wolfSSL/wolfssl #7512: [Bug]: DTLS 1.2 handshake fails when 2 packets arrive OOO

nunojpg, 2024-05-08:
### Version

v5.7.0-stable and master

### Description

Initial debug with Wireshark, with wolfSSL as server with TLS_ECDHE_ECDSA_WITH_CHACHA20_POLY1305_SHA256,
- Handshake consists of 7 packets
[image]
- When "Certificate Verify" (DTLS sequence number 4) arrives after "Change Cipher Spec" (DTLS sequence number 5) wolfSSL will not complete the connection.

kareem-wolfssl, 2024-05-08:
Hi @nunojpg,

Thanks for the report.  One of my colleagues will be reviewing this and responding shortly.

julek-wolfssl, 2024-05-09:
Hi @nunojpg, I was not able to replicate the failure. I used our udp-proxy tool to delay the certificate verify message after the change cipher spec. Please share your configuration details. Is wolfSSL being used on both sides?

```
# wolfSSL configured with
./configure --enable-dtls CPPFLAGS='-DWOLFSSL_DEBUG_DTLS' # WOLFSSL_DEBUG_DTLS added to force the change cipher spec to be sent separately from the finished
# test run with
./udp_proxy -p 12345 -s 127.0.0.1:11111 -S client -b 4
./examples/server/server -u -l TLS_ECDHE_ECDSA_WITH_CHACHA20_POLY1305_SHA256 -c certs/server-ecc.pem -k certs/ecc-key.pem -A certs/client-ecc-cert.pem
./examples/client/client -u -l TLS_ECDHE_ECDSA_WITH_CHACHA20_POLY1305_SHA256 -c certs/client-ecc-cert.pem -k certs/ecc-client-key.pem -A certs/ca-ecc-cert.pem -p 12345
```

[image]
The highlighted section is the client's flight with the re-arranged messages. Tested both with 5.7 and master.

Juliusz

nunojpg, 2024-05-10:
@julek-wolfssl 

I don't use wolfSSL on the client side. 

Thanks for your demo. I would have a much harder time to find how to test it in isolation, didn't know about your udp_proxy tool.

So, your demo shows the bug. As you can see the first handshake fails and the negotiation restarts with a 1s delay. Then udp_proxy doesn't reorder any more, so it works.

But your demo exposes a second bug.

The successful transaction is 38 packets (including payload). When I get it to fail then I loose the first handshake (about 15 packets) and retry, so let's say the "dumb" recovery is 53 packets.

Your demo shows a automatic retry but in total consumes 96 packets. The handshake state machine is probably deeply confused about this. 

Repeating your test without delay (./udp_proxy -p 12345 -s 127.0.0.1:11111 -S client)

[image]

And with delay (./udp_proxy -p 12345 -s 127.0.0.1:11111 -S client -b 4)

[image]

nunojpg, 2024-05-10:
I went to check if this is a DTLS requirement and it is:

https://datatracker.ietf.org/doc/html/rfc6347#section-3.2.2

julek-wolfssl, 2024-05-10:
@nunojpg 

I understand what you are saying with the disruption being incorrectly handled. I believe that the issue is that the dropped packet is the Change Cipher Spec message. We only store handshake messages for later processing and not any other type of message. The Change Cipher Spec message is currently only processed as it is received and it is not stored for later processing. That is the reason why the peer waits on a timeout for a retransmission from the peer.

Juliusz

nunojpg, 2024-05-10:
For some reason the client in my network started doing this, when 2 packets are transmitted to the internet in immediate sequence, the smaller one reaches the server first in 95% of the times. No sure if because of my network or something on the internet path. This means that wolfSSL fails the handshake 95% of the times. Otherwise I would have never found this. I have solved it by adding a delay on the client handshake transmission.

julek-wolfssl, 2024-05-10:
That is an interesting corner case. wolfSSL by default combines the Change Cipher Spec and Finished messages but it appears that your client does not concatenate any packets. Perhaps you can look into seeing if your client can combine messages that would fit into one MTU. This could decrease the amount of small packets being sent side by side.

nunojpg, 2024-05-10:
The wireshark logs were with WolfSSL on both sides. Only in my first comment I showed the Client->Server packets with Botan. But they appear to be identical. 
What do you see WolfSSL combining?

julek-wolfssl, 2024-05-10:
To group messages use [wolfSSL_set_group_messages](https://www.wolfssl.com/documentation/manuals/wolfssl/group__Setup.html#function-wolfssl_set_group_messages).

nunojpg, 2024-05-10:
Nice...that changed my handshake from 7/7 to 3/7. Have to see if the client supports the same.

julek-wolfssl, 2024-05-13:
@nunojpg, since we found a solution that works for you, can we marked this issue as solved?

Juliusz

nunojpg, 2024-05-13:
Hi, yes. The workaround suggested completely resolves the issue for me. I appreciate your help.

closed 2024-05-13 by nunojpg