wolfSSL/wolfssl #7521: How can I contact support@wolfssl.com

x509-name-testing, 2024-05-11:
### Version

None

### Description

Dear WolfSSL Team,

I want to report some issues, but my email to support@wolfssl.com is blocked. What should I do?

Here is the auto-reply from support@wolfssl.com:
```
We're writing to let you know that the group you tried to contact (support) may not exist, or you may not have permission to post messages to the group. A few more details on why you weren't able to post:

 * You might have spelled or formatted the group name incorrectly.
 * The owner of the group may have removed this group.
 * You may need to join the group before receiving permission to post.
 * This group may not be open to posting.
```

Many thanks.

x509-name-testing, 2024-05-13:
Can anyone have a look at this? Thanks!

embhorn, 2024-05-13:
Hello @x509-name-testing 

I can confirm that we are receiving messages at the support email alias. Please send an email to me directly and I can help you open a support ticket.

Please send it to `<my first name>@wolfssl.com` (I obfuscated the address to help prevent unwanted spam. My first name is below...)

Thanks,
Eric
wolfSSL Support

closed 2024-05-13 by embhorn