wolfSSL/wolfssl #7530: multiplication by 0 in wc_ecc_mulmod and points at infinity

a-def, 2024-05-14:
### Version

5.7.0

### Description

Hello,

I have some questions regarding `wc_ecc_mulmod` in `wolfcrypt/src/ecc.c`. Using secp256r1, when scalar `k` is zero, the result `R` is not at infinity and not on the curve. It seems rather odd. Shouldn't it be at infinity? Is it an expected/desired behavior in WolfSSL? 

To illustrate, I included these lines in:
- `test_wc_ecc_mulmod` 
```c
    /* Test multiplication by 0 (after Test bad args) */
    mp_int zero;
    ExpectIntEQ(mp_init(&zero), MP_OKAY);
    ExpectIntEQ(mp_set(&zero, 0), 0);
    ExpectIntEQ(wc_ecc_import_raw_ex(&key2, key1.dp->Gx, key1.dp->Gy,
        key1.dp->Af, ECC_SECP256R1), 0);
    ExpectIntEQ(wc_ecc_import_raw_ex(&key3, key1.dp->Gx, key1.dp->Gy,
        key1.dp->prime, ECC_SECP256R1), 0);
    ExpectIntEQ(wc_ecc_mulmod(&zero, &key2.pubkey,
        &key3.pubkey, wc_ecc_key_get_priv(&key2), wc_ecc_key_get_priv(&key3),
        1), 0);
    ExpectIntEQ(wc_ecc_point_is_at_infinity(&key2.pubkey), 1);
    mp_free(&zero);
```
- `test_wolfSSL_EC_POINT`
```c
    /* check multiplication by zero */
    BIGNUM* zero = NULL;
    ExpectNotNull(zero = BN_new());
    BN_zero(zero);
    ExpectIntEQ(EC_POINT_mul(group, new_point, zero, NULL, NULL, ctx), 1);
    ExpectIntEQ(
        EC_POINT_is_on_curve(group, new_point, ctx) &&
        EC_POINT_is_at_infinity(group, new_point), 1);
    BN_free(zero);
```

configuration: --enable-all-crypto, --enable-opensslextra, --enable-opensslall, --enable-asm, --enable-intelasm

SparkiDev, 2024-05-14:
Hi @a-def

The function doesn't support zero in 5.7.0.

I've put up a PR that fixes this and added a test case to ensure we keep this behavior.

Id this all you need?

dgarske, 2024-05-15:
Hi @a-def ,

I've re-opened this (it auto closed) as I would like to have your feedback on the PR. Please let us know if this PR resolves the issue or if you see anything else.

Thanks,
David Garske, wolfSSL

a-def, 2024-05-15:
Hi,

Great, thank you for this quick fix. That's all.

SparkiDev, 2024-05-30:
Hi @a-def 

I've put up a pull request with the change to set z to 1.
I did debate this in my mind but the safe add and double don't use the z so I thought it was safe!

Let me know if this fixes things for you.

Sean :-)

PR #7532 (2024-05-15): ECC: handle zero in wc_ecc_mulmod()
# Description

Public API needs to handle multiplying by zero as the underlying code doesn't and needn't.

Fixes #7530

# Testing

Added test case.
Failed before change made to wc_ecc_mulmod().

# Checklist

 - [x] added tests
 - [ ] updated/added doxygen
 - [ ] updated appropriate READMEs
 - [ ] Updated manual and documentation
PR #7600 (2024-06-18): ECC: when multiplying by zero, set z to 1
# Description

Make sure zero times a point is infinity but z is 1 as it is assumed later on.

Fixes #7530

# Testing

./configure '--disable-shared' 'CFLAGS=-DWOLFSSL_PUBLIC_MP'

# Checklist

 - [x] added tests
 - [ ] updated/added doxygen
 - [ ] updated appropriate READMEs
 - [ ] Updated manual and documentation
closed 2024-06-18 by JacobBarthelmeh