wolfSSL/wolfssl #7543: Request more openssl compatibility for gSOAP.

STVHA, 2024-05-16:
I found several incompatibility issues when I built gSOAP's plugins with wolfssl.

1. OpenSSL version checking issue
One of gSOAP's plugins (mecevp) uses OPENSSL_VERSION_NUMBER to choose functions to compile, the check like this
```
#if (OPENSSL_VERSION_NUMBER < 0x30000000L)
// use deprecated functions
#else
// use new functions 
#endif
```
Unfortunately the API version compatibility defined in opensslv.h always clamps the OpenSSL version to a range of [0x009070dfL, 0x10100000L]. So whatever OPENSSL_VERSION_NUMBER I defined finally it falls down to the range above.
I think wolfssl should allow the user to define OpenSSl version up to at least 3.0.0, and it'd be better ready for new versions in the future.

2. Functions do not exist in wolfssl
Please append more these functions to the part of EVP:
```
int EVP_SealInit(EVP_CIPHER_CTX *ctx, const EVP_CIPHER *type,
                 unsigned char **ek, int *ekl, unsigned char *iv,
                 EVP_PKEY **pubk, int npubk);
int EVP_SealUpdate(EVP_CIPHER_CTX *ctx, unsigned char *out,
                   int *outl, unsigned char *in, int inl);
int EVP_SealFinal(EVP_CIPHER_CTX *ctx, unsigned char *out, int *outl);

int EVP_OpenInit(EVP_CIPHER_CTX *ctx, EVP_CIPHER *type, unsigned char *ek,
                 int ekl, unsigned char *iv, EVP_PKEY *priv);
int EVP_OpenUpdate(EVP_CIPHER_CTX *ctx, unsigned char *out,
                   int *outl, unsigned char *in, int inl);
int EVP_OpenFinal(EVP_CIPHER_CTX *ctx, unsigned char *out, int *outl);

int EVP_EncryptUpdate(EVP_CIPHER_CTX *ctx, unsigned char *out,
                                 int *outl, const unsigned char *in, int inl);
```
3. Conflict struct name
The `struct COND_TYPE` defined in wolfcrypt\types.h has name conflicts with another struct defined in a gSOAP's thread plugin having different members but same purpose. I am not sure which is more dominant than the other.
Is it possible to rename this one to something like `WOLFSSL_COND_TYPE`?

anhu, 2024-05-16:
Hello @STVHA , 

My name is Anthony and I am a member of the wolfSSL team. I see you are trying to build wolfSSL with gSOAP.  It appears gSOAP is using openSSL3 APIs.   We haven't seen much interest in  OpenSSL 3.x.y. I think this would be classified as a feature request on our end.  

Can you let us know a bit more about yourself and your project? If for privacy reasons you don't want to reveal details about yourself or project, you can open a feature request by sending a mail to support@wolfssl.com and giving details there.  Please also reference this github issue.  

I shall leave this open for now.  I will be monitoring this issue as well as our technical support channels and await a message from you. 

Warm regards, Anthony.

STVHA, 2024-05-17:
Thank you Anthony for your reply.

I am working on a project using gSOAP to communicate with IP cameras, it is working fine with the latest version of openssl. Recently one new module already using wolfssl needs to be integrated into the project so I have an idea to unify the usage of crypto libraries to a single one, wolfssl, I prefer it. Honestly I don't want to use deprecated functions of openssl, not to mention the wolfssl's openssl lacks many functions of the openssl v1.x also, that is why I posted the request here.
According to your reply I understand that is the business strategy of your team. It seem not possible to have the requested functions in the near future. Currently I should find another way or keep using openssl v3.x.
Please feel free to close this request for me.

Thank you. 
Regards

anhu, 2024-06-27:
Thanks @STVHA  !!  As you suggested, this message will close this ticket.

closed 2024-06-27 by anhu