wolfSSL/wolfssl #7579: error: api_kyber.h No such file

jelliyjane, 2024-05-23:
### Version

5.7.0

### Description

The file 'api_kyber.h' is missing when included in the 'wolfssl/wolfcrypt/ext_kyber.h' file. I am using stm Cube IDE. If anyone has experienced this issue, how can i fix it?

[image]

embhorn, 2024-05-23:
Hi @jelliyjane 

We are updating the cube packs. @lealem47 please share the updated version when available.

anhu, 2024-05-23:
Hi, @jelliyjane 

My name is Anthony and I am a member of the wolfSSL team. 

I think you'll find this documentation very helpful: 

https://github.com/wolfSSL/wolfssl-examples/tree/master/pq/stm32

If you'd prefer a video tutorial, please see this recorded webinar: 

https://www.youtube.com/watch?v=OK6MKXYiVBY

I hope this helps!

Here at wolfSSL, we love knowing how our code is being used.  Can you please let us know a bit about your project and yourself?  For example: 

- Is this effort out of personal, academic or business interest? 
- What are your goals? 
- Are there any institutions that are associated with this project? 
- Where are you geographically located? 

If you do not want to share this information on a public forum such as GitHub, you can email me at anthony@wolfssl.com

Warm regards, Anthony

jelliyjane, 2024-06-09:
Thank you for your help. I successfully executed it!

I'm a graduate student currently conducting research on PQC and TLS implementation on embedded boards. My project focuses on enhancing the efficiency of PQC-TLS for embedded systems.

Best regards,

closed 2024-06-19 by jelliyjane