wolfSSL/wolfssl #7635: Guidance to use with Arduino as an IDF component in PIO

hitecSmartHome, 2024-06-10:
### Version

latest

### Description

I can't find a comprehensive guide to replace ESP-TLS with WolfSSL.

I have inluded wolfssl in my libdeps in `platformio.ini`

```ini
lib_deps = 
    https://github.com/joltwallet/esp_littlefs.git
    wolfssl/wolfSSL@^5.7.0-rev.3b
```

`pio -t run menuconfig -> Component config -> ESP-TLS -> Choose SSL/TLS library for ESP-TLS`
**I only have mbedTLS**

kareem-wolfssl, 2024-06-10:
Hi @hitecSmartHome ,

To get wolfSSL working with ESP-IDF/Espressif, check out our example with documentation here: https://github.com/wolfSSL/wolfssl/tree/master/IDE/Espressif

gojimmypi, 2024-06-11:
Hi @hitecSmartHome 

> I can't find a comprehensive guide to replace ESP-TLS with WolfSSL.

We do not at this time have such a guide. Further, as with many engineering topics, replacing ESP-TLS: "_it depends_".

Without additional details on your project, my recommendation on the replacement is just that: completely remove the ESP-TLS and use wolfSSL directly unless there's a compelling reason to need both.

Although the ESP-IDF offers the ability to switch between crypto libraries, it adds an unnecessary layer of complexity. Further, the [espressif/esp-wolfssl](https://github.com/espressif/esp-wolfssl) is not managed by wolfSSL and is typically stale. As of today, the submodule is 2 years old.

That said, I'd like to help. Can you provide a code example? 

I wrote a [wolfSSL blog on PlatformIO](https://www.wolfssl.com/wolfssl-now-supported-on-platformio/), There's also the [IDE/PlatformIO](https://github.com/wolfSSL/wolfssl/tree/master/IDE/PlatformIO) with benchmark and test examples, but I've not had a chance to add the TLS Client and Server examples that are [available for Arduino](https://github.com/wolfSSL/wolfssl/tree/master/IDE/ARDUINO/sketches).

Regarding the error: `Found both ESPIDF and ARDUINO. Pick one.` - Please note that there are two PlatformIO libraries: 

- [wolfssl/wolfssl](https://registry.platformio.org/libraries/wolfssl/wolfssl)
- [wolfssl/Arduino-wolfSSL](https://registry.platformio.org/libraries/wolfssl/Arduino-wolfSSL)

I've not yet used both the IDF and Arduino concurrently. Are you using the [Arduino as an ESP-IDF component](https://docs.espressif.com/projects/arduino-esp32/en/latest/esp-idf_component.html)? If so, perhaps try using the `Arduino-wolfSSL` library instead.

Thank you for your interest in wolfSSL. We look forward to learning more about your project.

hitecSmartHome, 2024-06-12:
Thank you very much for the explanation.
I tought I can just replace mbedTLS in menuconfig with wolfssl if it's in libdeps and let the IDF libraries use that instead.
I'm not using it in my own code for any specific reasons, just https, wss connections trought IDF libraries.

ESP-TLS uses a lot of RAM, I needed a way to minimise this. But if it does not work out of the box I have no time to write my own wrappers.

Thank you very much again!

closed 2024-06-12 by hitecSmartHome

gojimmypi, 2024-06-12:
@hitecSmartHome that's correct: one _should_ be able to select wolfSSL in the ESP-IDF. 

Coincidentally, I'll be focusing on improving this in the very near future and updating the referenced wolfSSL library at [espressif/esp-wolfssl](https://github.com/espressif/esp-wolfssl). I have a new [CMakeLists.txt in the examples](https://github.com/wolfSSL/wolfssl/blob/master/IDE/Espressif/ESP-IDF/examples/template/components/wolfssl/CMakeLists.txt) that allows pointing to wolfSSL source code via searching parent directories, assigning cmake variable, or setting environment variable for `WOLFSSL_ROOT`.

If you'd like, I could help with your project? It would be great to have your feedback for either the ESP-TLS, or using wolfSSL directly. Note there are also wolfSSL TLS [client](https://github.com/wolfSSL/wolfssl/tree/master/IDE/Espressif/ESP-IDF/examples/wolfssl_client) and [server](https://github.com/wolfSSL/wolfssl/tree/master/IDE/Espressif/ESP-IDF/examples/wolfssl_server) examples to help you get started.

I've been able to do TLS 1.3 in as little as 32KB RAM + 256KB flash on the Arduino Nano 33 IoT SAMD21.   See [tweet](https://x.com/gojimmypi/status/1765449044669923814) thread.

Best Regards.