wolfSSL/wolfssl #7641: --enable-quic --enable-opensslcoexist options are mutually exclusive

alexsn, 2024-06-13:
### Version

5.7.0

### Description

`./configure --enable-quic --enable-opensslcoexist` fails with the following error:
**configure: error: Cannot use --enable-opensslcoexist with --enable-opensslextra**

This occurs due to [this setting](https://github.com/wolfSSL/wolfssl/blame/master/configure.ac#L2034-2037).

Any workaround I can use as I need both wolfssl and openssl to be linked into my application.

kareem-wolfssl, 2024-06-13:
Hi @alexsn ,

This is an intentional error as our QUIC support depends extensively on our OpenSSL compatibility layer which is not compatible with OpenSSL coexistence.  Are you able to share some information about your use case for linking wolf and OpenSSL into your application?  Our OpenSSL compatibility layer supports many OpenSSL APIs and can be used in place of OpenSSL.

alexsn, 2024-06-14:
Hey @kareem-wolfssl, thanks for the reply. The original issue I encountered was multiple function definitions when statically linking openssl with wolfssl.
I was able to get openssl / wolfssl to coexist without `--enable-opensslcoexist` by compiling wolfssl with NO_WOLFSSL_STUB and adding the following small change:
```
#if !defined(NETOS) && !defined(NO_WOLFSSL_STUB)
void ERR_load_SSL_strings(void)
{

}
#endif
```
Is this patch acceptable as I want `ERR_load_SSL_strings` to be removed in the presence of `NO_WOLFSSL_STUB`

closed 2024-07-08 by alexsn