wolfSSL/wolfssl #7674: Can not Collect Coverage Properly

songshangru, 2024-06-22:
### Version

wolfssl-5.7.0-stable

### Description

I compile wolfssl in ubuntu 20 , cmd as follow:

autoreconf --install 
CFLAGS="-g -O0 -ftest-coverage" LDFLAGS="-fprofile-arcs -ftest-coverage" ./configure "--enable-tls13" "--enable-session-ticket" "--enable-debug" "--disable-shared"
CFLAGS="-g -O0 -ftest-coverage" LDFLAGS="-fprofile-arcs -ftest-coverage" make

But the compiled program ./examples/server/server fails to correctly collect coverage, with all coverage metrics showing 0. Additionally, I couldn't find the corresponding .gcno files, only server-server.gcno. What could be the problem?

embhorn, 2024-06-24:
Hi @songshangru 

Thanks for contacting wolfSSL Support. It looks like you missed running the tests with `make check`. Here is how we generate a code coverage report:

```
./autogen.sh
./configure LDFLAGS=-lgcov CFLAGS="-O0 -fprofile-arcs -ftest-coverage" CXXFLAGS="-O0 -fprofile-arcs -ftest-coverage" --enable-all 
make check
lcov -t 'wolfssl' -o coverage.info -c -d .
mkdir wolfSSL-Code-Coverage-Report
cd wolfSSL-Code-Coverage-Report
genhtml ../coverage.info
cd ..
zip -r wolfSSL-Code-Coverage-Report.zip wolfSSL-Code-Coverage-Report
```

Here is an example:
[wolfSSL-Code-Coverage-Report.zip](https://github.com/user-attachments/files/15956929/wolfSSL-Code-Coverage-Report.zip)

songshangru, 2024-06-25:
Sorry, I would not want to run 'make check', because I only want to collect the coverage when run './example/server/server' with arguements. But  all coverage metrics showing 0 when I run this cmd.