wolfSSL/wolfssl #7686: [5.7.0] Unit.test failed on DTLS v1.3 during make test proccess

armdn, 2024-06-26:
### Version

5.7.0

### Description

In 'make test' process, section unit.test ended up with FAIL status.

Here is the log of test-suite:

`SSL version is DTLSv1.3
SSL cipher suite is TLS_AES_256_GCM_SHA384
SSL signature algorithm is SHA256
Session timeout set to 500 seconds
Client Random : 5E0576979C06815FDC4367088C070727D6D1DB1EE50BwolfSSL_connect error -308, error state on socket
SSL_accept error -308, error state on socket
wolfSSL error: SSL_accept failed
wolfSSL error: wolfSSL_connect failed
F5C56E2F50DDE8BF6F0C
Client message: hello wolfssl!
I hear you fa shizzle!
trying server command line[1831]: SuiteTest -u -v 4 -l TLS13-AES256-GCM-SHA384 --pqc KYBER_LEVEL3 -2 -p 0
Using Post-Quantum KEM: KYBER_LEVEL3
trying client command line[1832]: SuiteTest -u -v 4 -l TLS13-AES256-GCM-SHA384 --pqc KYBER_LEVEL3 -2 -p 55848 -H defCipherList
Using default cipher list for testing
Using Post-Quantum KEM: KYBER_LEVEL3
FAIL scripts/unit.test (exit status: 1)`

Weird error 'SSL_accept error -308, error state on socket' occurs during DTLS v1.3 test...

Compilation goes smoothly without warnings... Am i missed something in config?
Thanks for the help!

anhu, 2024-06-27:
Hi @armdn , 

Can you please let us know what configure flags you were using? 

Warm regards, Anthony

armdn, 2024-06-27:
> Hi @armdn ,
> 
> Can you please let us know what configure flags you were using?
> 
> Warm regards, Anthony

Sure

`--enable-experimental --enable-kyber --enable-libssh2 --enable-openssh --enable-aescfb --enable-aesni --enable-intelasm --enable-dsa --enable-dtls --enable-dtls13 --enable-dtls-frag-ch --enable-aesxts`

anhu, 2024-06-28:
Hi @armdn 

I've been able to simplify the steps to reproduce this: 

```
./configure --enable-experimental --enable-kyber --enable-dtls --enable-dtls13 --enable-dtls-frag-ch
make all 
./examples/server/server  -u -v 4 -l TLS13-AES256-GCM-SHA384 --pqc KYBER_LEVEL3
```
Then in another terminal: 
```
./examples/client/client -u -v 4 -l TLS13-AES256-GCM-SHA384 --pqc KYBER_LEVEL3 -H defCipherList
```

At this point both the server and and client just wait forever.  I will need to talk to my colleagues about this to determine the best course of action. 

Thank you so much for letting us know about this issue.  Please stay tuned.  There should be updates to our progress. 

Warm regards, Anthony

julek-wolfssl, 2024-08-28:
@armdn thank you for the report. The issue was that we were incorrectly checking the size of the ClientHello message. We didn't detect that we were going to fragment and didn't deal with it appropriately. https://github.com/wolfSSL/wolfssl/pull/7912 fixes the issue.

Juliusz

armdn, 2024-08-28:
Thats great! Will test it at last!

closed 2024-08-29 by douzzer

julek-wolfssl, 2024-08-30:
https://github.com/wolfSSL/wolfssl/pull/7912 has been merged.