wolfSSL/wolfssl #7720: Licensing: FLOSS license exception, what is it?

rmg-x, 2024-07-06:
### Version

N/A

### Description

The wolfSSL website has a page dedicated to a "FLOSS exception" here:
https://www.wolfssl.com/docs/floss-exception/

However, neither the website nor this repository seem to link or mention this exception anywhere. The webpage is only discoverable when using a search engine.

The `COPYING` file in this repository has the following section:
```
10. If you wish to incorporate parts of the Program into other free
programs whose distribution conditions are different, write to the author
to ask for permission.  For software which is copyrighted by the Free
Software Foundation, write to the Free Software Foundation; we sometimes
make exceptions for this.  Our decision will be guided by the two goals
of preserving the free status of all derivatives of our free software and
of promoting the sharing and reuse of software generally.
```
...which seems similar but not the exact same. 

Could all of this be clarified by wolfSSL?

cconlon, 2024-07-08:
Hi @rmg-x, we'll look adjusting our website linkage of this page.

Can you please email licensing@wolfssl.com to continue the licensing discussion around our FLOSS exception?

rmg-x, 2024-07-08:
That's okay, I was just curious what it was and how it worked. 
Being a GPLv2 project, I thought it was interesting that exceptions are offered for more permissively licensed projects.

closed 2024-07-08 by rmg-x

bgermann, 2026-01-13:
Can you update the exception to apply to the GPL-v3-or-later that current wolfSSL versions are licensed under?