wolfSSL/wolfssl #7724: Duplicate poly1305_blocks symbol when building with --enable-armasm and linking with both {wolf,open}ssl

alexsn, 2024-07-08:
### Contact Details

alexsn@meta.com

### Version

5.7.0

### Description

When building with `WOLFSSL_ARMASM` and linking with both openssl and wolfssl I get a duplicate poly1305_blocks symbol as openssl also defined it [here](https://github.com/openssl/openssl/blob/master/crypto/poly1305/poly1305.c#L427).

I was able to resolve the issue by renaming poly1305_blocks -> wc_Poly1305_Blocks thought the code base:
```
diff --git a/third-party/wolfssl/wolfcrypt/src/poly1305.c b/third-party/wolfssl/wolfcrypt/src/poly1305.c
--- a/third-party/wolfssl/wolfcrypt/src/poly1305.c
+++ b/third-party/wolfssl/wolfcrypt/src/poly1305.c
@@ -263,8 +263,8 @@
 This local function operates on a message with a given number of bytes
 with a given ctx pointer to a Poly1305 structure.
 */
-static int poly1305_blocks(Poly1305* ctx, const unsigned char *m,
-                     size_t bytes)
+static int wc_Poly1305_Blocks(Poly1305* ctx, const unsigned char *m,
+                              size_t bytes)
 {
 #if defined(WOLFSSL_X86_64_BUILD) && defined(USE_INTEL_SPEEDUP)
     /* AVX2 is handled in wc_Poly1305Update. */
@@ -398,7 +398,7 @@
 This local function is used for the last call when a message with a given
 number of bytes is less than the block size.
 */
-static int poly1305_block(Poly1305* ctx, const unsigned char *m)
+static int wc_Poly1305_Block(Poly1305* ctx, const unsigned char *m)
 {
 #if defined(WOLFSSL_X86_64_BUILD) && defined(USE_INTEL_SPEEDUP)
     /* No call to poly1305_block when AVX2, AVX2 does 4 blocks at a time. */
@@ -407,7 +407,7 @@
     RESTORE_VECTOR_REGISTERS();
     return 0;
 #else
-    return poly1305_blocks(ctx, m, POLY1305_BLOCK_SIZE);
+    return wc_Poly1305_Blocks(ctx, m, POLY1305_BLOCK_SIZE);
 #endif
 }
 #endif /* !defined(WOLFSSL_ARMASM) || !defined(__aarch64__) */
@@ -541,7 +541,7 @@
         for (i = i + 1; i < POLY1305_BLOCK_SIZE; i++)
             ctx->buffer[i] = 0;
         ctx->finished = 1;
-        poly1305_block(ctx, ctx->buffer);
+        wc_Poly1305_Block(ctx, ctx->buffer);
     }
     /* fully carry h */
@@ -609,7 +609,7 @@
         for (; i < POLY1305_BLOCK_SIZE; i++)
             ctx->buffer[i] = 0;
         ctx->finished = 1;
-        poly1305_block(ctx, ctx->buffer);
+        wc_Poly1305_Block(ctx, ctx->buffer);
     }
     /* fully carry h */
@@ -772,7 +772,7 @@
             ctx->leftover += want;
             if (ctx->leftover < POLY1305_BLOCK_SIZE)
                 return 0;
-            poly1305_block(ctx, ctx->buffer);
+            wc_Poly1305_Block(ctx, ctx->buffer);
             ctx->leftover = 0;
         }
@@ -781,11 +781,11 @@
             size_t want = ((size_t)bytes & ~((size_t)POLY1305_BLOCK_SIZE - 1));
 #if !defined(WOLFSSL_ARMASM) || !defined(__aarch64__)
             int ret;
-            ret = poly1305_blocks(ctx, m, want);
+            ret = wc_Poly1305_Blocks(ctx, m, want);
             if (ret != 0)
                 return ret;
 #else
-            poly1305_blocks(ctx, m, want);
+            wc_Poly1305_Blocks(ctx, m, want);
 #endif
             m += want;
             bytes -= (word32)want;
diff --git a/third-party/wolfssl/wolfcrypt/src/port/arm/armv8-poly1305.c b/third-party/wolfssl/wolfcrypt/src/port/arm/armv8-poly1305.c
--- a/third-party/wolfssl/wolfcrypt/src/port/arm/armv8-poly1305.c
+++ b/third-party/wolfssl/wolfcrypt/src/port/arm/armv8-poly1305.c
@@ -187,8 +187,7 @@
     );
 }
-void poly1305_blocks(Poly1305* ctx, const unsigned char *m,
-                            size_t bytes)
+void wc_Poly1305_Blocks(Poly1305* ctx, const unsigned char *m, size_t bytes)
 {
     __asm__ __volatile__ (
         /* If less than 4 blocks to process then use regular method */
@@ -872,7 +871,7 @@
     poly1305_blocks_16(ctx, m, bytes);
 }
-void poly1305_block(Poly1305* ctx, const unsigned char *m)
+void wc_Poly1305_Block(Poly1305* ctx, const unsigned char *m)
 {
     poly1305_blocks_16(ctx, m, POLY1305_BLOCK_SIZE);
 }
@@ -1092,7 +1091,7 @@
         for (; i < POLY1305_BLOCK_SIZE; i++)
             ctx->buffer[i] = 0;
         ctx->finished = 1;
-        poly1305_block(ctx, ctx->buffer);
+        wc_Poly1305_Block(ctx, ctx->buffer);
     }
     __asm__ __volatile__ (
diff --git a/third-party/wolfssl/wolfssl/wolfcrypt/poly1305.h b/third-party/wolfssl/wolfssl/wolfcrypt/poly1305.h
--- a/third-party/wolfssl/wolfssl/wolfcrypt/poly1305.h
+++ b/third-party/wolfssl/wolfssl/wolfcrypt/poly1305.h
@@ -125,9 +125,9 @@
     word32 addSz, const byte* input, word32 sz, byte* tag, word32 tagSz);
 #if defined(__aarch64__ ) && defined(WOLFSSL_ARMASM)
-void poly1305_blocks(Poly1305* ctx, const unsigned char *m,
+void wc_Poly1305_Blocks(Poly1305* ctx, const unsigned char *m,
                             size_t bytes);
-void poly1305_block(Poly1305* ctx, const unsigned char *m);
+void wc_Poly1305_Block(Poly1305* ctx, const unsigned char *m);
 #endif
 #ifdef __cplusplus
```
Is it possible to ask for a rename of poly1305_blocks to anyone else or have a compile time option to do so.

### Reproduction steps

./configure --enable-quic --enable-earlydata --enable-psk --enable-harden --enable-altcertchains --prefix=$PWD/build --enable-armasm --enable-fastmath --enable-postauth

SparkiDev, 2024-07-09:
Hi @alexsn,

I've put up a pull request that changes the names of functions and symbols to include 'aarch64'.
This will make them unique compared to OpenSSL.

Let me know if this works for you!

Sean

alexsn, 2024-07-09:
> Hi @alexsn,
> 
> I've put up a pull request that changes the names of functions and symbols to include 'aarch64'. This will make them unique compared to OpenSSL.
> 
> Let me know if this works for you!
> 
> Sean

Hey Sean,

I've tested this locally and it now links without conflicts.

Thanks, Alex.

SparkiDev, 2024-07-09:
Hi @alexsn

Thanks for the review!
I'll leave the ticket to be closed when the PR is merged if that is alright with you.

Thanks,
Sean

PR #7728 (2024-07-16): Poly1305 AArch64: unique naming of asm funcs
# Description

Change function names to ensure no clash with OpenSSL.
Specifically: poly1305_blocks()

Fixes: #7724

# Testing

./configure '--disable-shared' 'LDFLAGS=--static' '--host=aarch64' 'CC=aarch64-linux-gnu-gcc' '--enable-armasm'
make
poly1305_blocks() no longer a function name in wolfSSL.

# Checklist

 - [ ] added tests
 - [ ] updated/added doxygen
 - [ ] updated appropriate READMEs
 - [ ] Updated manual and documentation
closed 2024-07-16 by JacobBarthelmeh