wolfSSL/wolfssl #7729: pkcs11 TLS example: how to avoid performing symmetric crypto on token?

space88man, 2024-07-09:
### Version

5.7.2

### Description

Using wolfSSL TLS with a PKCS#11 private key : how can I use the token for only asymmetric operations and perform symmetric operations in the library?

With the PKCS#11 server example : https://github.com/wolfSSL/wolfssl-examples/blob/master/pkcs11/
server-tls-pkcs11-ecc.c

wolfSSL seems to use the token for symmetric operations - but my token doesn't like this template -

```
# pkcs11-spy dump
30: C_CreateObject
2024-07-09 22:50:11.314
[in] hSession = 0x1
[in] pTemplate[5]: 
    CKA_CLASS             CKO_SECRET_KEY       
    CKA_KEY_TYPE          CKK_AES            
    CKA_ENCRYPT           True
    CKA_DECRYPT           True
    CKA_VALUE             0000000000b0f66c / 32
    00000000  57 CE E3 50 D8 E0 F9 C3 2E 3F D7 83 BA 1B 5B 13  W..P.....?....[.
    00000010  0E DF 00 98 ED 2B 50 AB 72 7C CD 25 D5 10 8B 1A  .....+P.r|.%....
Returned:  209 CKR_TEMPLATE_INCONSISTENT
```

With NSS softoken, the symmetric operations also fail:

```
127: C_CreateObject
2024-07-09 23:15:01.487
[in] hSession = 0x1000001
[in] pTemplate[5]: 
    CKA_CLASS             CKO_SECRET_KEY       
    CKA_KEY_TYPE          CKK_AES            
    CKA_ENCRYPT           True
    CKA_DECRYPT           True
    CKA_VALUE             000000000233bbec / 32
    00000000  A7 5F 7C 5F 20 E8 88 2F 99 03 0D 91 71 24 02 AF  ._|_ ../....q$..
    00000010  22 FC 6A 70 D7 63 4C E8 E4 B5 AC 7A 12 53 98 51  ".jp.cL....z.S.Q
[out] *phObject = 0xf
Returned:  0 CKR_OK

128: C_DecryptInit
2024-07-09 23:15:01.487
[in] hSession = 0x1000001
[in] pMechanism->type = CKM_AES_GCM                  
[in] pMechanism->pParameter->pIv[ulIvLen] 000000000233d180 / 12
    00000000  2C 4D EC 1D 18 85 0B A1 81 1C 57 90              ,M........W.    
[in] pMechanism->pParameter->ulIvBits = 0x0
[in] pMechanism->pParameter->pAAD[ulAADLen] 000000000233a560 / 13
    00000000  00 00 00 00 00 00 00 00 16 03 03 00 10           .............   
[in] pMechanism->pParameter->ulTagBits = 128
[in] hKey = 0xf
Returned:  113 CKR_MECHANISM_PARAM_INVALID

```

SparkiDev, 2024-07-09:
Hi @space88man,

If you do not want to perform symmetric algorithm operations with PKCS#11 then you can define: NO_PKCS11_AES.
You may also want to define: NO_PKCS11_HMAC and NO_PKCS11_RNG.

You can also implement your own callback that calls wc_Pkcs11_CryptoDevCb for only the algorithms you want.
Anything not using PKCS#11 needs to return NOT_COMPILED_IN.
The devId can be used to identify where the operation is coming from. That is use a different identifier for TLS to another context.

Sean

space88man, 2024-07-10:
Hi @SparkiDev 

Thanks — I tried something like this: where 1 = real PKCS#11 devID, 2 = filter devID,
and it works. Does this look correct to you?

```
int my_FilterCb(int devId, wc_CryptoInfo* info, void* ctx)
{
        int ret = 0;
        // assert (devId == 2);
        fprintf(stderr, "[%d] filter callback -> %d\n", devId, info->algo_type);
        if (info->algo_type != WC_ALGO_TYPE_PK)
                return NOT_COMPILED_IN;
        // call the real PKCS#11 device
        return wc_Pkcs11_CryptoDevCb(1, info, ctx);
}
```

SparkiDev, 2024-07-10:
Hi @space88man,

The code looks good assuming you are setting the devId for TLS connections to 2.
Note: 'ret' is not needed in the function as you have it.

Is there anything more for this issue?

Thanks,
Sean

closed 2024-07-10 by space88man