wolfSSL/wolfssl #7739: RFE: modernize PKCS#11 RSA mechanism to PKCS/PSS/OAEP

space88man, 2024-07-11:
### Version

5.7.2

### Description

It is discouraged to call the raw RSA mechanims on a token `CKM_RSA_X_509` which is used by wolfCrypt PKCS#11.
This mechanism is not supported by Thales LUNA.

The enhancement would be for wolfCrypt -> CryptoDev -> token to use modern mechamisms like RSA_PKCS/RSA_PKCS_PSS/RSA_PKCS_OAEP for signing and decryption.

My testing observations:
* wolfCrypt only uses CKM_RSA_X_509 (supported by softoken but not LUNA)
* OpenSSL openssl-pkcs11 or pkcs11-provider use RSA_PKCS or RSA_PKCS_PSS

space88man, 2024-07-11:
I am interested in this feature but not sure how the wolfCrypt RSA operation splits work between software (padding/hashing(PKCS, PSS)) and CryptoDev(raw encryption/decryption). Any pointers to the source code would be very helpful.

SparkiDev, 2024-07-11:
Hi @space88man,

While it would be nice to have the whole operation done on the device, it was cleaner for our code to have only the RSA operation done down there.

The wc_CryptoCb_Rsa() calls in rsa.c are where the PKCS#11 code is invoked.

Note that the RSA-PSS sign could be easily modified to call a new callback, the verify is broken into two parts which would take a bit more effort but again is not difficult. Similarly the OAEP encrypt/decrypt would require new callback but would be easy to implement.

Sean

Sean

space88man, 2024-07-12:
> 
> While it would be nice to have the whole operation done on the device, it was cleaner for our code to have only the RSA operation done down there.
> 

Yes - it looks cleaner to only dispatch the raw RSA operation and keep the rest the same.

The glitch for me ATM is that LUNA disables CKM_RSA_X_509 in FIPS mode 😣. It supports raw RSA in non-FIPS mode.

SparkiDev, 2024-07-12:
Hi @space88man,

My preference is to have new calls to wc_CryptoCb_Rsa higher up.
New types could be added: WC_PK_TYPE_RSA_PSS, WC_PK_TYPE_RSA_OAEP.
Then it could be handled in the callback which operation is wanted.
This risks existing callback implemenations failing but can be disables with a #ifdef.

Sean

space88man, 2024-07-12:
> My preference is to have new calls to wc_CryptoCb_Rsa higher up. New types could be added: WC_PK_TYPE_RSA_PSS, WC_PK_TYPE_RSA_OAEP. Then it could be handled in 

Let me work out something for WC_PK_TYPE_RSA_PKCS.

space88man, 2024-07-14:
WIP: test files both software RSA(to ensure nothing broke) and token RSA for upcoming PR for RSA PKCS-offload. NSS Softoken with RSA private key included.

[testing-v1.zip](https://github.com/user-attachments/files/16215323/testing-v1.zip)

dgarske, 2024-09-04:
Hi @space88man ,

This support has been added in PR https://github.com/wolfSSL/wolfssl/pull/7907. Please review and let us know if you have any feedback.

Thanks,
David Garske, wolfSSL

closed 2024-11-15 by dgarske