wolfSSL/wolfssl #7749: Cannot build wolfssl with liboqs integration

notyan, 2024-07-13:
### Version

wolfssl-5.7.2

### Description

## Description
trying yo install and build wolfSSL with liboqs following the https://github.com/wolfSSL/wolfssl/blob/master/INSTALL point 15
Following all the step provided by the documentation, but the wolfssl build have this problem
Here the configure i set
```
$ ./configure --with-liboqs --enable-experimental  
Configuration summary for wolfssl version 5.7.2

   * Installation prefix:        /usr/local
   * System type:                pc-linux-gnu
   * Host CPU:                   x86_64
   * C Compiler:                 gcc
   * C Flags:                       -Werror -Wno-pragmas -Wall -Wextra -Wunknown-pragmas --param=ssp-buffer-size=1 -Waddress -Warray-bounds -Wbad-function-cast -Wchar-subscripts -Wcomment -Wfloat-equal -Wformat-security -Wformat=2 -Wmaybe-uninitialized -Wmissing-field-initializers -Wmissing-noreturn -Wmissing-prototypes -Wnested-externs -Wnormalized=id -Woverride-init -Wpointer-arith -Wpointer-sign -Wshadow -Wsign-compare -Wstrict-overflow=1 -Wswitch-enum -Wundef -Wunused -Wunused-result -Wunused-variable -Wwrite-strings -fwrapv
   * C++ Compiler:
   * C++ Flags:
   * CPP Flags:
   * CCAS Flags:
   * LD Flags:
   * LIB Flags:                   -pie -z relro -z now -Werror
   * Library Suffix:
   * Debug enabled:              no
   * Coverage enabled:
   * Warnings as failure:        yes
   * make -j:                    3
   * VCS checkout:               yes

   Features
   * Experimental settings:      Allowed
   * FIPS:                       no
   * Single threaded:            no
   * Filesystem:                 yes
   * OpenSSH Build:              no
   * OpenSSL Extra API:          no
   * OpenSSL Coexist:            no
   * Old Names:                  yes
   * Max Strength Build:         no
   * Distro Build:               no
   * Reproducible Build:         no
   * Side-channel Hardening:     yes
   * Single Precision Math:      no
   * SP implementation:          all
   * Fast Math:                  no
   * Heap Math:                  no
   * Assembly Allowed:           yes
   * sniffer:                    no
   * snifftest:                  no
   * ARC4:                       no
   * AES:                        yes
   * AES-NI:                     no
   * AES-CBC:                    yes
   * AES-CBC length checks:      no
   * AES-GCM:                    yes
   * AES-GCM streaming:          no
   * AES-CCM:                    no
   * AES-CTR:                    no
   * AES-CFB:                    no
   * AES-OFB:                    no
   * AES-XTS:                    no
   * AES-XTS streaming:          no
   * AES-SIV:                    no
   * AES-EAX:                    no
   * AES Bitspliced:             no
   * AES Key Wrap:               no
   * ARIA:                       no
[... 217 lines cut ...]
./wolfssl/wolfcrypt/dilithium.h:546:12: note: in expansion of macro ‘DILITHIUM_MAX_PUB_KEY_SIZE’
  546 |     byte p[DILITHIUM_MAX_PUB_KEY_SIZE];
      |            ^~~~~~~~~~~~~~~~~~~~~~~~~~
./wolfssl/wolfcrypt/dilithium.h:484:39: error: ‘OQS_SIG_ml_dsa_87_ipd_length_secret_key’ undeclared here (not in a function); did you mean ‘OQS_SIG_dilithium_5_length_secret_key’?
  484 | #define DILITHIUM_LEVEL5_KEY_SIZE     OQS_SIG_ml_dsa_87_ipd_length_secret_key
      |                                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
./wolfssl/wolfcrypt/dilithium.h:484:39: note: in definition of macro ‘DILITHIUM_LEVEL5_KEY_SIZE’
  484 | #define DILITHIUM_LEVEL5_KEY_SIZE     OQS_SIG_ml_dsa_87_ipd_length_secret_key
      |                                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
./wolfssl/wolfcrypt/dilithium.h:547:12: note: in expansion of macro ‘DILITHIUM_MAX_KEY_SIZE’
  547 |     byte k[DILITHIUM_MAX_KEY_SIZE];
      |            ^~~~~~~~~~~~~~~~~~~~~~
  CC       tests/unit_test-hash.o
  CC       tests/unit_test-w64wrapper.o
  CC       tests/unit_test-srp.o
  CC       tests/unit_test-quic.o
  CC       examples/client/tests_unit_test-client.o
  CC       examples/server/tests_unit_test-server.o
make[2]: *** [Makefile:8224: tests/unit_test-api.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[2]: Leaving directory '/home/royan/wolfssl'
make[1]: *** [Makefile:8578: all-recursive] Error 1
make[1]: Leaving directory '/home/royan/wolfssl'
make: *** [Makefile:5159: all] Error 2
```

iyanmv, 2024-07-15:
Those instructions seem a bit outdated. Try to build the latest liboqs instead of version 0.8.0. In particular, repeat the steps to compile liboqs but replace the command `git checkout 0.8.0` with `git checkout 0.10.1`. I can confirm that wolfssl 5.7.2 works with that version of liboqs (although I'm using the shared library, not the static one).

kareem-wolfssl, 2024-07-16:
Hi @notyan ,

As iyanmv said, please retry your build with the latest liboqs version.  I will update our INSTALL instructions for liboqs.

closed 2025-12-19 by kareem-wolfssl