wolfSSL/wolfssl #7752: can I use wolfSSL without a CA cert?

4rtemis-4rrow, 2024-07-15:
### Version

latest, preferably

### Description

ok so, this isn't a bug oran issue, but I didn't know where else to ask, stack overflow is...... less than helpful

I am working on a program where having bullet proof security is not the top priority, the top priority would be package size, so, I want to skip any sort of CA loading

is that something that wolfSSL allows me to do, or should I look at a different library?

thanks in advance

dgarske, 2024-07-15:
Hi @4rtemis-4rrow ,

The CA certificate is used to verify the peer's certificate. Usually those are loaded with wolfSSL_CTX_load_verify or wolfSSL_CTX_load_verify_locations.

The easiest way is to disable the certification verification using something like: `wolfSSL_CTX_set_verify(ctx, WOLFSSL_VERIFY_NONE, NULL);`. This will not verify the peers certificate. The third argument is an optional callback function that could also be used to override a failure.

Can you tell us more about your project and use-case? Are you the client or server?

Thanks,
David Garske, wolfSSL

4rtemis-4rrow, 2024-07-15:
thanks for your response @dgarske 

it's a POC linux malware, I want to have TLS encryption on it's comms, I will be using the linux/tls.h header file for the encryption, as described in [this document](https://kernel.org/doc/Documentation/networking/tls.txt), but it's only issue is that it can't do a handshake, so I'm looking at the options I have for the handshake, considering I need it to be as lightweight as possible, I am contemplating either wolfSSL or implementing it manually

closed 2024-07-30 by 4rtemis-4rrow