wolfSSL/wolfssl #7756: [Bug]: The value of WOLFSSL_MAX_ALT_NAMES causes an error of ASN_ALT_NAME_E

ycdev1, 2024-07-17:
### Version

5.7.2

### Description

I updated to version 5.7.2 via vcpkg but I got ASN_ALT_NAME_E error while handshaking YouTube domain (www.youtube.com)
The error is generated at this line:
https://github.com/wolfSSL/wolfssl/blob/ec9fcf53539755ed3a38e386a00d91478ff357b3/wolfcrypt/src/asn.c#L18986
due to numNames reaching 129 which is larger than WOLFSSL_MAX_ALT_NAMES
I don't know if numNames reaching 129 is normal and the bug is from the new introduced WOLFSSL_MAX_ALT_NAMES (as per the release notes!) or there is something wrong with the received certificate? The same code is working with previous versions

embhorn, 2024-07-18:
Hi @ycdev1 

Thanks for reporting this issue. The default value is arbitrary, we were not aware of real-world use cases that employed more than a few alt names in a cert. We will review this and provide an update.

In the mean time, you can redefine `WOLFSSL_MAX_ALT_NAMES` to a larger value in the configuration as a workaround.

ColtonWilley, 2024-07-18:
Hi @ycdev1

I have put up a PR that should fix this issue for you here: #7762 

Please let me know if that works for you.

Thanks,
Colton

PR #7762 (2024-07-19): Increase default max alt names from 128 to 1024
# Description

Increase default max alt names from 128 to 1024
Fixes #7756
closed 2024-07-19 by dgarske