wolfSSL/wolfssl #7767: AES CTR wolfcrypt interface need to support keys with ENCRYPT usage when encrypting

DmitryOcheretyany, 2024-07-19:
### Version

5.7.0

### Description

The AES CTR interface to encrypt is run backwards to decrypt and use IV as input. This will prevent keys with usage set to ENCRYPT from encrypting data.

Now, AES-CTR encyption uses C_Decrypt, so it's impossible to use AES-CTR encryption for keys with encrypt key purpose. It's required to extend WolfSSL, so AES-CTR encryption can be performed using C_Encrypt, but without breaking/changing behavior of the existing code.

Way to Resolve:
 int wc_AesGetIV(Aes* aes, byte* iv) is added

embhorn, 2024-07-19:
Hi @DmitryOcheretyany 

Thanks for your interest in the wolfSSL project. It's not clear to me what you are requesting here. The AES-CTR APIs do use the same encryption key for both directions. Here is an example:
https://github.com/wolfSSL/wolfssl-examples/blob/master/crypto/aes/aesctr-file-encrypt.c

Could you modify that example to help show us the issue you are trying to resolve?

Thanks,
@embhorn - wolfSSL Support

DmitryOcheretyany, 2024-07-24:
Hi @embhorn,
Thank you for your response and for providing the example.
Our objective is to implement AES-CTR encryption in wolfSSL using C_Encrypt and IV generation while ensuring backward compatibility. Currently, AES-CTR encryption uses C_Decrypt, which prevents using encryption keys with the encrypt key purpose.
To address this, we propose the following changes:
1. Add the function `int wc_AesGetIV(Aes* aes, byte* iv)` to support IV retrieval.
2. Modify the AES-CTR encryption to use `C_Encrypt` with IV generation.
Here is a specific example of the added functionality:
```c
#ifdef WOLF_CRYPTO_CB
WOLFSSL_API int wc_AesGetIV(Aes* aes, byte* iv);
#endif
#ifdef PKCS11_NVIDIA_VENDOR
WOLFSSL_API int wc_AesSetKeyType(Aes* aes, int keyType);
#endif
```
We aim to implement these changes without affecting the existing behavior. Your guidance on modifying the provided example to demonstrate this issue would be greatly appreciated.
Best regards,  
Dmitry Ocheretyany

embhorn, 2024-07-24:
Hi @DmitryOcheretyany 

Thanks for sharing those details. I suggest sending an email to support@wolfssl.com to create a feature request. There we can have a more in depth conversation about the feature requirements.

Thanks,
@embhorn

embhorn, 2024-09-30:
Since we never heard back I will go ahead and close out this support inquiry. If any other questions come up please do not hesitate to open a new issue anytime by emailing [support@wolfssl.com](mailto:support@wolfssl.com) or through the zendesk portal at https://wolfssl.zendesk.com/. We are always happy to help out in any way we can.

Until next time all the best and thank you for using wolfSSL support.

closed 2024-09-30 by embhorn