wolfSSL/wolfssl #7778: Feature request: support for additional software

wellloaded, 2024-07-23:
### Version

1

### Description

As we are moving away from openssl for dimension reason (in freshtomato.org), we are currently stuck as there's no support yet for the following software:

- getdns
- mysql
- tor
- tinc
- vsftpd

Is there any hope these might be covered in the near feature?
Thanks

dgarske, 2024-07-23:
Hi @wellloaded , 

Can you tell us a bit more about your project and what's driving the change? I am not familiar with the "dimension reason" in freshtomoto. 

We have many open source projects maintained here -> https://github.com/wolfSSL/osp. 

Have you tried building these with our openssl compatibility layer enabled? It's likely these could work with very little changes to the code. We have over 1600 openssl compatibility API's.

Thanks, David Garske, wolfSSL

wellloaded, 2024-07-23:
Freshtomato is a firmware replacement project (www.freshtomato.org), it's an alternative to OpenWRT or similar although restricted to Broadcom routers.

You can check the latest commits (and what it was done for wolfSSL) at this URL:
https://bitbucket.org/pedro311/freshtomato-arm/commits/branch/arm-master

Thanks!

dgarske, 2024-07-24:
Hi @wellloaded ,

I'd like to escalate this request internally and assist your effort. We very much like what you are doing with FreshTomato and would like to support your efforts to move to wolfSSL.

Can you email support at wolfssl dot com and link to this issue? I'd like to get your email address so we could arrange a call to discuss collaboration.

We do have a mysql patch available. See: https://github.com/wolfSSL/mysql-patch

Thanks,
David Garske, wolfSSL