wolfSSL/wolfssl #7816: Should --enable-defaultdhparams place HAVE_DH_DEFAULT_PARAMS into options.h?

knight-of-ni, 2024-07-31:
### Version

5.7.0

### Description

I am building wolfssl rpm to support the Netatalk project, which recently started using this library.

Netatalk looks for HAVE_DH_DEFAULT_PARAMS, OPENSSL_EXTRA, and OPENSSL_ALL to be present as shown here:
https://github.com/Netatalk/netatalk/blob/main/meson.build#L560

Since the wolfssl rpm in [copr](https://copr.fedorainfracloud.org/coprs/beaveryoga/wolfSSL/) is not built with these parameters, I edited the [wolfssl specfile](https://download.copr.fedorainfracloud.org/results/beaveryoga/wolfSSL/fedora-40-x86_64/07218457-wolfssl42/wolfssl42-1.spec) and rebuilt with:

`%configure  --enable-distro --enable-pkcs11 --enable-defaultdhparams ---enable-opensslextra --enable-opensslall`

No other changes were made.
The build succeeds and I inserted the resulting rpms into my mock builds of Netatalk.

However, Netatalk does not find wolfssl because HAVE_DH_DEFAULT_PARAMS is not present in options.h.

I've gone through the built output and I don't see any signs the build rejected this option.
Interestingly, the [cmake template file](https://github.com/wolfSSL/wolfssl/blob/master/cmake/options.h.in) does not mention this option.

At the moment I'm not sure what kind of problem this is. 

Did I build this incorrectly? Is this an issue with WolfSSL? ...or should Netatalk check for HAVE_DH_DEFAULT_PARAMS be done differently?

anhu, 2024-07-31:
Hi @knight-of-ni , My name is Anthony and I am a member of the wolfSSL team.  I'll be helping you on this ticket.  Please stay tuned.

anhu, 2024-07-31:
As an initial observation, your configure commandline has `---enable-opensslextra ` (tripple dash would be unrecognized) .

knight-of-ni, 2024-07-31:
Sorry about that. That was a typo, which has now been fixed.

anhu, 2024-07-31:
Interestingly, by default, `HAVE_DH_DEFAULT_PARAMS` gets added. That is `./configure`  adds it into `options.h` but your command line does not add it. I suspect that `--enable-distro` then enables QT ENABLED_QT is then true. 

```
if test "x$ENABLED_DH" = "xyes" && test "x$ENABLED_DHDEFAULTPARAMS" = "xyes" && test "x$ENABLED_QT" != "xyes"
then
    ENABLED_DHDEFAULTPARAMS=yes
    AM_CFLAGS="$AM_CFLAGS -DHAVE_DH_DEFAULT_PARAMS"
fi
```

I'll need to figure out why QT disables it.  Please stay tuned. 

Warm regards, Anthony. 
```

knight-of-ni, 2024-07-31:
Ah good catch. While I did see in configure.ac that HAVE_DH_DEFAULT_PARAMS was dependent on QT being disabled, I did not realize the enable-distro flag enabled QT.

knight-of-ni, 2024-08-01:
After removing the --enable-distro flag from configure, I built a new set of packages on Copr. 
HAVE_DH_DEFAULT_PARAMS is now present in options.h. Thank you for your help.

closed 2024-08-01 by knight-of-ni