wolfSSL/wolfssl #7827: [Bug]: FIPS_VERSION3_GE not defined - error: missing binary operator before token

knight-of-ni, 2024-08-03:
### Contact Details

knight-of-ni

### Version

5.7.2

### Description

When building Netatalk 3.2.5 against wolfssl 5.7.2, the build fails with the error below because FIPS_VERSION3_GE macro is not defined. 

Note that Netatalk does not use FIPS_VERSION3_GE macro, or fips for that matter, which is why wolfssl was built w/o fips.

This appears to be the relevant line of code:
https://github.com/wolfSSL/wolfssl/blob/v5.7.2-stable/wolfssl/wolfcrypt/sha3.h#L39

...which was added with commit 81f5ac7f6c0c01c70ada7ab316109de3edcf59c2.

Note that Netatalk 3.2.5 builds fine against wolfssl 5.7.0, which does not have that commit.

### Reproduction steps

Build wolfssl in the usual manner like so:
```
configure \
            --enable-pkcs11 \
            --enable-defaultdhparams \
            --enable-opensslextra \
            --enable-opensslall \
            --enable-krb

```
Then build Netatalk against the wolfssl.

### Relevant log output

```shell
gcc -Ietc/uams/uams_dhx_passwd.so.p -Ietc/uams -I../etc/uams -I. -I.. -Iinclude -I../include -Ietc/afpd -I../etc/afpd -fdiagnostics-color=always -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -Wextra -Wpedantic -std=c11 -DHAVE_CONFIG_H '-D_U_=__attribute__((unused))' -Wno-pedantic -Wno-extra -Wno-all -Wno-deprecated-declarations -O2 -flto=auto -ffat-lto-objects -fexceptions -g -grecord-gcc-switches -pipe -Wall -Werror=format-security -Wp,-U_FORTIFY_SOURCE,-D_FORTIFY_SOURCE=3 -Wp,-D_GLIBCXX_ASSERTIONS -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -fstack-protector-strong -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1 -m64 -march=x86-64 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -fno-omit-frame-pointer -mno-omit-leaf-frame-pointer -fPIC -MD -MQ etc/uams/uams_dhx_passwd.so.p/uams_dhx_passwd.c.o -MF etc/uams/uams_dhx_passwd.so.p/uams_dhx_passwd.c.o.d -o etc/uams/uams_dhx_passwd.so.p/uams_dhx_passwd.c.o -c ../etc/uams/uams_dhx_passwd.c
In file included from ../include/wolfssl/wolfcrypt/hash.h:47,
                 from ../include/wolfssl/wolfcrypt/sp_int.h:34,
                 from ../include/wolfssl/wolfcrypt/wolfmath.h:48,
                 from ../include/wolfssl/openssl/bn.h:34,
                 from ../etc/uams/uams_dhx_passwd.c:33:
/usr/include/wolfssl/wolfcrypt/sha3.h:39:21: error: missing binary operator before token "("
   39 | #if FIPS_VERSION3_GE(6,0,0)
```

knight-of-ni, 2024-08-19:
I am going to try and move [this conditional](https://github.com/wolfSSL/wolfssl/blob/master/wolfssl/wolfcrypt/sha3.h#L39) up into the ifdef HAVE_FIPS block a [few lines up](https://github.com/wolfSSL/wolfssl/blob/master/wolfssl/wolfcrypt/sha3.h#L30).

Might not be until later this week before I can test it.

kareem-wolfssl, 2024-08-19:
Hi @knight-of-ni ,

Thanks for your patience on this issue and my apologies for the delay.  FIPS_VERSION3_GE is defined in settings.h, one common issue we see with applications building against wolfSSL is not including our build settings before including other headers.  Please confirm that your application is including the following before all other wolf/OpenSSL headers in all files which include these headers:
```
#ifndef WOLFSSL_USER_SETTINGS
    #include <wolfssl/options.h>
#endif
#include <wolfssl/wolfcrypt/settings.h>
```

knight-of-ni, 2024-08-19:
Ah, thank you. I'll work with the Netatalk project team and let you know.

knight-of-ni, 2024-08-23:
Netatalk was patched to include settings.h like so:
https://github.com/knight-of-ni/Netatalk/commit/847036e55604baff0bcb202ec5a6a5da99108cd7

Unfortunately, the same error persists:

```
In file included from ../include/wolfssl/wolfcrypt/hash.h:47,
                 from ../include/wolfssl/wolfcrypt/sp_int.h:34,
                 from ../include/wolfssl/wolfcrypt/wolfmath.h:48,
                 from ../include/wolfssl/openssl/bn.h:34,
                 from ../etc/uams/uams_dhx_passwd.c:34:
/usr/include/wolfssl/wolfcrypt/sha3.h:39:21: error: missing binary operator before token "("
   39 | #if FIPS_VERSION3_GE(6,0,0)
```

Here is the content of settings.h:
https://gist.githubusercontent.com/knight-of-ni/71754e39a638758ebd558d5e390401f6/raw/601687c3b852e308d1d0777fed4ebea8d4b4be9a/gistfile1.txt

Here is where the macro is defined in settings.h:
```
#define FIPS_VERSION3_GE(major,minor,patch) \
    (WOLFSSL_FIPS_VERSION_CODE >= WOLFSSL_MAKE_FIPS_VERSION3(major,minor,patch))
```

Since options.h does NOT define HAVE_FIPS, WOLFSSL_FIPS_VERSION_CODE should be set like so, a few lines prior:

```
#if !defined(HAVE_FIPS)
    #define WOLFSSL_FIPS_VERSION_CODE WOLFSSL_MAKE_FIPS_VERSION3(0,0,0)
    #define WOLFSSL_FIPS_VERSION2_CODE WOLFSSL_FIPS_VERSION_CODE
```

If I read settings.h right,  WOLFSSL_FIPS_VERSION_CODE is (0,0,0) in my case, but I'm not sure what the value of WOLFSSL_MAKE_FIPS_VERSION3 is.

knight-of-ni, 2024-08-25:
Providing another data point...

For troubleshooting purposes, I included wolfssl/wolfcrypt/settings.h directly into sha3.h like so then rebuilt wolfssl:
https://github.com/knight-of-ni/wolfssl/commit/c52b8e7174223a7616c12b5cae65e19fa01aea3b

I then built netatalk against the modified wolfssl package and received the same error:

```
[194/274] gcc -Ietc/uams/uams_dhx_pam.so.p -Ietc/uams -I../etc/uams -I. -I.. -Iinclude -I../include -Ietc/afpd -I../etc/afpd -fdiagnostics-color=always -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -Wextra -Wpedantic -std=c11 -DHAVE_CONFIG_H '-D_U_=__attribute__((unused))' -Wno-pedantic -Wno-extra -Wno-all -Wno-deprecated-declarations -O2 -flto=auto -ffat-lto-objects -fexceptions -g -grecord-gcc-switches -pipe -Wall -Werror=format-security -Wp,-U_FORTIFY_SOURCE,-D_FORTIFY_SOURCE=3 -Wp,-D_GLIBCXX_ASSERTIONS -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -fstack-protector-strong -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1 -m64 -march=x86-64 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -fno-omit-frame-pointer -mno-omit-leaf-frame-pointer -fPIC -MD -MQ etc/uams/uams_dhx_pam.so.p/uams_dhx_pam.c.o -MF etc/uams/uams_dhx_pam.so.p/uams_dhx_pam.c.o.d -o etc/uams/uams_dhx_pam.so.p/uams_dhx_pam.c.o -c ../etc/uams/uams_dhx_pam.c
FAILED: etc/uams/uams_dhx_pam.so.p/uams_dhx_pam.c.o 
gcc -Ietc/uams/uams_dhx_pam.so.p -Ietc/uams -I../etc/uams -I. -I.. -Iinclude -I../include -Ietc/afpd -I../etc/afpd -fdiagnostics-color=always -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -Wextra -Wpedantic -std=c11 -DHAVE_CONFIG_H '-D_U_=__attribute__((unused))' -Wno-pedantic -Wno-extra -Wno-all -Wno-deprecated-declarations -O2 -flto=auto -ffat-lto-objects -fexceptions -g -grecord-gcc-switches -pipe -Wall -Werror=format-security -Wp,-U_FORTIFY_SOURCE,-D_FORTIFY_SOURCE=3 -Wp,-D_GLIBCXX_ASSERTIONS -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -fstack-protector-strong -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1 -m64 -march=x86-64 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -fno-omit-frame-pointer -mno-omit-leaf-frame-pointer -fPIC -MD -MQ etc/uams/uams_dhx_pam.so.p/uams_dhx_pam.c.o -MF etc/uams/uams_dhx_pam.so.p/uams_dhx_pam.c.o.d -o etc/uams/uams_dhx_pam.so.p/uams_dhx_pam.c.o -c ../etc/uams/uams_dhx_pam.c
In file included from ../include/wolfssl/wolfcrypt/hash.h:47,
                 from ../include/wolfssl/wolfcrypt/sp_int.h:34,
                 from ../include/wolfssl/wolfcrypt/wolfmath.h:48,
                 from ../include/wolfssl/openssl/bn.h:34,
                 from ../etc/uams/uams_dhx_pam.c:34:
/usr/include/wolfssl/wolfcrypt/sha3.h:40:21: error: missing binary operator before token "("
   40 | #if FIPS_VERSION3_GE(6,0,0)
```

The build complains the macro isn't defined, even though I included settings.h just prior.

ColtonWilley, 2024-08-26:
@knight-of-ni 

I took a crack at it myself but was unable to reproduce, I think likely because I am not replicating your build. The wolfssl piece is clear, I was hoping you could elaborate with more details exactly how I would reproduce by "Then build Netatalk against wolfssl".

knight-of-ni, 2024-08-27:
Thank you @ColtonWilley . I appreciate the assist. 

An easy way to replicate this would be on Copr.

My results are here:
https://copr.fedorainfracloud.org/coprs/kni/wolfssl/builds/

Full build log here:
https://download.copr.fedorainfracloud.org/results/kni/wolfssl/fedora-rawhide-x86_64/07938805-netatalk/builder-live.log.gz

To duplicate in your own Copr repo, you can use the source rpm and specfile here:
https://download.copr.fedorainfracloud.org/results/kni/wolfssl/fedora-rawhide-x86_64/07938770-wolfssl/

After that build succeeds, build netatalk using the source rpm and specfile here:
https://download.copr.fedorainfracloud.org/results/kni/wolfssl/fedora-rawhide-x86_64/07938805-netatalk/

You can do the same thing from the local command line, using mock. It's a multi step process I can step you through, if you would rather do it this way.

If this isn't what you had in mind, tell me your preferred method and I can step you through it.

ColtonWilley, 2024-08-27:
@knight-of-ni 

I am not familiar with copr myself. I installed a Fedora VM and tried the commands under "You can reproduce this build on your computer by running" in the full build log, but hit errors on PGP signature verification failures on standard packages. 

Ideally I would like to reproduce this locally, as I agree this seems like it could be an odd bug. Once I am reproducing I can dig into root cause and figure this thing out. If you have any additional information on reproducing locally that would be very helpful.

Thanks,
Colton Willey, wolfSSL.

knight-of-ni, 2024-08-27:
Just a bit more information, for troubleshooting I duplicated the [relevant lines in settings.h](https://github.com/wolfSSL/wolfssl/blob/master/wolfssl/wolfcrypt/settings.h#L333-L386) directly in sha3.h. 

I then built wolfssl with this change. Netatalk build succeeded in this case. 

This would imply the compiler is not reaching [line 333](https://github.com/wolfSSL/wolfssl/blob/master/wolfssl/wolfcrypt/settings.h#L333) in settings.h when I try to include it the normal way e.g. `#include <wolfssl/wolfcrypt/settings.h>`.  This is certainly a weird one, that I can't seem to pin down.

In any case, here are instructions to locally build wolfssl and netatalk, using mock, in Fedora 40.

1. `sudo dnf install rpm-build mock mock-filessytem mock-core-configs`
2. `mkdir -p ~/rpmbuild/{BUILD,RPMS,SOURCES,SPECS,SRPMS}`
3. `echo '%_topdir %(echo $HOME)/rpmbuild' > ~/.rpmmacros`
4. add your user to the "mock" group then log out and back in
5. `cd ~/rpmbuild/SRPMS`
6. `wget https://download.copr.fedorainfracloud.org/results/kni/wolfssl/fedora-rawhide-x86_64/07938770-wolfssl/wolfssl-5.7.2-1.fc42.src.rpm`
7. `wget https://download.copr.fedorainfracloud.org/results/kni/wolfssl/fedora-rawhide-x86_64/07938805-netatalk/netatalk-3.2.5-1.fc42.src.rpm`
8. `mock wolfssl-5.7.2-1.fc42.src.rpm`
9. If successful then `cp /var/lib/mock/fedora-40-x86_64/result/*.rpm ~/rpmbuild/RPMS/`
10. `mock --install ~/rpmbuild/RPMS/x86_64/wolfssl-5.7.2-2.fc40.x86_64.rpm ~/rpmbuild/RPMS/x86_64/wolfssl-devel-5.7.2-2.fc40.x86_64.rpm`
11. `mock --no-clean ~/rpmbuild/SRPMS/netatalk-3.2.5-1.fc42.src.rpm`

The build log can be found in `/var/lib/mock/fedora-40-x86_64/result/build.log`.

I wrote this more or less off-the-cuff, so there may be minor mistakes. Just tell me what error(s) you hit, and we'll work through them.

Lastly, I know this might seem like a lot of steps, but doing it this way saves one from manually downloading all the netatalk dependencies (there are a lot) and then patching netatalk to include wolfssl/wolfcryptsettings.h... mock does all that for you.

Appreciate the help.

ColtonWilley, 2024-08-27:
Thank you very much for the detailed instructions. I was able to get to the last step `mock --no-clean ~/rpmbuild/SRPMS/netatalk-3.2.5-1.fc42.src.rpm` which failed and generated the expected failures in the log at `/var/lib/mock/fedora-40-x86_64/result/build.log`.

I hate to continue to ask for additional assistance, but I was wondering if you knew how I might change the wolfssl source code and retest using this environment, as I am not sure how the original .rpms are created.

knight-of-ni, 2024-08-27:
Yes, good question.

To make and test changes, first install the source rpm (as a normal user):
`rpm -ivh ~/rpmbuild/SRPMS/wolfssl-5.7.2-1.fc42.src.rpm`

This will put the rpm specfile under `~/rpmbuild/SPECS` and the tarball under `~/rpmbuild/SOURCES/`.

At this point you have a couple choices when making changes.

What I do is make a change in my fork of the project on github, then wget the patch:
`wget https://github.com/knight-of-ni/wolfssl/commit/{commit hash}`

After saving that to `~/rpmbuild/SOURCES`, I then rename it to something human friendly and insert the following line into the wolfssl.spec, just under Source0: `Patch0 wolfssl-mypatchfile.patch`

This is a bit inefficient, but github helps me visual the process.

You may find it easier to just directly modify the contents of `~/rpmbuild/SOURCES/wolfssl-5.7.2.tar.gz`. 

Once your changes are complete, build a new source rpm like so:
`rpmbuild -bs --nodeps ~/rpmbuild/SPECS/wolfssl.spec`

Then build it with the command from earlier:
```
mock ~/rpmbuild/SRPMS/wolfssl-5.7.2-1.fc42.src.rpm
cp /var/lib/mock/fedora-40-x86_64/result/*.rpm ~/rpmbuild/RPMS/
mock --install ~/rpmbuild/RPMS/x86_64/wolfssl-5.7.2-2.fc40.x86_64.rpm ~/rpmbuild/RPMS/x86_64/wolfssl-devel-5.7.2-2.fc40.x86_64.rpm
mock --no-clean ~/rpmbuild/SRPMS/netatalk-3.2.5-1.fc42.src.rpm
```

The `no-clean` option in the last step is important. That ensures mock uses the wolfssl rpm you just installed, rather than the wolfssl that now lives in Fedora repos.

Hope that helps.

ColtonWilley, 2024-08-28:
@knight-of-ni 

Thank you for the detailed instructions, it made debugging much easier. 

I believe the issue here lies with the wolfssl headers bundled with netatalk. On my setup, there are two settings.h files in the mock filesystem:
`/var/lib/mock/fedora-40-x86_64/root/include/usr/include/wolfssl/wolfcrypt/settings.h` and
`/var/lib/mock/fedora-40-x86_64/root/builddir/build/BUILD/netatalk-3.2.5/include/wolfssl/wolfcrypt/settings.h`

Extracting the netatalk rpm and inspecting the tar shows the same, that an older version of settings.h is bundled with netatalk. I believe this older settings.h in the BUILD dir, which does not define FIPS_VERSION3_GE, is the one included instead of the newer one. I tried to verify this by rebuilding the netatalk rpm using the newer settings.h but ran into some RPM build dependency issues. 

I believe this issue can be resolved by either bundling the newer headers instead, or bundling no headers and letting it pick up the installed ones. 

Please let me know if this solves your issues.

Thanks,
Colton Willey, wolfSSL.

knight-of-ni, 2024-08-28:
Ah yes, thank you. I needed another set of eyes to spot that. I believe that is in fact the root cause of the problem, but that has uncovered a new, [unrelated issue](https://github.com/Netatalk/netatalk/issues/1421) with Netatalk. 

I now need to work with the Netatalk team to resolve it. Please stand by.

knight-of-ni, 2024-08-29:
I have confirmed that this indeed was the problem. Netatalk builds just fine now, against an external wolfssl library, once the bundled wolfssl is removed just prior to build. Thank you very much for you help.

closed 2024-08-29 by knight-of-ni

ColtonWilley, 2024-08-30:
Glad to hear you issues were resolved successfully!