wolfSSL/wolfssl #7838: [Bug]:

rampageX, 2024-08-06:
### Contact Details

punkdm@gmail.com

### Version

5.7.2

### Description

cross-compile curl with wolfSSL on Alpine for armv5 device

### Reproduction steps

```
        ./configure \
          --host=${host} \
          CC=${host}-gcc \
          AR=${host}-ar \
          STRIP=${host}-strip \
          RANLIB=${host}-ranlib \
          --prefix=${install_dir} \
          CFLAGS="-march=armv5t -mfloat-abi=soft -msoft-float \
                -DHAVE_PK_CALLBACKS -DWOLFSSL_USER_IO -DNO_WRITEV -DTIME_T_NOT_64BIT" \
            --enable-quic --enable-session-ticket --enable-earlydata --enable-psk --enable-harden \
            --enable-altcertchains --enable-fastmath --enable-curl --enable-ip-alt-name --disable-shared
```

and cURL:

```
    CC=${host}-gcc CXX=${host}-g++ CPPFLAGS="${CPPFLAGS}" LDFLAGS="${LDFLAGS}" \
    LDFLAGS="--static -static -Wl,--no-as-needed -L${lib_dir}" LIBS="-lpthread" \
    LD_LIBRARY_PATH="-L${lib_dir}" PKG_CONFIG_PATH="${lib_dir}/pkgconfig" \
    ./configure \
    --build="x86_64-alpine-linux-musl" \
    --host="${host}" \
    --target="${host}" \
    --prefix="${install_dir}" \
    --enable-static --disable-shared \
    --with-wolfssl --with-ngtcp2 --with-brotli --with-zstd \
    --with-nghttp2 --with-nghttp3 \
    --with-libidn2 --with-libssh2 \
    --enable-hsts --enable-mime --enable-cookies \
    --enable-http-auth --enable-manual \
    --enable-proxy --enable-file --enable-http \
    --enable-ftp --enable-telnet --enable-tftp \
    --enable-pop3 --enable-imap --enable-smtp \
    --enable-gopher --enable-mqtt \
    --enable-doh --enable-dateparse --enable-verbose \
    --enable-alt-svc --enable-websockets \
    --enable-ipv6 --enable-unix-sockets --enable-socketpair \
    --enable-headers-api --enable-versioned-symbols \
    --enable-threaded-resolver --enable-optimize --enable-pthreads \
    --enable-warnings --enable-werror \
    --enable-curldebug --enable-dict --enable-netrc \
    --enable-bearer-auth --enable-tls-srp --enable-dnsshuffle \
    --enable-get-easy-options --enable-progress-meter \
    --with-ca-bundle=/etc/ssl/certs/ca-certificates.crt \
    --with-ca-path=/etc/ssl/certs \
    --enable-ares \
    --disable-ldap --disable-ldaps
```

### Relevant log output

```shell
When i run `/tmp/curl --alt-svc altsvc.cache https://curl.se/ -Iv`, i got:

* Alt-svc connecting from [h2]curl.se:443 to [h3]curl.se:443
* Host curl.se:443 was resolved.
* IPv6: (none)
* IPv4: 151.101.193.91, 151.101.129.91, 151.101.65.91, 151.101.1.91
*   Trying 151.101.193.91:443...
*  CAfile: /etc/ssl/certs/ca-certificates.crt
*  CApath: /etc/ssl/certs
* successfully set certificate verify locations:
*  CAfile: /etc/ssl/certs/ca-certificates.crt
*  CApath: /etc/ssl/certs
* QUIC connect to 151.101.193.91 port 443 failed: SSL peer certificate or SSH remote key was not OK
*   Trying 151.101.129.91:443...
*  CAfile: /etc/ssl/certs/ca-certificates.crt
*  CApath: /etc/ssl/certs
* successfully set certificate verify locations:
*  CAfile: /etc/ssl/certs/ca-certificates.crt
*  CApath: /etc/ssl/certs
* QUIC connect to 151.101.129.91 port 443 failed: SSL peer certificate or SSH remote key was not OK
*   Trying 151.101.65.91:443...
*  CAfile: /etc/ssl/certs/ca-certificates.crt
*  CApath: /etc/ssl/certs
* successfully set certificate verify locations:
*  CAfile: /etc/ssl/certs/ca-certificates.crt
*  CApath: /etc/ssl/certs
* QUIC connect to 151.101.65.91 port 443 failed: SSL peer certificate or SSH remote key was not OK
*   Trying 151.101.1.91:443...
*  CAfile: /etc/ssl/certs/ca-certificates.crt
*  CApath: /etc/ssl/certs
* successfully set certificate verify locations:
*  CAfile: /etc/ssl/certs/ca-certificates.crt
*  CApath: /etc/ssl/certs
* QUIC connect to 151.101.1.91 port 443 failed: SSL peer certificate or SSH remote key was not OK
* Failed to connect to curl.se port 443 after 1139 ms: SSL peer certificate or SSH remote key was not OK
* closing connection #0
curl: (60) Failed to connect to curl.se port 443 after 1139 ms: SSL peer certificate or SSH remote key was not OK
More details here: https://curl.se/docs/sslcerts.html

curl failed to verify the legitimacy of the server and therefore could not
establish a secure connection to it. To learn more about this situation and
how to fix it, please visit the webpage mentioned above.
```

or run `/tmp/curl -Iv https://curl.se`, i got:

```
* Host curl.se:443 was resolved.
* IPv6: (none)
* IPv4: 151.101.193.91, 151.101.129.91, 151.101.65.91, 151.101.1.91
*   Trying 151.101.193.91:443...
* Connected to curl.se (151.101.193.91) port 443
* ALPN: curl offers h2,http/1.1
* successfully set certificate verify locations:
*  CAfile: /etc/ssl/certs/ca-certificates.crt
*  CApath: /etc/ssl/certs
* SSL_connect failed with error -308: error state on socket
* closing connection #0
curl: (35) SSL_connect failed with error -308: error state on socket
```

On same device/system, build with openSSL one worked fine.
```

embhorn, 2024-08-06:
Hello @rampageX 

Thanks for this report. Could you please enable debug logging in wolfSSL and share the log? You'll need to add `--enable-debug` to the wolfSSL configure and add call to turn on logging to curl:

```
diff --git a/lib/vtls/wolfssl.c b/lib/vtls/wolfssl.c
index 2ef0af661..36b959694 100644
--- a/lib/vtls/wolfssl.c
+++ b/lib/vtls/wolfssl.c
@@ -1698,6 +1698,9 @@ static int wolfssl_init(void)
 {
   int ret;
 
+#ifdef DEBUG_WOLFSSL
+    wolfSSL_Debugging_ON();
+#endif
 #ifdef OPENSSL_EXTRA
   Curl_tls_keylog_open();
 #endif
```

rampageX, 2024-08-06:
```
/tmp/curl -V
wolfSSL Entering wolfSSL_Init
wolfSSL Entering wolfCrypt_Init
RNG_HEALTH_TEST_CHECK_SIZE = 128
sizeof(seedB_data)         = 128
opened /dev/urandom.
rnd read...
wolfSSL Entering wolfSSL_add_all_algorithms
curl 8.9.1 (arm-unknown-linux-musleabi) libcurl/8.9.1 wolfSSL/5.7.2 zlib/1.3.1 brotli/1.1.0 zstd/1.5.6 c-ares/1.32.3 libpsl/0.21.5 libssh2/1.11.0 nghttp2/1.62.1 ngtcp2/1.6.0 nghttp3/1.4.0
Release-Date: 2024-07-31
Protocols: dict file ftp ftps gopher gophers http https imap imaps ipfs ipns mqtt pop3 pop3s rtsp scp sftp smb smbs smtp smtps telnet tftp ws wss
Features: alt-svc AsynchDNS brotli HSTS HTTP2 HTTP3 IPv6 Largefile libz NTLM PSL SSL threadsafe TrackMemory UnixSockets zstd
wolfSSL Entering wolfSSL_Cleanup
wolfSSL Entering wolfCrypt_Cleanup
```

/tmp/curl -Iv https://curl.se 2> /tmp/cURL_wolfSSL_https.log
[cURL_wolfSSL_https.zip](https://github.com/user-attachments/files/16516639/cURL_wolfSSL_https.zip)

/tmp/curl --alt-svc altsvc.cache https://curl.se/ -Iv 2>/tmp/cURL_wolfSSL_http3.log
[cURL_wolfSSL_http3.zip](https://github.com/user-attachments/files/16516641/cURL_wolfSSL_http3.zip)

bagder, 2026-01-16:
curl 8.18.0 with wolfSSL 5.8.4 seems to work fine for me on Linux x86-64 for me (and is verified in curl's CI).

The h3 problem seems to be related to the certificate check, while the second, `SSL_connect failed with error -308: error state on socket` seems to be a `wolfSSL_connect()` returning error.

jackctj117, 2026-01-30:
@rampageX Your build configurations look very complicated, have they have ever worked?  If so, what changed?

anhu, 2026-01-30:
Hi, in your options.h you have  WOLFSSL_USER_IO.  Can you let me know how that got there?

anhu, 2026-01-31:
Also, can you let us know a bit about yourself?  What are your goals and how is it that you are doing this.  I have to admit that your configuration is quite unique and I would be curious to know how you came about to use it.