wolfSSL/wolfssl #7844: [Bug]: curl: (35) SSL_connect failed with error -140: ASN parsing error, invalid input

Zoey2936, 2024-08-06:
### Contact Details

zoey@z0ey.de

### Version

v5.7.2-stable

### Description

1. Build curl using this Dockerfile: https://github.com/ZoeyVid/curl-quic/blob/latest/Dockerfile
2. run `curl https://api.telegram.org`
3. get: `curl: (35) SSL_connect failed with error -140: ASN parsing error, invalid input` as error
4. rebuilt using v5.7.0 and everything works

embhorn, 2024-08-06:
Hi @Zoey2936 

Thanks for letting us know about this issue. Could you please test against the latest master and let us know if it works? I tried unsuccessfully to build your Docker file.

Thanks,
@embhorn

Zoey2936, 2024-08-07:
you can find a curl built with master as base here: https://github.com/ZoeyVid/curl-quic/actions/runs/10287471865 (the "artifacts" Artifact), and no building with master as base still does not work

tmael, 2024-08-07:
Thanks @Zoey2936 
I was able to reproduce the issue on both the master and v5.7.2-stable.
Seems like https://github.com/wolfSSL/wolfssl/pull/7625/files might have introduced the issue.

Zoey2936, 2024-08-07:
ok, until it is fixed I will use v5.7.0

Zoey2936, 2024-08-07:
```
PC-Zoey:~# echo | openssl s_client -connect api.telegram.org:443 -servername api.telegram.org 2>/dev/null | openssl x509 -noout -serial
serial=DF94C51B5A76E3CB
```
not sure...

Zoey2936, 2024-09-11:
so the current solution would be  CFLAGS="-DWOLFSSL_NO_ASN_STRICT"?

RealYukiSan, 2024-10-14:
I tried `CFLAGS="-DWOLFSSL_NO_ASN_STRICT" make && sudo make install`

but still encounter the same error, I don't use curl, and using wolfssl as library, the code is hosted on [gitlab repository](https://gitlab.com/realyukisan/d-net/-/tree/cbot?ref_type=heads)

RealYukiSan, 2024-10-14:
I tried a workaround mentioned by @Zoey2936 
switch to commit 8970ff4c34034dbb3594943d11f8c9d4c5512bd5 (where [5.7.0 released](https://github.com/wolfSSL/wolfssl/releases/tag/v5.7.0-stable)) and build from there:
```bash
# cleanup previous build
make clean
git ls-files -i -o --exclude-standard | xargs rm -rf
# setup, build and install
./autogen.sh
./configure --enable-sys-ca-certs
make
sudo make install
```
but I still got the `asn parsing error` message :(

RealYukiSan, 2024-10-14:
Woops, I made a mistake. I didn't aware that I've installed wolfssl (implicitly) from package manager, it might be linked to the library that provided by package manager instead of newly build one

RealYukiSan, 2024-10-20:
nevermind, using the suggestion mentioned by @Zoey2936 actually works, I just need to make sure that the appropriate compiled library indeed loaded, and `patchelf` come in handy to solve this.

btw, `CFLAGS="-DWOLFSSL_NO_ASN_STRICT"` didn't work.

Berthalamew, 2024-12-05:
Getting this as well, trying to upgrade from curl 5.7.0 to 5.7.4.

Zoey2936, 2025-04-16:
Issue still exists:
```
docker run --rm -it curl https://api.telegram.org
curl: (35) SSL_connect failed with error -140: ASN parsing error, invalid input
```
Dockerfile to build curl: https://github.com/ZoeyVid/curl-quic/blob/latest/Dockerfile, only change is that I removed -DWOLFSSL_NO_ASN_STRICT

Zoey2936, 2025-12-23:
When building with `-DWOLFSSL_NO_ASN_STRICT` it works, without it not:
```
/src/curl # /src/curl/src/curl -V
curl 8.17.0 (x86_64-pc-linux-musl) libcurl/8.17.0 wolfSSL/5.8.4 zlib/1.3.1 nghttp2/1.68.0 ngtcp2/1.19.0 nghttp3/1.14.0
Release-Date: [unreleased]
Protocols: dict file ftp ftps gopher gophers http https imap imaps ipfs ipns mqtt pop3 pop3s rtsp smb smbs smtp smtps telnet tftp ws wss
Features: alt-svc AsynchDNS ECH HSTS HTTP2 HTTP3 HTTPS-proxy HTTPSRR IPv6 Largefile libz NTLM SSL threadsafe UnixSockets
/src/curl # /src/curl/src/curl https://api.telegram.org
curl: (35) SSL_connect failed with error -140: ASN parsing error, invalid input
```

tmael, 2025-12-23:
@Zoey2936, Thanks for confirming that -DWOLFSSL_NO_ASN_STRICT works. We have a pull request that allows root CAs with serial number 0. Could you please give it a try? https://github.com/wolfSSL/wolfssl/pull/9567/

tmael, 2025-12-23:
Thanks, @Zoey2936. Excellent—thanks for confirming that #9567 fixes the issue. It will be included in the next wolfSSL release. Feel free to reopen this issue or create a new one if you run into any further problems. For now, I’m closing the issue.

closed 2025-12-23 by tmael

Zoey2936, 2026-04-10:
> Thanks, [@Zoey2936](https://github.com/Zoey2936). Excellent—thanks for confirming that [#9567](https://github.com/wolfSSL/wolfssl/pull/9567) fixes the issue. It will be included in the next wolfSSL [...]

do you know when it will be merged and released?