wolfSSL/wolfssl #7883: [Question] In TLS 1.2 client, VerifyMac fails, with bad record alert, after about 1-2 mins of good exchange

antonjfernando2021, 2024-08-18:
### Version

wolfssl 4.3.0

### Description

 Hi, 
 I have enabled these 
#define DEBUG_WOLFSSL
#define WOLFSSL_EXTRA_ALERTS
I suspect our custom socket layer as primary cause, but wireshark and wolfssl logs/prints are all I got initially.
Would printing sequence numbers, from keys structure, to see if records are out of order, or dropped etc, help? 
+    printf("(VerifyMac) peer_sequence_number_hi  0x%08x, peer_sequence_number_lo  0x%08x\n",
+    printf("(VerifyMac) sequence_number_hi 0x%08x, sequence_number_lo 0x%08x\n",
any suggestions on what else I could print/log on client side, to find what might be contributing to bad record?
Wireshark seems ok, no indication bad incoming tls packets. 
I know mine is an old release, I am curious if any verifymac related fixes went in near 4.3.0 release.
Thanks,

dgarske, 2024-08-19:
Hi @antonjfernando2021 ,

I agree this sounds like a legitimate issue with data integrity. Can you tell us more about your transport and the hardware you are using? If you'd like to keep this discussion private please open a support ticket using an email to support at wolfssl dot com and reference this issue.

Updating your release would be a good idea in general and is a good experiment, however I am not aware of anything that would improve a MAC issue.

Thanks,
David Garske, wolfSSL

dgarske, 2024-08-22:
Hi @antonjfernando2021 ,

How goes the investigation?

Thanks,
David Garske, wolfSSL

dgarske, 2024-11-15:
Hi @antonjfernando2021 ,

Since we did not hear back I will assume the issue has been resolved. Please let us know if you have any followup from this.

Thanks,
David Garske, wolfSSL

closed 2024-11-15 by dgarske