wolfSSL/wolfssl #7885: can't compile a project that uses wolfssl

4rtemis-4rrow, 2024-08-19:
### Version

latest

### Description

so, for context, I'm using arch linux, the project needs to be statically linked, so I chose to link it with musl, I installed the wolfssl package and the musl package

attempting to compile the project fails, a wolfssl header's file or directory is not found

I know this is an error in the way I'm trying to do this, but idk what the error is, do I need to manually compile wolfssl? or is there some compiler flag I'm missing?

thanks in advance

kareem-wolfssl, 2024-08-20:
Hi @4rtemis-4rrow ,

The wolfSSL Arch Linux package is linked against glibc, so you will need to download wolfSSL's source and build it against musl for it to work.  If you continue to run into errors after this, please share your wolfSSL ./configure line and a log of your build including any errors you're seeing.

4rtemis-4rrow, 2024-08-20:
thanks for responding @kareem-wolfssl 

I successfully built it from source using musl-gcc and the --enable-static flag to the config script, and now I successfully have a libwolfssl.a file in /usr/local/lib

as musl-gcc isn't configured to search in that path, I added the compiler flag -I/usr/local/lib

however, the build is still failing

```
⮚ ./Build --verbose
INFO: Generated PASSWORD: 6yaq~*3>Id2Fc2D,myc/izVSUaY9Riz*^fD[*7Srys//22CqM;
INFO: [!] Building Euryale
./Euryale/Euryale.c:20:10: fatal error: wolfssl/options.h: No such file or directory
   20 | #include <wolfssl/options.h>
      |          ^~~~~~~~~~~~~~~~~~~
compilation terminated.
ERROR: Command failed: musl-gcc -Wall -Wextra -D_XOPEN_SOURCE=500 -fomit-frame-pointer -fno-inline-functions -fno-ident -z -static execstack -Os -s -fvisibility=hidden -DPASSWORD="6yaq~*3>Id2Fc2D,myc/izVSUaY9Riz*^fD[*7Srys//22CqM;" -I/usr/local/lib -c ./Euryale/Euryale.c -o ./Euryale/Euryale.o
```

any and all help would be greatly appreciated, thank you

4rtemis-4rrow, 2024-08-20:
I am facing no errors when building wolfssl though, in either case, here are the logs

```
⮚ export CC=musl-gcc
⮚ ./configure --enable-static
checking for gcc... musl-gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables...
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether the compiler supports GNU C... yes
checking whether musl-gcc accepts -g... yes
checking for musl-gcc option to enable C11 features... none needed
checking whether musl-gcc understands -c and -o together... yes
checking build system type... x86_64-pc-linux-musl
checking host system type... x86_64-pc-linux-musl
checking target system type... x86_64-pc-linux-musl
checking for a BSD-compatible install... /usr/bin/install -c
checking whether sleep supports fractional seconds... yes
checking filesystem timestamp resolution... 0.01
checking whether build environment is sane... yes
checking for a race-free mkdir -p... /usr/bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports the include directive... yes (GNU style)
checking whether make supports nested variables... yes
checking xargs -n works... yes
checking whether UID '1000' is supported by ustar format... yes
checking whether GID '1000' is supported by ustar format... yes
checking how to create a ustar tar archive... gnutar
checking dependency style of musl-gcc... gcc3
checking how to print strings... printf
checking for a sed that does not truncate output... /usr/bin/sed
checking for grep that handles long lines and -e... /usr/bin/grep
checking for egrep... /usr/bin/grep -E
checking for fgrep... /usr/bin/grep -F
checking for ld used by musl-gcc... /usr/bin/ld
checking if the linker (/usr/bin/ld) is GNU ld... yes
checking for BSD- or MS-compatible name lister (nm)... /usr/bin/nm -B
checking the name lister (/usr/bin/nm -B) interface... BSD nm
checking whether ln -s works... yes
checking the maximum length of command line arguments... 1572864
checking how to convert x86_64-pc-linux-musl file names to x86_64-pc-linux-musl format... func_convert_file_noop
checking how to convert x86_64-pc-linux-musl file names to toolchain format... func_convert_file_noop
checking for /usr/bin/ld option to reload object files... -r
checking for file... file
checking for objdump... objdump
checking how to recognize dependent libraries... pass_all
checking for dlltool... no
checking how to associate runtime and link libraries... printf %s\n
checking for ar... ar
checking for archiver @FILE support... @
checking for strip... strip
checking for ranlib... ranlib
checking command to parse /usr/bin/nm -B output from musl-gcc object... ok
checking for sysroot... no
checking for a working dd... /usr/bin/dd
checking how to truncate binary pipes... /usr/bin/dd bs=4096 count=1
checking for mt... no
checking if : is a manifest tool... no
checking for stdio.h... yes
checking for stdlib.h... yes
checking for string.h... yes
checking for inttypes.h... yes
checking for stdint.h... yes
checking for strings.h... yes
checking for sys/stat.h... yes
checking for sys/types.h... yes
checking for unistd.h... yes
[... 524 lines cut ...]
libtool: finish: PATH="/usr/local/sbin:/usr/local/bin:/usr/bin:/sbin" ldconfig -n /usr/local/lib
----------------------------------------------------------------------
Libraries have been installed in:
   /usr/local/lib

If you ever happen to want to link against installed libraries
in a given directory, LIBDIR, you must either use libtool, and
specify the full pathname of the library, or use the '-LLIBDIR'
flag during linking and do at least one of the following:
   - add LIBDIR to the 'LD_LIBRARY_PATH' environment variable
     during execution
   - add LIBDIR to the 'LD_RUN_PATH' environment variable
     during linking
   - use the '-Wl,-rpath -Wl,LIBDIR' linker flag
   - have your system administrator add LIBDIR to '/etc/ld.so.conf'

See any operating system documentation about shared libraries for
more information, such as the ld(1) and ld.so(8) manual pages.
----------------------------------------------------------------------
make[3]: Leaving directory '/home/artemis-arrow/Downloads/wolf/wolfssl-5.7.2-stable'
make[2]: Leaving directory '/home/artemis-arrow/Downloads/wolf/wolfssl-5.7.2-stable'
make[1]: Leaving directory '/home/artemis-arrow/Downloads/wolf/wolfssl-5.7.2-stable'
⮚ ls /usr/local/lib
 pkgconfig   libwolfssl.a   libwolfssl.la   libwolfssl.so   libwolfssl.so.42   libwolfssl.so.42.2.0
```

4rtemis-4rrow, 2024-08-20:
forget that, I'm an idiot, made a very dumb mistake

closed 2024-08-20 by 4rtemis-4rrow