wolfSSL/wolfssl #7900: Feature Request: Allow test suite to run w/o internet access

knight-of-ni, 2024-08-25:
### Version

5.7.2

### Description

For wolfssl rpm in fedora, I have to modify the test suite like so, in order to allow it to succeed in a build environment with no access to the outside world:

```
# Disable tests that need Internets
sed -i 's/^if BUILD_OCSP$/if FALSE/' scripts/include.am
sed -i 's/^if BUILD_OCSP_STAPLING$/if FALSE/' scripts/include.am

```

It would be nice if there was an autotools flag I could set (e.g. --disable-online-checks) to do this for me.

mpsuzuki, 2024-08-28:
I'm interested in this issue. Should I draft something for the discussion material?

kareem-wolfssl, 2024-09-03:
Hi knight-of-ni,

Thanks for your patience on this.  I've assigned this issue to my colleague who will work on getting this implemented.

kareem-wolfssl, 2025-12-19:
Fixed in https://github.com/wolfSSL/wolfssl/pull/7945.

closed 2025-12-19 by kareem-wolfssl