wolfSSL/wolfssl #7905: how to disable single precision features?

mpsuzuki, 2024-08-27:
### Version

5.7.2

### Description

Now I'm trying to test wolfssl with ancient macOS 10.4 running on PowerPC G4.

```
  CC       wolfcrypt/src/src_libwolfssl_la-sp_int.lo
sp_int.s:3564:Parameter syntax error (parameter 1)
sp_int.s:3565:Parameter syntax error (parameter 1)
sp_int.s:3566:Parameter syntax error (parameter 3)
sp_int.s:3567:Parameter syntax error (parameter 3)
sp_int.s:5883:Parameter syntax error (parameter 1)
sp_int.s:5884:Parameter syntax error (parameter 2)
sp_int.s:6950:Parameter syntax error (parameter 1)
sp_int.s:6951:Parameter syntax error (parameter 2)
...
sp_int.s:26006:Parameter syntax error (parameter 3)
sp_int.s:26083:Parameter syntax error (parameter 1)
sp_int.s:26084:Parameter syntax error (parameter 1)
sp_int.s:26085:Parameter syntax error (parameter 3)
sp_int.s:26086:Parameter syntax error (parameter 3)
make[2]: *** [wolfcrypt/src/src_libwolfssl_la-sp_int.lo] Error 1
make[1]: *** [all-recursive] Error 1
make: *** [all] Error 2
```

Yet I have not confirmed whether this is a bug of the code generator
of GCC, or, a bug of the assembler (the last version of GCC provided
by MacPorts for PowerPC is gcc-7.5.0, I got same errors, so I guess the
latter). Anyway, as an easiest workaround for the first step, I want to
disable the single precision features.

The configure with all disable for SP-related features, like,
```./configure --disable-sp --disable-sp-math-all --disable-sp-math --disable-sp-asm```,
the compilation fails by the missing type definition of ```mp_int```.

```
make  all-recursive
  CC       wolfcrypt/benchmark/benchmark.o
In file included from wolfcrypt/benchmark/benchmark.c:77:
./wolfssl/wolfcrypt/wolfmath.h:89: error: parse error before '*' token
cc1: warnings being treated as errors
./wolfssl/wolfcrypt/wolfmath.h:90: warning: function declaration isn't a prototype
In file included from ./wolfssl/wolfcrypt/asn.h:50,
                 from wolfcrypt/benchmark/benchmark.c:81:
./wolfssl/wolfcrypt/dh.h:65: error: parse error before 'mp_int'
./wolfssl/wolfcrypt/dh.h:65: warning: no semicolon at end of struct or union
./wolfssl/wolfcrypt/dh.h:78: error: parse error before '}' token
...
```

This failure is not specific to the ancient macOS. If I test with same configuration
on GNU/Linux on x86_64 machine, the result is same (missing definition of ```mp_int```).

How can I disable SP-related features?

embhorn, 2024-08-27:
Hi @mpsuzuki 

You'll still need to enable an alternative math component. Try enabling TFM with `--enable-fastmath`

Let us know if that helps.

Thanks,
@embhorn - wolfSSL Support

mpsuzuki, 2024-08-27:
Thanks! ```./configure --disable-sp --disable-asm --enable-fastmath``` makes working binary, at least on GNU/Linux:
```
PASS: scripts/resume.test
PASS: scripts/google.test
PASS: scripts/tls13.test
PASS: scripts/pem.test
PASS: testsuite/testsuite.test
PASS: scripts/external.test
PASS: scripts/openssl.test
PASS: scripts/unit.test
```
Also, manual execution of scripts/ocsp.test passed either.

Unfortunately, I found a few other issues on macOS/ppc (specific to old GCC) and building is not finished yet. When I find other issues, like, unexpected dependency with SP module, I would reopen this, but now I can close this issue.

Again thank you very much for prompt help!

closed 2024-08-27 by mpsuzuki