wolfSSL/wolfssl #7910: POSIX requests the "sleep" command to accept an integer, but not a floating number.

mpsuzuki, 2024-08-28:
### Version

5.7.2

### Description

According to [POSIX specification](https://pubs.opengroup.org/onlinepubs/9699919799/utilities/sleep.html), the "sleep" command is not required to support non-integer number, like, "sleep 0.1". For example, [HP-UX 11 manual](https://man.freebsd.org/cgi/man.cgi?query=sleep&apropos=0&sektion=0&manpath=HP-UX+11.22&arch=default&format=html) says a non-integer argument causes an error. But, current testing scripts use "sleep 0.1" in a few cases (scripts/openssl.test, scripts/resume.test). I encountered this issue on ancient macOS X 10.4.

It is not so difficult to write an user program to imitate modern "sleep", like, [this](https://github.com/mpsuzuki/wolfssl-patch-dev/blob/fix-posix-sleep/tests/xsleep.c). Using Perl or Python might be another considerable option too.

However, yet I'm unsure about the appropriate directory to put such program in wolfssl, and how to set the environmental variables for testing. How can I learn the machanism?

embhorn, 2024-08-28:
Hi @mpsuzuki 

The test scripts are not required to use the library, and in fact, many embedded systems would never even use them. They are there for testing the wolfSSL code on modern systems.

Could you please clarify what you mean by this:
> However, yet I'm unsure about the appropriate directory to put such program in wolfssl, and how to set the environmental variables for testing. How can I learn the machanism?

Thanks,
@embhorn - wolfSSL Support

mpsuzuki, 2024-08-28:
Ah, sorry for my poor description.

- I want to use pre-installed ```/bin/sleep``` as far as it supports a floating number, to minimize the degrading trouble.
- I want to use the substitution program ("xsleep.c" in my example), if ```/bin/sleep``` does not supports a floating number.
- It is possible for configure (or CMake) to check whether ```/bin/sleep``` supports a floating number.
- But I don't know how to reflect the checking result in configure to the testing scripts.

Looking at the internal of the testing scripts, some scripts refer the environmental variables like WOLFSSL_EXTERNAL_TEST to control the behaviors of the testing scripts. But I could not find how these variables are configured. I want to understand its mechanism.

This explanation is better than my original comment?

embhorn, 2024-08-28:
The scripts use `WOLFSSL_EXTERNAL_TEST` and `WOLFSSL_OPENSSL_TEST` when those as environment variables in the system running the test.

For example, our CI tests set `export WOLFSSL_EXTERNAL_TEST=1` before running `make check`

For the sleep issue, I'll check with the team tomorrow to see if there are suggestions.

mpsuzuki, 2024-08-29:
I found that I overlooked the automake's ```AM_SANITY_CHECK``` macro checks the sleep command by ```_AM_SLEEP_FRACTIONAL_SECONDS```, and try ```sleep 0.001```, so, current configure script does such check sleep command already.

However, this existing check seems to be for the timestamp checking on the filesystem, there is no substitution program supporting fractional seconds.

bandi13, 2024-08-29:
A simple trick that you could do is set `PATH` to have your version of `sleep` higher in the list. Then the scripts should inherit that and run your `xsleep.c` instead of the system installed `sleep`.