wolfSSL/wolfssl #7922: Build options

parsahatami98, 2024-08-29:
Hello there
I am trying to build wolfssl to use with libz-ng , nginx ,wolfclu, xtls/xray-core , with tls1.2-1.3 also with dtls1.2-1.3 and post-quantum compability options in a debian based linux x86_64. What feature should i enable while building? I tried many times with no error in make check. But failing while installing wolfclu and nginx build after patching.
Please help and guide me
Thank you

parsahatami98, 2024-08-31:
[image]

Xray core is a proxy to bypass China's GFW and also useful in Iran and Russia. If it is possible, please consider to add supporting it. With your libs i can assume it may be better than lightway core by Expressvpn.

anhu, 2024-09-03:
Hi @parsahatami98 , 

Please have a look at https://github.com/wolfSSL/wolfssl-nginx?tab=readme-ov-file#post-quantum-algorithms and see if that is helpful.  Let me know if you have further problems. 

Warm regards, Anthony

anhu, 2024-09-03:
By the way, can you please let us know about your project, goals and where you are located?

anhu, 2024-09-09:
Hi, Any further questions?

parsahatami98, 2024-09-09:
Hello
Yup i tried to build wolfssl with these options : all, experimental, kyber, liboqs, des3, oldnames, aescbc, aescbc-lenght-check, dtls3, dtls-frag-ch, dtlscid, hkdf, hkpe, srtp-kdf, sctp, x963kdf, libz path to zlib-ng, wolfclu, dual-alg-certs, smallcache, fastmatg, nginx, and harden and disabling oldtls
WolfClu showing error not finding wolfssl_P521_SABER_level5 , also not finding Kyber_90s and NTRU_HRSS and NTRU_HPS
And tried to modifying nginx quic in github VirtuBox/nginx-ee and installing nginx mainline and patching wolfssl-nginx after patching dynamic tls records in github kn007 , all unsuccessful and showed errors

parsahatami98, 2024-09-09:
Project : making a vpn  for self use 
Goal : using other than usual libraries using Xtls/x-ray core that may work with openssl or as a reverse proxy using nginx
Location : somewhere in IRAN

anhu, 2024-09-10:
Please open a ticket by sending a message to support@wolfssl.com .

closed 2024-09-10 by anhu