wolfSSL/wolfssl #7941: [Bug]: Rpmlint reports `crypto-policy-non-compliance-openssl /usr/lib64/libwolfssl.so.42.1.0 SSL_CTX_set_cipher_list` with Fedora packaging

topazus, 2024-09-04:
### Version

5.7.0

### Description

When packaging wolfssl on Fedora Linux, the rpmlint tool will report `wolfssl.x86_64: W: crypto-policy-non-compliance-openssl /usr/lib64/libwolfssl.so.42.1.0 SSL_CTX_set_cipher_list`, as the issue was raised on package review of wolfssl [1, 2]. Any ideas on this?

[1] https://bugzilla.redhat.com/show_bug.cgi?id=2302646#c11
[2] https://docs.fedoraproject.org/en-US/packaging-guidelines/CryptoPolicies/#_new_crypto_libraries

topazus, 2024-09-04:
@douzzer @dgarske

anhu, 2024-09-04:
Hi @topazus 

Let me look into this for you.  I'll check the links you referenced first.  Please stay tuned. 

Warm regards, Anthony

anhu, 2024-09-04:
@topazus 

I'm not sure I understand their objection.  Is there an action that needs to be taken?  The interaction is very confusing as there is back and forth about a security team that no longer exists in the same capacity. 

Warm regards, Anthony

topazus, 2024-09-04:
@decathorpe Fabio is proficient with Fedora crypto policies, would you mind sheding light on this?

cc @knight-of-ni

decathorpe, 2024-09-04:
Short summary (I'm not at my PC right now):

There is a system-wide security policy for cryptography in Fedora. It defines which ciphers / algorithms are allowed to be used, and which are not. There are policies for OpenSSL, GnuTLS, OpenJDK, etc. They're loaded at runtime by these libraries, and applications are not allowed to override this policy without approval.

The policies are developed here:
https://gitlab.com/redhat-crypto/fedora-crypto-policies

It sounds like WolfSSL doesn't have a mechanism to load such a policy at runtime, which is a problem for Fedora - new libraries that implement cryptography MUST follow the system-wide policy, unless granted an explicit exception.

anhu, 2024-09-04:
Interesting.  Thanks for the summary of the situation. Is there a procedure for getting the exception? 

Warm regards, Anthony

topazus, 2024-09-04:
Wolfssl package is disapproved and retired for now.
https://pagure.io/fesco/issue/3267#comment-929888

anhu, 2024-09-04:
@decathorpe , 

Are you saying we need to be able to parse files like this: https://gitlab.com/redhat-crypto/fedora-crypto-policies/-/blob/master/policies/DEFAULT.pol?ref_type=heads

....and then only allow usage of  the algorithms specified in there? 

Warm regards, Anthony

decathorpe, 2024-09-05:
Actual installed policy files are parsed from these .pol files and look a bit different, they're available in `/usr/share/crypto-policies/` (etc.) on an installed Fedora Linux system.

There are policies in different formats for different applications, I assume that one in a specific format for wolfSSL could be added here, too.

closed 2024-09-05 by anhu

topazus, 2024-09-05:
> @topazus , Can you please send a message to [support@wolfssl.com](mailto:support@wolfssl.com) referencing this github issue asking for a feature request?

Ok, will send the email.

knight-of-ni, 2024-09-08:
Here is a reference, showing how the [gnutls](https://github.com/gnutls/gnutls) project handles this request.

configure supports a `--with-system-priority-file flag`:
https://github.com/gnutls/gnutls/blob/master/configure.ac#L908

We then set the path to the appropirate policy file when we call configure in the rpm specfile:
https://src.fedoraproject.org/rpms/gnutls/blob/rawhide/f/gnutls.spec#_322

topazus, 2024-09-08:
> @topazus Would you please CC me on the email. Thank you.

Here is the related link of feature request. https://wolfssl.zendesk.com/hc/en-us/requests/18593
I may not have time on this in the coming few days. You can comment on the above feature request link. Thanks.

knight-of-ni, 2024-09-08:
> Here is the related link of feature request. https://wolfssl.zendesk.com/hc/en-us/requests/18593
Thanks @topazus. This link is requires a zendesk login that is associated with this request. I've created an account and emailed wolfssl support, requesting access to 18593.

> I may not have time on this in the coming few days. You can comment on the above feature request link. Thanks.
Understood. If I recall, you mentioned you were graduating in another thread. Congratulations on this important step in your life.