wolfSSL/wolfssl #7943: Riscv Qemu

shirinebadi, 2024-09-04:
### Version

latest

### Description

Hi.

I want to cross compile WolfSSL to Risc-v on Qemu. I use x64 architecture. I want to know if it is possible to do that and if there is any tutorials for it?

dgarske, 2024-09-04:
Hi @shirinebadi ,

Thanks for your interest in using wolfSSL. Have you seen our new `--enable-riscv-asm` speedup option? It provides significant speedups for symmetric algorithms like AES/SHA on RISC-V. You can see some of the recent pull requests here: https://github.com/wolfSSL/wolfssl/pulls?q=is%3Apr+is%3Aclosed+RISC-V
Can you tell us more about your project?

@SparkiDev will you share some of your tips for testing RISC-V on QEMU?

Thanks,
David Garske, wolfSSL

shirinebadi, 2024-09-04:
This command should work?
`./configure --enable-riscv-asm --enable-all`

I'm trying to integrating SSL with Keystone

SparkiDev, 2024-09-04:
Hi @shirinebadi,

I use the gcc cross-compiler.
To install on Ubuntu:
```
sudo apt install gcc-riscv64-linux-gnu
```

If your CPU you want to emulate has just the base standard and no extensions, the QEMU you have will work.
Otherwise, you will need to get the very latest.

The configuration and make lines:
```
autogen.sh
./configure --disable-shared --enable-all LDFLAGS=--static --host=riscv64 CC=riscv64-linux-gnu-gcc --enable-riscvasm
make
```
Note this will build a static application that has all the QEMU code built-in.

To run testing:
```
./wolfcrypt/test/testwolfcrypt
```

To run benchmarking:
```
./wolfcrypt/benchmark/benchmark
```

Give it a go and let know how it goes.

Sean

SparkiDev, 2024-10-31:
Hi @shirinebadi,

Have you had any success cross-compiling for RISC-V?

Thanks,
Sean

shirinebadi, 2024-10-31:
Hi @SparkiDev. Thanks for your complete explanation. My research requirements changes and we decided to use BoringSSL. Thanks.

closed 2024-10-31 by SparkiDev