wolfSSL/wolfssl #7969: Using Custom user_settings.h with wolfSSL in HomeKit after Version Update

AchimPieters, 2024-09-12:
### Version

5.7.2

### Description

Hi everyone and @gojimmypi,

I'm working on an ESP-IDF HomeKit project and using wolfSSL as a managed component to simplify version updates. I've added wolfSSL as a component by including it in components/homekit/idf_component.yml, which makes it easier to update just by adjusting the version number.

[image]

In the components/homekit/user_settings.h file, I’ve set the necessary configurations to ensure smooth compilation.
 
[image]

However, my challenge is maintaining these custom settings across version updates. After updating to a new version of wolfSSL (e.g., from 5.7.2 to 5.7.3), I'd like my user_settings.h file to be used by default, without requiring additional manual steps.

[image]

## Here’s a summary of what I've done so far:

wolfSSL is added as a managed component in components/homekit/idf_component.yml
Custom configurations are set in components/homekit/user_settings.h
wolfSSL is automatically downloaded as a managed component during the build process.

## Question:

What adjustments or changes can I make to ensure my custom user_settings.h file is used automatically after updating the wolfSSL version in idf_component.yml? In other words, how can I ensure the project works "out of the box" with custom configurations after a version update?

Any suggestions or insights on how to approach this would be greatly appreciated!

Thanks in advance.

Best regards,
Achim Pieters

gojimmypi, 2024-09-12:
Hi @AchimPieters

Thank you again for all your help in getting wolfSSL working well with your awesome [ESP32 Apple HomeKit examples](https://github.com/AchimPieters/esp32-homekit-demo/tree/main/examples). Not only is that a cool project, but your use-case has been helpful in the design of the [wolfSSL Managed Component](https://components.espressif.com/components/wolfssl/wolfssl).

> What adjustments or changes can I make to ensure my custom user_settings.h file is used automatically ?

That's an excellent question! The answer is that _to use the Managed Components_, a custom `user_settings.h` is not possible at this time. The `user_settings.h` is currently _part of the component_; If the [component manager](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-guides/tools/idf-component-manager.html) sees that any part of the component has changed, then the component needs to be converted to a non-managed component. This of course defeats the purpose and value of having a Managed Component in the first place.

In order to customize wolfSSL for a given project, the current plan is to use the Espressif method of having all settings defined via the `idy.py menuconfig`. This means having `sdkconfig.defaults` available with an appropriate `Kconfig` file as described in the [Project Configuration Documentation](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/kconfig.html). 

I suppose in theory, it _might_ be possible to have a cleverly-crafted `CMakeLists.txt` that points to a wolfSSL `user_settings.h` in a user-defined directory _outside_ of the component. I could see that as being both beneficial, but also potentially problematic.  I'll need to experiment a bit with that. The trick would be having a user settings both inside _and outside_ of the component. Perhaps some sort of "supplementary user settings file"? I don't know.

For now, please take a look at my current [reference template app](https://github.com/wolfSSL/wolfssl/tree/master/IDE/Espressif/ESP-IDF/examples/template). Introduced in https://github.com/wolfSSL/wolfssl/pull/7866 I have many new settings available in the respective [Kconfig file](https://github.com/wolfSSL/wolfssl/blob/master/IDE/Espressif/ESP-IDF/examples/template/components/wolfssl/Kconfig) that will allow use of `menuconfig` and the respective `sdkconfig.default` files. 

* Please let me know what settings you need that are not included in the template app ASAP.

I'm expecting a wolfSSL release in the relatively near future (and a Managed Component update shortly thereafter).  I'd like to add any relevant settings that the Apple HomeKit support might need. It would be great to have those included in the next release. 

Like your examples, I have many different ESP32 examples, not only [here](https://github.com/wolfSSL/wolfssl/tree/master/IDE/Espressif) in the wolfSSL repo, but also:

- [wolfssl-examples](https://github.com/wolfSSL/wolfssl-examples/tree/master/ESP32)  
- [wolfssh](https://github.com/wolfSSL/wolfssh/tree/master/ide/Espressif)
- [wolfssh-examples](https://github.com/wolfSSL/wolfssh-examples/tree/main/Espressif)
- [wolfmqtt](https://github.com/wolfSSL/wolfMQTT/tree/master/IDE/Espressif)
- [wolftpm](https://github.com/wolfSSL/wolfTPM/tree/master/IDE/Espressif)

I would also like to have a common `user_settings.h`, `CMakeLists.txt`, and `Kconfig` in the [components/wolfssl](https://github.com/wolfSSL/wolfssl/tree/master/IDE/Espressif/ESP-IDF/examples/template/components/wolfssl) directory used by all of the examples.

> how can I ensure the project works "out of the box" with custom configurations after a version update?

I think the best method is with the `sdkconfig.defaults` and respective support in the `user_settings.h` file as well as in `settings.h` (see also https://github.com/wolfSSL/wolfssl/pull/7953 and the related changes to `settings.h`). 

Note that the [template user_settings.h](https://github.com/wolfSSL/wolfssl/blob/master/IDE/Espressif/ESP-IDF/examples/template/components/wolfssl/include/user_settings.h) already has a  [specific Apple HomeKit Section](https://github.com/wolfSSL/wolfssl/blob/9e2a7b36537ce6bc713405f4f08f8f50843bca87/IDE/Espressif/ESP-IDF/examples/template/components/wolfssl/include/user_settings.h#L135) ready for any defaults across all HomeKit apps.

I'm interested in your feedback and suggestions. 

Cheers

AchimPieters, 2024-09-15:
Hi @gojimmypi 

Thank you for your detailed response. I’ve attached my current user_settings.h file. 

[user_settings.h.txt](https://github.com/user-attachments/files/17005121/user_settings.h.txt)

After going through your message, I agree that using a Kconfig file could indeed be the solution here. This way, you can set up the necessary configurations per project, outside of the wolfSSL managed component. This would allow for seamless updates to the latest version of wolfSSL without having to manually adjust everything.

One idea could be to include a sample Kconfig file as a template, which could be copied for each project and modified as needed. This would make it easier for users to customize settings while maintaining the benefits of the managed component.

Looking forward to hearing your thoughts on this approach!

Cheers

AchimPieters, 2024-09-27:
Hi @gojimmypi,

Have you made any progress on the above?

gojimmypi, 2024-09-27:
Hi @AchimPieters , no, not yet. Sorry for the delay.

I've been working on improving the esp-tls with wolfSSL in #7936. See also the related `settings.h` [changes](https://github.com/wolfSSL/wolfssl/pull/7936/files#diff-9efb2cca94d744438b9691461eb2bf010ecf7b2e9a6e24c8d6fd7872a663197a) there.

If you could take a look at my [template](https://github.com/gojimmypi/wolfssl/tree/ED25519_SHA2_fix/IDE/Espressif/ESP-IDF/examples/template/components/wolfssl) and/or [esp_http_client_example](https://github.com/gojimmypi/wolfssl/tree/ED25519_SHA2_fix/IDE/Espressif/ESP-IDF/examples/esp_http_client_example/components/wolfssl) and let me know which settings are missing, that would be very helpful.

I'm also taking any other suggestions, and even pull requests.

Otherwise I'll try to get to that as soon as I can. Thanks for your patience. I'd definitely like to have this update in the next release of wolfSSL that I suspect will be in the coming weeks.

M10tech, 2024-10-04:
About WolfSSL (WS) in ESP idf and managed components

Let me try to make an analogy with a pizza restaurant.

There are pizzas on the menu and pizzas where you select your own toppings.
A crossover also exists with a menu pizza with extra toppings.

Now, WS as a managed component is menu pizza.
We can only get it as a fixed item and with the template initiative, WS creates the menu.
If we want to mess with the toppings, we have to invade the kitchen.

If we use WS as a git component, we can build our pizza and choose all the toppings.
This gives us maximum flexibility. And the git component system gives us the possibility to
adjust WS or upstream components any way we want in our own fork.

The key problem I have as an inventor is the inflexibility of tinkering with the toppings in managed mode.
If there is a new version of WS, it will overwrite my settings, so I will avoid that , which is bad also.
And once I have created my perfect recipe, it will take a lot of effort to get it on the menu of the restaurant.

The other side is that the offering for copy-cats will be easier to use.

In my opinion, there are no copy-cats in the target group of WS.
There are inventors that integrate WS and then there are copy-cats that use these inventions.
Like the HomeKit stack Achim ported and where I have been able to contribute.
Those contributions worked because the component system, but would be terrible in the managed component style.
This is because the managend components are no longer git aware of their own, but become part of the main project.
That last bit makes source tracking difficult and bloats the project repository.

Now, to make it worse, the HomeKit stack needs to provide the settings for WS,
not the main project that uses HomeKit as a component.

So, what to do?

I would want the settings of WS to be taken from a location set in the component that includes it in the dependency file. And maybe also, a layered approach where every higher layer can add a few details to the settings.

How to do that, I do not know yet, but I think the current approach is not appealing to inventors...

And one more thing: it took me an hour to make sure that what was in the managed component was actually derived 1:1 from the official WS repository. With security in mind, we do not want a backdoor to be installed by Espressif without detection.

@gojimmypi, I hope this helps in your quest to make your great product even better!

My 2 cents,
M10

gojimmypi, 2024-10-04:
Hi @M10tech and thank you for your feedback!

Yes, you have some completely valid and accurate points.

> managed components are no longer git aware of their own, but become part of the main project. That last bit makes source tracking difficult and bloats the project repository.

This in particular is unfortunately very true. The Managed Components definitely have there place:  making it easy to add wolfSSL library to a project and get started easily. The unfortunate thing, as you pointed out, is that the code is completely disconnected from GitHub source control. In fact, it is even painful to make the simplest changes, as the component is then no longer "managed".

As for bloat, well, the installed files could be added to `.gitignore`, but otherwise I agree that the files add to project size. It does, however, completely contain everything in a simple, easy-to-use example.

> we do not want a backdoor to be installed by Espressif without detection

This seems unlikely, but then once "published" you are correct that there's currently no way to _prove_ the code is the same as-published. When the publishing is completed, it is all in a tgz package from my (the publisher's) perspective when uploaded. I suspect there's a way to have a hash to prove the downloaded library is the exact same as published. I'll need to do a bit more homework on this. This is an excellent suggestion, thanks for pointing it out.

This same problem exists of course, when [wolfSSL is copied to the HomeKit repo](https://github.com/AchimPieters/esp32-homekit-demo/tree/main/components/wolfssl). Actually, this is probably even worse from a security perspective.

In a commercial situation, we can provide guidance to ensure code integrity and reliability. 

> it took me an hour to make sure that what was in the managed component was actually derived 1:1 from the official WS repository

I am open to suggestions how to make this easier, in particular if I don't find a hash feature as described above for the Managed Component. I could perhaps create my own in-directory hashes at publish time and include the hash fingerprint in an official wolfSSL source.

> So, what to do?

Well, like many things: it depends. I would probably not use a Managed Component in a project actively in development. As you noted, the disconnect from GitHub and inability to contribute upstream is a pretty big disadvantage.

In the case of the Apple HomeKit project, @AchimPieters  _really_ wants to make things as _simple_ for new users as possible. I get it: that's important. I'm trying to discourage from having an _entire_ local copy of wolfSSL _in the project_. (mainly for all the same issues of source integrity, and contributing to upstream). If you think the Managed Component adds bloat, OMG, a copy of the entire repo is even worse! lol

Using a Managed Component seemed to be a good way to accomplish "_using wolfSSL but not having an entire copy_".

There's also of course the option of having a submodule, the way the Espressif [esp_wolfssl](https://github.com/espressif/esp-wolfssl) does. There's a place for that. I'm not particularly fond of the idea for ESP32 projects. It also adds a bit of complexity for the new user.

Me, well I use my development branch of wolfSSL when working on any project. A very small number of files are needed in the local project. See the [template example](https://github.com/wolfSSL/wolfssl/tree/master/IDE/Espressif/ESP-IDF/examples/template/components/wolfssl). 

All my projects are in a workspace directory:

```
workspace/wolfssl
workspace/myproject
```

The [CMakeLists.txt](https://github.com/wolfSSL/wolfssl/blob/master/IDE/Espressif/ESP-IDF/examples/template/components/wolfssl/CMakeLists.txt) can find wolfSSL in a parent directory, or have a specific location by setting `WOLFSSL_ROOT`. This is great for development, but admittedly a bit more complex for things like the Apple HomeKit.

For example, in the case of the [HomeKit LED](https://github.com/AchimPieters/esp32-homekit-demo/tree/main/examples/led), one could add a local wolfSSL component, and simply remove the wolfSSL from  [EXTRA_COMPONENT_DIRS](https://github.com/AchimPieters/esp32-homekit-demo/blob/0d1708a4aa349ed1df69d4f527ae2df6dc6baa50/examples/led/CMakeLists.txt#L34) from the project's `MakeLists.txt`.

> I would want the settings of WS to be taken from a location set in the component that includes it in the dependency file.

I believe having your own clone of wolfSSL and using the `WOLFSSL_ROOT` would accomplish this. Although the "dependency file" part would not work for the Managed Components. It would need to be a regular, non-managed component. I believe this is the only option for how the Espressif Managed Components work.

I know this all needs to be documented better. In fact, just this week I've been working on this.

> I hope this helps in your quest to make your great product even better!

Thank you! I sincerely appreciate your feedback. You have a well informed and accurate assessment of the pros and cons of the  Managed Components.

Do you have an interest in wolfSSL beyond the public HomeKit examples? Please reach out to support@wolfss.com if there are things you'd like to discuss for a private project.

I'm certainly open to any additional ideas and feedback you have. Thanks again for taking the time to describe your concerns.

M10tech, 2024-10-06:
Hi @gojimmypi ,

>As for bloat, well, the installed files could be added to .gitignore,

excellent idea but hard to get in place, right?

Just for brainstorming, without thought of feasibility or usefulness:
- make a branch or a separate repo fork with submodule of wolfssl in the wolf GitHub account
- put enough .gitignore to kill the bloat
- upload that branch in your .tgz including the .git stuff
- the git engine will detect any type of mismatches and the commit-id will be your fingerprint
- with a way to point to settings outside of that managed component we might be in business.

In answer of your other replies:
When I develop, I want the develop environment to be the same as the final delivery.
Setting a WOLFSSL_ROOT does not achieve that.
I use LCM4ESP32 from HomeACcessoryKid and really like that flow.
It has a beta mode for testing stuff before it hits the final users.
But like this I know that every loose end of the integration is covered.

gojimmypi, 2024-10-29:
Hi @AchimPieters - many Espressif updates where included in the latest [wolfSSL 5.7.4 Release](https://www.wolfssl.com/wolfssl-5-7-4-release/), however I did not have time to determine if all the of the desired Apple HomeKit settings can be configured with the Kconfig (menuconfig) settings, nor have I had a chance to publish that release of [wolfSSL as a Managed Component](https://components.espressif.com/components/wolfssl/wolfssl).

The good thing is that your HomeKit examples are all _functional_, but yes: I do think it would be better to have a Managed Component rather than an entire copy of wolfSSL in your repo.

@M10tech sorry for the delay in responding. There's a somewhat related topic [Program can't find WolfSSL headers in ESP-IDF](https://www.wolfssl.com/forums/topic2207-$2.html) on the wolfSSL forum that may be of interest. You have some good ideas; let me know what I can do to help.

> I want the develop environment to be the same as the final delivery. Setting a WOLFSSL_ROOT does not achieve that.

I'm curious why? I would think that specifically having a static version of wolfSSL (say, a clone checked out on a particular release tag), and shared between projects (or develop / release builds) would accomplish that goal. I must be missing something, so perhaps you could give an example? Thank you.

gojimmypi, 2025-03-02:
Hi @AchimPieters - have you had a chance to test the 5.7.4 release? Please let me know if there's anything outstanding for this issue.

I recently published another update: [wolfssl v5.7.6 Managed Component](https://components.espressif.com/components/wolfssl/wolfssl/versions/5.7.6?language=en).

Both of these recent versions are working well and include many new user settings options that can be selected when using the command-line `idf.py menuconfig`.

Options can be found at `(top) -> Component config -> wolfSSL`:

[image]

Reminders: 

- Erase: `idf.py erase-flash -b 115200 -p /dev/ttyS44`
- Set WiFi SSID and password before flash
- Program: ` idf.py flash -p /dev/ttyS44 -b 115200 monitor -b 115200`
- Add dee manually
- the default Apple Homekit code is `338-77-883` 

My serial monitor output today using an ESP32-C3:

```
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5820,len:0x1738
load:0x403cc710,len:0xb9c
load:0x403ce710,len:0x2e40
entry 0x403cc71a
I (30) boot: ESP-IDF 5.2.1.2407012 2nd stage bootloader
I (30) boot: compile time Mar  2 2025 09:32:32
I (30) boot: chip revision: v0.3
I (34) boot.esp32c3: SPI Speed      : 80MHz
I (39) boot.esp32c3: SPI Mode       : DIO
I (43) boot.esp32c3: SPI Flash Size : 2MB
I (48) boot: Enabling RNG early entropy source...
I (53) boot: Partition Table:
I (57) boot: ## Label            Usage          Type ST Offset   Length
I (64) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (72) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (79) boot:  2 factory          factory app      00 00 00010000 00100000
I (87) boot: End of partition table
I (91) esp_image: segment 0: paddr=00010020 vaddr=3c0b0020 size=2be00h (179712) map
I (127) esp_image: segment 1: paddr=0003be28 vaddr=3fc8f400 size=03000h ( 12288) load
I (130) esp_image: segment 2: paddr=0003ee30 vaddr=40380000 size=011e8h (  4584) load
I (134) esp_image: segment 3: paddr=00040020 vaddr=42000020 size=af4e4h (718052) map
I (254) esp_image: segment 4: paddr=000ef50c vaddr=403811e8 size=0e01ch ( 57372) load
I (270) boot: Loaded app from partition at offset 0x10000
I (270) boot: Disabling RNG early entropy source...
>>> HomeKit: Starting server
>>> HomeKit: Formatting HomeKit storage
>>> HomeKit: Generated new accessory ID: 03:D8:1A:74:00:84
>>> HomeKit: Generated new accessory key
>>> HomeKit: Configuring mDNS
>>> homekit_setup_mdns: Accessory Setup ID = 1QJ8
>>> homekit_run_server: Starting HTTP server
>>> homekit_server_accept_client: Free heap: 218352
>>> HomeKit: [Client 1] Got new client connection from 192.168.1.122
>>> homekit_client_process: [Client 1] Got 132 incoming data
>>> homekit_server_on_pair_setup: Pair Setup
>>> homekit_server_on_pair_setup: Free heap: 220088
>>> tlv_debug: Got following TLV values:
>>> tlv_debug: Type 0 value (1 bytes): \x00
>>> tlv_debug: Type 6 value (1 bytes): \x01
>>> HomeKit: [Client 1] Pair Setup Step 1/3
>>> homekit_server_on_pair_setup: Free heap: 219988
>>> crypto_srp_new: Initializing SRP
>>> homekit_server_on_pair_setup: [Client 1] Initializing crypto
>>> homekit_server_on_pair_setup: Free heap: 211044
[... 178 lines cut ...]
>>> homekit_server_process_notifications: Free heap: 220108
>>> client_sendv: [Client 1] Sending payload: EVENT/1.0 200 OK\x0D\x0AContent-Type: application/hap+json\x0D\x0ATransfer-Encoding: chunked\x0D\x0A\x0D\x0A
>>> client_sendv: [Client 1] Sending payload: 35\x0D\x0A{"characteristics":[{"aid":1,"iid":10,"value":true}]}\x0D\x0A
>>> homekit_client_process: [Client 1] Got 195 incoming data0D\x0A
>>> homekit_client_process: [Client 1] Decrypting data
>>> homekit_client_process: [Client 1] Decrypted 177 bytes, available 0
Decrypted data (177 bytes): "\x50\x55\x54\x20\x2F\x63\x68\x61\x72\x61\x63\x74\x65\x72\x69\x73\x74\x69\x63\x73\x20\x48\x54\x54\x50\x2F\x31\x2E\x31\x0D\x0A\x48\x6F\x73\x74\x3A\x20\x48\x6F\x6D\x65\x4B\x69\x74\x5C\x30\x33\x32\x4C\x45\x44\x2E\x5F\x68\x61\x70\x2E\x5F\x74\x63\x70\x2E\x6C\x6F\x63\x61\x6C\x0D\x0A\x43\x6F\x6E\x74\x65\x6E\x74\x2D\x4C\x65\x6E\x67\x74\x68\x3A\x20\x35\x30\x0D\x0A\x43\x6F\x6E\x74\x65\x6E\x74\x2D\x54\x79\x70\x65\x3A\x20\x61\x70\x70\x6C\x69\x63\x61\x74\x69\x6F\x6E\x2F\x68\x61\x70\x2B\x6A\x73\x6F\x6E\x0D\x0A\x0D\x0A\x7B\x22\x63\x68\x61\x72\x61\x63\x74\x65\x72\x69\x73\x74\x69\x63\x73\x22\x3A\x5B\x7B\x22\x61\x69\x64\x22\x3A\x31\x2C\x22\x69\x69\x64\x22\x3A\x31\x30\x2C\x22\x76\x61\x6C\x75\x65\x22\x3A\x30\x7D\x5D\x7D"
>>> HomeKit: [Client 1] Update Characteristics
>>> homekit_server_on_update_characteristics: Free heap: 220132
>>> homekit_server_on_update_characteristics: [Client 1] Processing element {
        "aid":  1,
        "iid":  10,
        "value":        0
}
>>> HomeKit: [Client 1] Updating characteristic 1.10 ("On") with boolean false
>>> homekit_characteristic_notify: Got characteristic 1.10 change event
>>> homekit_server_on_update_characteristics: [Client 1] There were no processing errors, sending No Content response
>>> client_sendv: [Client 1] Sending payload: HTTP/1.1 204 No Content\x0D\x0A\x0D\x0A
>>> homekit_client_process: [Client 1] Finished processing
>>> homekit_server_process_notifications: [Client 1] Sending EVENT
>>> homekit_server_process_notifications: Free heap: 220096
>>> client_sendv: [Client 1] Sending payload: EVENT/1.0 200 OK\x0D\x0AContent-Type: application/hap+json\x0D\x0ATransfer-Encoding: chunked\x0D\x0A\x0D\x0A
>>> client_sendv: [Client 1] Sending payload: 36\x0D\x0A{"characteristics":[{"aid":1,"iid":10,"value":false}]}\x0D\x0A
>>> client_sendv: [Client 1] Sending payload: \x30\x0D\x0A\x0D\x0A
```

AchimPieters, 2025-03-09:
@gojimmypi Will test ASAP, will come back with the results.

AchimPieters, 2025-03-15:
@gojimmypi I have tested the new version of wolfs version: `5.7.6`

It looks like wolfSSL is working out of the box on most ESP platforms, including ESP32, ESP32C3, ESP32C5, ESP32C6, ESP62C61, ESP32S2, and ESP32S3. 🎉

However, there are some issues with:

ESP32C2
ESP32H2
ESP32P4

All of these are failing due to missing esp32/rom/aes.h, which seems to be unavailable in their respective ESP-IDF component directories. The error messages suggest that the missing header might need to be explicitly added to the INCLUDE_DIRS in idf_component_register within the CMakeLists.txt file.

For ESP32P4, there’s also an additional warning about an unexpected CONFIG_IDF_TARGET_NN value in user_settings.h. This might need further investigation to confirm if it's related to the unsupported target.

Possible next steps:

Verify if esp32/rom/aes.h is available for these chips in the ESP-IDF version being used.
Check if these targets require a different approach for hardware crypto acceleration.
Investigate the CONFIG_IDF_TARGET_NN issue on ESP32P4 and whether it's a misconfiguration or missing support.

ESP32C2 error:

```
In file included from /project/examples/led/managed_components/wolfssl__wolfssl/wolfssl/wolfcrypt/sha256.h:85,
                 from /project/examples/led/managed_components/wolfssl__wolfssl/wolfssl/wolfcrypt/random.h:109,
                 from /project/examples/led/managed_components/wolfssl__wolfssl/wolfssl/wolfcrypt/tfm.h:48,
                 from /project/examples/led/managed_components/wolfssl__wolfssl/wolfssl/wolfcrypt/wolfmath.h:44,
                 from /project/examples/led/managed_components/wolfssl__wolfssl/wolfssl/wolfcrypt/rsa.h:61,
                 from /project/examples/led/managed_components/wolfssl__wolfssl/wolfcrypt/src/asn.c:114:
/project/examples/led/managed_components/wolfssl__wolfssl/wolfssl/wolfcrypt/port/Espressif/esp32-crypt.h:723:18: fatal error: esp32/rom/aes.h: No such file or directory
  723 |         #include "esp32/rom/aes.h"
      |                  ^~~~~~~~~~~~~~~~~
compilation terminated.
ninja: build stopped: subcommand failed.
Missing "esp32/rom/aes.h" file name found in the following component(s): esp_rom(/opt/esp/idf/components/esp_rom/esp32/include/esp32/rom/aes.h). Maybe one of the components needs to add the missing header directory to INCLUDE_DIRS of idf_component_register call in CMakeLists.txt.
ninja failed with exit code 1, output of the command is in the /project/examples/led/build/log/idf_py_stderr_output_11179 and /project/examples/led/build/log/idf_py_stdout_output_11179
root@2a26bb907e69:/project/examples/led# 
```

ESP32H2 error:

```
In file included from /project/examples/led/managed_components/wolfssl__wolfssl/wolfcrypt/src/tfm.c:57:
/project/examples/led/managed_components/wolfssl__wolfssl/wolfssl/wolfcrypt/port/Espressif/esp32-crypt.h:723:18: fatal error: esp32/rom/aes.h: No such file or directory
  723 |         #include "esp32/rom/aes.h"
      |                  ^~~~~~~~~~~~~~~~~
compilation terminated.
[840/898] Building C object esp-idf/wo...ssl.dir/wolfcrypt/src/wc_encrypt.c.obj
ninja: build stopped: subcommand failed.
Missing "esp32/rom/aes.h" file name found in the following component(s): esp_rom(/opt/esp/idf/components/esp_rom/esp32/include/esp32/rom/aes.h). Maybe one of the components needs to add the missing header directory to INCLUDE_DIRS of idf_component_register call in CMakeLists.txt.
ninja failed with exit code 1, output of the command is in the /project/examples/led/build/log/idf_py_stderr_output_16992 and /project/examples/led/build/log/idf_py_stdout_output_16992
root@2a26bb907e69:/project/examples/led# 
```

ESP32P4 error:

```
In file included from /project/examples/led/managed_components/wolfssl__wolfssl/wolfssl/wolfcrypt/settings.h:333,
                 from /project/examples/led/managed_components/wolfssl__wolfssl/wolfcrypt/src/tfm.c:39:
/project/examples/led/managed_components/wolfssl__wolfssl/include/user_settings.h:794:6: warning: #warning "Unexpected CONFIG_IDF_TARGET_NN value" [-Wcpp]
  794 |     #warning "Unexpected CONFIG_IDF_TARGET_NN value"
      |      ^~~~~~~
In file included from /project/examples/led/managed_components/wolfssl__wolfssl/wolfcrypt/src/tfm.c:57:
/project/examples/led/managed_components/wolfssl__wolfssl/wolfssl/wolfcrypt/port/Espressif/esp32-crypt.h:723:18: fatal error: esp32/rom/aes.h: No such file or directory
  723 |         #include "esp32/rom/aes.h"
      |                  ^~~~~~~~~~~~~~~~~
compilation terminated.
[863/932] Building C object esp-idf/wo...fssl.dir/wolfcrypt/src/wc_pkcs11.c.obj
In file included from /project/examples/led/managed_components/wolfssl__wolfssl/wolfssl/wolfcrypt/settings.h:333,
                 from /project/examples/led/managed_components/wolfssl__wolfssl/wolfcrypt/src/wc_pkcs11.c:26:
[... 55 lines cut ...]
  794 |     #warning "Unexpected CONFIG_IDF_TARGET_NN value"
      |      ^~~~~~~
[873/932] Building C object esp-idf/wo...olfssl.dir/wolfcrypt/src/wc_port.c.obj
In file included from /project/examples/led/managed_components/wolfssl__wolfssl/wolfssl/wolfcrypt/settings.h:333,
                 from /project/examples/led/managed_components/wolfssl__wolfssl/wolfcrypt/src/wc_port.c:31:
/project/examples/led/managed_components/wolfssl__wolfssl/include/user_settings.h:794:6: warning: #warning "Unexpected CONFIG_IDF_TARGET_NN value" [-Wcpp]
  794 |     #warning "Unexpected CONFIG_IDF_TARGET_NN value"
      |      ^~~~~~~
[... cut ...]
Missing "esp32/rom/aes.h" file name found in the following component(s): esp_rom(/opt/esp/idf/components/esp_rom/esp32/include/esp32/rom/aes.h). Maybe one of the components needs to add the missing header directory to INCLUDE_DIRS of idf_component_register call in CMakeLists.txt.
ninja failed with exit code 1, output of the command is in the /project/examples/led/build/log/idf_py_stderr_output_23609 and /project/examples/led/build/log/idf_py_stdout_output_23609
root@2a26bb907e69:/project/examples/led# 
```

AchimPieters, 2025-03-22:
@gojimmypi I also made a Major change to the [ESP32-Homekit](https://github.com/AchimPieters/esp32-homekit) and [ESP](https://github.com/AchimPieters/esp32-homekit-demo) I made the ESP32-HomeKit available as Component at the [ESP Registry](https://components.espressif.com/components/achimpieters/esp32-homekit/versions/1.2.4)

gojimmypi, 2025-04-06:
Hi @AchimPieters 

> it looks like wolfSSL is working out of the box on most ESP platforms, including ESP32, ESP32C3, ESP32C5, ESP32C6, ESP62C61, ESP32S2, and ESP32S3. 

YAY!

> I also made a Major change to the ESP32-Homekit and ESP32-homekit-demo I made the ESP32-HomeKit available as Component at the ESP Registry

That's very cool! I might suggest not instructing the user to use docker. Most people will probably be using the [Official Espressif Instructions](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/get-started/windows-setup.html). (e.g. `export.bat`, etc.)

Otherwise that's really awesome to have the Apple Homekit as a Managed Cmoponent!

> However, there are some issues with: ESP32C2, ESP32H2, ESP32P4

I believe some of the issues have ben worked ut since the most recent release of the 5.7.6 [Managed Component](https://components.espressif.com/components/wolfssl/wolfssl).
 
Others were not, and some new polishing was needed; I've created https://github.com/wolfSSL/wolfssl/pull/8641 to address.

In any case, we'd really like to keep issues small and succinct. As such, since the original topic of _this_ PR regarding `user_settings.h` has been addressed, I will close this issue.

Please feel free to open any new issues. Thank you again your help in testing and the awesome integration into your Apple Homekit examples.