wolfSSL/wolfssl #7984: [Bug]: Unable to fully disable AES with preprocessor flags

rdmark, 2024-09-18:
### Version

5.7.2

### Description

I am trying to completely disable WolfSSL AES, to avoid conflicts with AES symbols in libnettle. This is when using WolfSSL as an embedded library in the Netatalk codebase.

I'm trying to disable AES in WolfSSL with these flags:

```
#define NO_AES
#define NO_AES_CBC
```

However, this particular macro in `evp.h` is still getting compiled, seemingly because both `HAVE_AESGCM` and `HAVE_AESCCM` are still resolving to non-zero.

```
#if defined(HAVE_AESGCM) || defined(HAVE_AESCCM) || defined(HAVE_ARIA)
    ALIGN16 unsigned char authTag[AES_BLOCK_SIZE];
```

https://github.com/wolfSSL/wolfssl/blob/c3900470aaff38f46fca225aec7b572c5ecda02d/wolfssl/openssl/evp.h#L516

### Reproduction steps

1. Check out this Netatalk git branch 
https://github.com/Netatalk/netatalk/tree/1430-nettle-and-wolfssl-both-define-aes_max_key_size
2. Follow steps in https://netatalk.io/4.0/htmldocs/compile to setup and compile the code
3. Observe the compile log

### Relevant log output

```shell
In file included from ../include/wolfssl/openssl/ssl.h:42:
../include/wolfssl/openssl/evp.h:517:35: error: ‘AES_BLOCK_SIZE’ undeclared here (not in a function); did you mean ‘DES_BLOCK_SIZE’?
  517 |     ALIGN16 unsigned char authTag[AES_BLOCK_SIZE];
      |                                   ^~~~~~~~~~~~~~
      |                                   DES_BLOCK_SIZE
```
```

rdmark, 2024-09-18:
The same problem happens in `cryptocb.h` here, which leads to compile errors with missing types defined in the AES headers.

https://github.com/wolfSSL/wolfssl/blob/c3900470aaff38f46fca225aec7b572c5ecda02d/wolfssl/wolfcrypt/cryptocb.h#L290

```
[205/255] Compiling C object etc/uams/uams_dhx_pam.so.p/uams_dhx_pam.c.o
FAILED: etc/uams/uams_dhx_pam.so.p/uams_dhx_pam.c.o 
cc -Ietc/uams/uams_dhx_pam.so.p -Ietc/uams -I../etc/uams -I. -I.. -Iinclude -I../include -Isys -I../sys -Ietc/afpd -I../etc/afpd -fdiagnostics-color=always -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -Wextra -Wpedantic -std=c11 -O0 -g -DHAVE_CONFIG_H '-D_U_=__attribute__((unused))' -Wno-pedantic -Wno-extra -Wno-all -Wno-deprecated-declarations -D_GNU_SOURCE -fPIC -MD -MQ etc/uams/uams_dhx_pam.so.p/uams_dhx_pam.c.o -MF etc/uams/uams_dhx_pam.so.p/uams_dhx_pam.c.o.d -o etc/uams/uams_dhx_pam.so.p/uams_dhx_pam.c.o -c ../etc/uams/uams_dhx_pam.c
In file included from ../include/wolfssl/ssl.h:50,
                 from ../include/wolfssl/openssl/ssl.h:37,
                 from ../etc/uams/uams_dhx_pam.c:35:
../include/wolfssl/wolfcrypt/cryptocb.h:289:17: error: unknown type name ‘Aes’
  289 |                 Aes*        aes;
      |                 ^~~
../include/wolfssl/wolfcrypt/cryptocb.h:301:17: error: unknown type name ‘Aes’
  301 |                 Aes*        aes;
      |                 ^~~
../include/wolfssl/wolfcrypt/cryptocb.h:315:17: error: unknown type name ‘Aes’
  315 |                 Aes*        aes;
      |                 ^~~
../include/wolfssl/wolfcrypt/cryptocb.h:327:17: error: unknown type name ‘Aes’
  327 |                 Aes*        aes;
      |                 ^~~
../include/wolfssl/wolfcrypt/cryptocb.h:349:17: error: unknown type name ‘Aes’
  349 |                 Aes*        aes;
      |                 ^~~
../include/wolfssl/wolfcrypt/cryptocb.h:357:17: error: unknown type name ‘Aes’
  357 |                 Aes*        aes;
      |                 ^~~
../include/wolfssl/wolfcrypt/cryptocb.h:434:9: error: unknown type name ‘Cmac’
  434 |         Cmac* cmac;
      |         ^~~~
../include/wolfssl/wolfcrypt/cryptocb.h:644:36: error: unknown type name ‘Cmac’; did you mean ‘Hmac’?
  644 | WOLFSSL_LOCAL int wc_CryptoCb_Cmac(Cmac* cmac, const byte* key, word32 keySz,
      |                                    ^~~~
      |                                    Hmac
```

rdmark, 2024-09-18:
Yet another problem I'm running into is that `WOLFSSL_CMAC` is also inexplicably enabled (CMAC is part of AES I believe) which leads to compiler errors in two additional places:

https://github.com/wolfSSL/wolfssl/blob/c3900470aaff38f46fca225aec7b572c5ecda02d/wolfssl/wolfcrypt/cryptocb.h#L435
https://github.com/wolfSSL/wolfssl/blob/c3900470aaff38f46fca225aec7b572c5ecda02d/wolfssl/wolfcrypt/cryptocb.h#L651

Is there another flag that turns off CMAC?

embhorn, 2024-09-18:
Hello @rdmark 

How are you configuring wolfSSL?

Also the branch you shared appears to be invalid:
```
git clone https://github.com/Netatalk/netatalk/tree/1430-nettle-and-wolfssl-both-define-aes_max_key_size
Cloning into '1430-nettle-and-wolfssl-both-define-aes_max_key_size'...
fatal: repository 'https://github.com/Netatalk/netatalk/tree/1430-nettle-and-wolfssl-both-define-aes_max_key_size/' not found
```

Thanks,
Eric - wolfSSL Support

rdmark, 2024-09-18:
My apologies, I have restored the branch now. Please try checking it out again.

We are configuring WolfSSL within the Meson build system of our project. It's treated as an embedded library. See the flags listed starting at https://github.com/Netatalk/netatalk/blob/c9265a4674536dc404a664f60bb66433e3148ba8/meson_config.h#L710

embhorn, 2024-09-18:
Hi @rdmark

How is wolfSSL picking up the config in https://github.com/Netatalk/netatalk/blob/c9265a4674536dc404a664f60bb66433e3148ba8/meson_config.h

Can you build wolfSSL as a shared lib in docker then just link against it? That would also make updating to a new version much easier!

Thanks,

rdmark, 2024-09-19:
WolfSSL is built as the "ssl" module in libatalk. When the `-Dwith-embedded-ssl` flag is set in Netatalk's build system (which is 'true' by default) the build file in https://github.com/Netatalk/netatalk/blob/main/libatalk/ssl/meson.build gets picked up and WolfSSL built as a shared library, using the same global flags as the entire package.

Thanks for the tip about Docker. In many instances this would be a good solution, I agree. However, our case is a little different. Our main "customers" are downstream package maintainers (FreeBSD, NetBSD, Fedora, Debian etc.) who take the tarball and integrate it with their packaging infrastructure for redistribution. The majority of distros out there aren't distributing packaged WolfSSL shared libraries (yet), which is why we decided to bundle it as an embedded library.

Does this explain our current setup sufficiently?

embhorn, 2024-09-19:
I was not able to reproduce the error using the instructions you shared:

```
~/test/gh7984/netatalk$ meson compile -C build
INFO: autodetecting backend as ninja
INFO: calculating backend command to run: /usr/bin/ninja -C /home/eric/test/gh7984/netatalk/build
ninja: Entering directory `/home/eric/test/gh7984/netatalk/build'
[13/387] Compiling C object libatalk/adouble/libadouble.a.p/ad_conv.c.o
../libatalk/adouble/ad_conv.c: In function ‘ad_conv_dehex’:
../libatalk/adouble/ad_conv.c:260:5: warning: argument 2 null where non-null expected [-Wnonnull]
  260 |     strlcpy(buf, bdata(newpath), sizeof(buf));
      |     ^~~~~~~
In file included from ../libatalk/adouble/ad_conv.c:29:
/usr/include/string.h:506:15: note: in a call to function ‘strlcpy’ declared ‘nonnull’
  506 | extern size_t strlcpy (char *__restrict __dest,
      |               ^~~~~~~
[201/387] Compiling C object etc/afpd/libafpd.a.p/uam.c.o
../etc/afpd/uam.c: In function ‘uam_getname’:
../etc/afpd/uam.c:213:17: warning: argument 1 null where non-null expected [-Wnonnull]
  213 |         pwent = getpwnam(bdata(princ));
      |                 ^~~~~~~~
In file included from ../include/atalk/uam.h:8,
                 from ../include/atalk/globals.h:23,
                 from ../include/atalk/dsi.h:17,
                 from ../etc/afpd/uam.c:30:
/usr/include/pwd.h:116:23: note: in a call to function ‘getpwnam’ declared ‘nonnull’
  116 | extern struct passwd *getpwnam (const char *__name) __nonnull ((1));
      |                       ^~~~~~~~
[207/387] Compiling C object etc/afpd/libafpd.a.p/afp_asp.c.o
../etc/afpd/afp_asp.c: In function ‘afp_authprint_remove’:
../etc/afpd/afp_asp.c:75:30: warning: format ‘%s’ expects argument of type ‘char *’, but argument 3 has type ‘int’ [-Wformat=]
   75 |     sprintf(addr_filename, "%s/net%d.%dnode%d", obj->options.authprintdir,
      |                             ~^                  ~~~~~~~~~~~~~~~~~~~~~~~~~
      |                              |                              |
      |                              char *                         int
      |                             %d
[285/387] Compiling C object etc/papd/papd.p/main.c.o
../etc/papd/main.c: In function ‘getstatus’:
../etc/papd/main.c:570:36: warning: ‘snprintf’ output may be truncated before the last format character [-Wformat-truncation=]
  570 |         snprintf(buf->buf, 254, "%s", pr->p_status);
      |                                    ^
../etc/papd/main.c:570:9: note: ‘snprintf’ output between 1 and 255 bytes into a destination of size 254
  570 |         snprintf(buf->buf, 254, "%s", pr->p_status);
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[377/387] Compiling C object test/afpd..._test_gh7984_netatalk_etc_afpd_uam.c.o
/home/eric/test/gh7984/netatalk/etc/afpd/uam.c: In function ‘uam_getname’:
/home/eric/test/gh7984/netatalk/etc/afpd/uam.c:213:17: warning: argument 1 null where non-null expected [-Wnonnull]
  213 |         pwent = getpwnam(bdata(princ));
      |                 ^~~~~~~~
In file included from ../include/atalk/uam.h:8,
                 from ../include/atalk/globals.h:23,
                 from ../include/atalk/dsi.h:17,
                 from /home/eric/test/gh7984/netatalk/etc/afpd/uam.c:30:
/usr/include/pwd.h:116:23: note: in a call to function ‘getpwnam’ declared ‘nonnull’
  116 | extern struct passwd *getpwnam (const char *__name) __nonnull ((1));
      |                       ^~~~~~~~
[387/387] Generating doc/manual/manual with a custom command
Writing table-toc.html for book(netatalk-manual)
Writing example-toc.html for book(netatalk-manual)
Writing intro.html for chapter(intro)
Writing installation.html for chapter(installation)
Writing configuration.html for chapter(configuration)
Writing appletalk.html for chapter(appletalk)
Writing upgrade.html for chapter(upgrade)
Writing a2boot.8.html for refentry(a2boot.8)
Writing ad.1.html for refentry(ad.1)
Writing aecho.1.html for refentry(aecho.1)
Writing afp.conf.5.html for refentry(afp.conf.5)
Writing afp_signature.conf.5.html for refentry(afp_signature.conf.5)
Writing afp_voluuid.conf.5.html for refentry(afp_voluuid.conf.5)
Writing afpd.8.html for refentry(afpd.8)
Writing afpldaptest.1.html for refentry(afpldaptest.1)
Writing afppasswd.1.html for refentry(afppasswd.1)
Writing afpstats.1.html for refentry(afpstats.1)
Writing apple_dump.1.html for refentry(apple_dump.1)
Writing asip-status.1.html for refentry(asip-status.1)
Writing atalk.4.html for refentry(atalk.4)
Writing atalkd.8.html for refentry(atalkd.8)
Writing atalkd.conf.5.html for refentry(atalkd.conf.5)
Writing atalk_aton.3.html for refentry(atalk_aton.3)
Writing cnid_dbd.8.html for refentry(cnid_dbd.8)
Writing cnid_metad.8.html for refentry(cnid_metad.8)
Writing dbd.1.html for refentry(dbd.1)
Writing extmap.conf.5.html for refentry(extmap.conf.5)
Writing getzones.1.html for refentry(getzones.1)
Writing macusers.1.html for refentry(macusers.1)
Writing nbp.1.html for refentry(nbp.1)
Writing nbp_name.3.html for refentry(nbp_name.3)
Writing netatalk.8.html for refentry(netatalk.8)
Writing netatalk-config.1.html for refentry(netatalk-config.1)
Writing pap.1.html for refentry(pap.1)
Writing papd.8.html for refentry(papd.8)
Writing papd.conf.5.html for refentry(papd.conf.5)
Writing papstatus.8.html for refentry(papstatus.8)
Writing timelord.8.html for refentry(timelord.8)
Writing man-pages.html for chapter(man-pages)
Writing compile.html for appendix(compile)
Writing gpl.html for appendix(gpl)
Writing manual-index.html for index(manual-index)
Writing index.html for book(netatalk-manual)
```

I am building in Ubuntu. Is there some other config different from the instruction?
> Follow steps in https://netatalk.io/4.0/htmldocs/compile to setup and compile the code

rdmark, 2024-09-20:
Ah of course, the branch I gave you actually contains the workaround for the issue, so of course you won’t see it…

Please check out the commit right before and try again, for instance: https://github.com/Netatalk/netatalk/tree/5682070edbce472047943922cdc57aee74def7ea