wolfSSL/wolfssl #8003: ML-DSA/Dilithium Calls ED

opencrypto, 2024-09-22:
There seem to be an issue with the ML-DSA / Dilithium code in WolfCrypt. Specifically, the `wc_Dilithium_PublicKeyToDer`, calls the `SetAsymDerPublic()` here:

- https://github.com/wolfSSL/wolfssl/blob/bc6881974d9a0e66118715812e7e10e9609a62cb/wolfcrypt/src/dilithium.c#L9730

The `SetAsymDerPublic()` is defined in the `asn.c` (L:12043) file and it is meant to be used for ED curves, not for generic ASN.1 (i.e., the documentation refers to RFC8410). The buffer size for the `dataASN` is set to `edPubKeyASN_Length`, which does not work for Dilithium.

Proposed Resolution:

1. Change the name of the function to be specific for ED curves, and
2. Fix the use in MlDsa/Dilithium code.

philljj, 2024-09-22:
Hi @opencrypto,

Thank you for the report, I will look at this.

Thanks,
Jordan

philljj, 2024-09-24:
Hi @opencrypto,

The documentation and variable names around `SetAsymKeyDerPublic()` are outdated, and we'll open a PR to update this. 

The `SetAsymKeyDerPublic()` is a generic function we use for sphincs, dilithium, falcon, Ed, etc. The `edPubKeyASN_Length` is just the number of ASN items in the template for this type of pub key.

Sorry for the confusion!

Thanks,
Jordan

dgarske, 2024-09-24:
https://github.com/wolfSSL/wolfssl/pull/8009

closed 2024-09-25 by JacobBarthelmeh