wolfSSL/wolfssl #8023: [Bug]: Kyber (original) is not working

Frauschi, 2024-09-27:
### Version

master

### Description

When compiling the WolfSSL version of Kyber with the Round 3 version enabled, tests are failing. The final FIPS version is working properly, however.

Failing configuration:
```
./configure --enable-all --enable-asn=template --enable-experimental --enable-kyber=yes,original
make
make check
```

Working configuration:
```
./configure --enable-all --enable-asn=template --enable-experimental --enable-kyber=yes
make
make check
```

SparkiDev, 2024-09-30:
Hi @Frauschi,

I've put up a pull request that fixes the original algorithm.
Thanks for raising the bug!

Sean

Frauschi, 2024-09-30:
Hi @SparkiDev,

I verified the patch on my side, it is working properly now. Thanks for the fast response!

Tobi

PR #8027 (2024-09-30): Kyber original: fix to work
# Description

Encapsulate the message (hash of rand) for original. Final of FIPS 203 uses rand.

Fixes #8023

# Testing

./configure --disable-shared --enable-kyber=yes,original

# Checklist

 - [ ] added tests
 - [ ] updated/added doxygen
 - [ ] updated appropriate READMEs
 - [ ] Updated manual and documentation
closed 2024-09-30 by dgarske