wolfSSL/wolfssl #8032: Force TLS client to receive stapled OCSP responses to all certificates in the chain

Dor-kpf, 2024-10-01:
### Version

5.7.2

### Description

WolfSSL built with `--enable-ocsp --enable-ocspstapling --enable-ocspstapling2`

Hello!
I have a TLS client application that uses OCSP stapling **only** (the client will not send OCSP requests to OCSP responder) to get the revocation status of all the certificates (**intermediates + server's certificate**) in the chain from the TLS server. If a stapled response is missing from the Certificate Status message, I would like to abort the TLS handshake.
I have enabled OCSP stapling V2 (for TLS 1.2) so the client will request multi-stapling and may receive the stapled responses to intermediates as well.

How I enabled OCSP stapling V2:
```
wolfSSL_EnableOCSPStapling(m_ssl)
wolfSSL_UseOCSPStaplingV2(m_ssl, WOLFSSL_CSR2_OCSP_MULTI, 0)
```
I have also called `wolfSSL_CTX_EnableOCSPMustStaple(m_ctx)` but looking at the code in `SanityCheckMsgReceived` function, this will only force the reception of the Certificate Status message, but this message may still be missing some responses.

My questions are:
1. Does WolfSSL provide a way for the TLS client to detect, during the TLS handshake, that a stapled response for either the intermediates or the server's certificate are missing from the Certificate Status message, so I can abort the TLS handshake in such case?
2. Alternatively, does WolfSSL provide a way to read the stapled responses in the Certificate Status message (inside the VerifyCallback for example) so I can examine them and check if there are missing staples and abort the TLS handshake accordingly?

Looks like all the OCSP stapling logic and checking are done internally by WolfSSL in `wolfSSL_connect` function and that the stapled responses are not exposed to the user by WolfSSL's API.

Thank you!

miyazakh, 2024-10-02:
@Dor-kpf 
Thank you for your interest in wolfSSL. 

#1, #2 : I don't think wolfSSL provides an interface or any way to access `Certificate Status Message` raw data as you mentioned. Let me double-check.
#3 : [PR#7766](https://github.com/wolfSSL/wolfssl/pull/7766) is supposed to support this.

miyazakh, 2025-12-19:
@Dor-kpf 
I’m going to close this issue. Please let me know if there’s anything else you’d like to discuss.

closed 2025-12-23 by miyazakh