wolfSSL/wolfssl #8041: Typo in kyber.h header prevents using wc_MlKemKey_Decapsulate alias

tobiasbrunner, 2024-10-04:
`wc_MlKemKey_Decapsulate` is incorrectly mapped to `wc_KyberKey_Encapsulate` instead of `wc_KyberKey_Decapsulate`:

https://github.com/wolfSSL/wolfssl/blob/2e539ef70cbf129dd2b1471ae0e0d243853880ca/wolfssl/wolfcrypt/kyber.h#L259

tobiasbrunner, 2024-10-04:
Just another small fixup:

https://github.com/wolfSSL/wolfssl/blob/2e539ef70cbf129dd2b1471ae0e0d243853880ca/configure.ac#L1272

Doesn't seem the experimental switch is required anymore.

anhu, 2025-01-31:
Hi @tobiasbrunner , 

Thank you so much for letting us know about these issues.  I will have fixes for them in a PR very soon. I sincerely apologize for the delay in my response time. 

Warm regards, Anthony

anhu, 2025-01-31:
Please see https://github.com/wolfSSL/wolfssl/pull/8405 .  I'm sure this will be merged fairly quickly.  As such I will now close this issue. Thank you again for letting us know about this. 

Warm regards, Anthony