wolfSSL/wolfssl #8067: examine what flags are enabled

RealYukiSan, 2024-10-13:
### Version

not related to specific version

### Description

Let's say I want to download [wolfssl binary](https://github.com/wolfSSL/wolfssl/releases/tag/v5.7.2-stable), and I want to know what flags are enabled of installed pre-built binaries, how to identify the flags?

by flags, I mean build flag such as `--enable-tlsv12`, `--enable-sys-ca-certs`, etc...

kareem-wolfssl, 2024-10-14:
Hi @RealYukiSan ,

Great question.  For wolfSSL packages, you can find the build settings used to configure it in <include path>/wolfssl/options.h (for example for a local installation, /usr/local/include/wolfssl/options.h).  There isn't a way to get the configure line itself, the configure line is translated to these build options by configure.

RealYukiSan, 2024-10-15:
Thanks for the information @kareem-wolfssl.

closed 2024-10-15 by RealYukiSan