wolfSSL/wolfssl #8121: MSVC: Missing 'nanosleep()'

gvanem, 2024-10-29:
### Version

From a `git pull` yesterday

### Description

Trying to build the `unit.exe` test-program using MSVC on Win-10, I'm getting a link-error; unknown function `nanosleep()`.
Should be easy enough to fix. Here is my attempt:
```diff
--- a/tests/api.c 2024-10-28 10:28:38
+++ b/tests/api.c 2024-10-29 08:34:59
@@ -591,6 +591,15 @@
 static int testDevId = INVALID_DEVID;
 #endif

+#ifdef _MSC_VER
+  static void nanosleep(const struct timespec *ts,
+                       struct timespec *rem)
+  {
+    SleepEx (1000*ts->tv_sec + ts->tv_nsec/1000000, TRUE);
+    (void) rem;
+  }
+#endif
+
 #if !defined(NO_FILESYSTEM) && !defined(NO_CERTS) && !defined(NO_RSA) && \
     !defined(NO_WOLFSSL_SERVER) && !defined(NO_WOLFSSL_CLIENT)
```
(I assume any MinGW does have it).

But `unit.exe` is crashing (on an unrelated issue) at runtime inside `UCRT`.
```c
cdb -c g bin\unit.exe
  ...
   383: test_wolfSSL_PEM_PrivateKey                         : passed (  0.00067)
   384: test_wolfSSL_PEM_file_RSAKey                        : skipped
   385: test_wolfSSL_PEM_file_RSAPrivateKey                 : skipped
   386: test_wolfSSL_BIO                                    :
(34d8.508c): Security check failure or stack buffer overrun - code c0000409 (!!! second chance !!!)
Subcode: 0x5 FAST_FAIL_INVALID_ARG
ucrtbase!invoke_watson+0x18:
00007ff8`a4131208 cd29            int     29h
0:000> kp
Child-SP          RetAddr               Call Site
000000e9`c554f390 00007ff8`a40e24b1     ucrtbase!invoke_watson+0x18
000000e9`c554f3c0 00007ff8`a40e2379     ucrtbase!_invalid_parameter+0x12d
000000e9`c554f400 00007ff8`a4134bcb     ucrtbase!invalid_parameter_noinfo+0x9
000000e9`c554f440 00007ff8`a413c902     ucrtbase!__acrt_stdio_parse_mode<char>+0x27f
000000e9`c554f470 00007ff8`a41354a8     ucrtbase!common_openfile<char>+0x22
*** WARNING: Unable to verify checksum for unit.exe
000000e9`c554f4e0 00007ff7`3b247106     ucrtbase!common_fsopen<char>+0x9c
000000e9`c554f530 00007ff7`3b253a5d     unit!test_wolfSSL_BIO(void)+0x6156
000000e9`c554f790 00007ff7`3b28a990     unit!ApiTest(void)+0xcd
000000e9`c554f7f0 00007ff7`3b299080     unit!unit_test(int argc = 0n1, char ** argv = 0x00000232`1cbd5120)+0x210
(Inline Function) --------`--------     unit!invoke_main(void)+0x22
000000e9`c554f840 00007ff8`a62f7374     unit!__scrt_common_main_seh(void)+0x10c
000000e9`c554f880 00007ff8`a687cc91     KERNEL32!BaseThreadInitThunk+0x14
000000e9`c554f8b0 00000000`00000000     ntdll!RtlUserThreadStart+0x21
```

Any pointers as to why? The `certs/server-cert.pem` exists.

philljj, 2024-10-30:
Hi @gvanem,

Thank you for the report!

How are you trying to build tests/api.c and tests/unit.c? Are you building through VS IDE or command line?

We have a number of different VS solutions for wolfSSL. E.g. we just added new VS 2022 solutions for wolfssl test and benchmark here in this PR https://github.com/wolfSSL/wolfssl/pull/8090.

There's also this documentation that's useful:
- https://www.wolfssl.com/documentation/manuals/wolfssl/chapter02.html#building-on-windows

You can also build on command line in WSL on Windows, and follow our unix-like system steps:
- https://www.wolfssl.com/documentation/manuals/wolfssl/chapter02.html#building-on-unix-like-systems

Best,
Jordan

gvanem, 2024-10-31:
> How are you trying to build tests/api.c and tests/unit.c?

Via a home-made GNU-makefile.

So the bottom-line is that MSVC is not supported for the tests? It seems PR https://github.com/wolfSSL/wolfssl/pull/8090 was
merged *before* I hit the issue. Even then `nanosleep()` is not there for MSVC/clang-cl?!

Regarding the `invoke_watson` stuff, it seems some `fopen()` flags like `"rwb"` and `"ar"` triggers this. Seems illegal.
I fixed that by adding a  `_set_invalid_parameter_handler (invalid_parameter_handler);`  call.