wolfSSL/wolfssl #8124: [Bug]: ARM ASM without crypto extensions doesn't compile

Frauschi, 2024-10-30:
### Version

master

### Description

When compiling for an ARMv8 target without support for the ARM crypto extensions (e.g. Raspberry Pi 4), compilation fails with undefined references.

I'm compiling with the following flags set to enable the code for the ARM port without crypto extensions:
- `WOLFSSL_ARMASM`
- `WOLFSSL_ARMASM_NO_HW_CRYPTO`
- `WOLFSSL_AARCH64_NO_SQRMLSH`

The undefined references are:
- `AES_CBC_decrypt`
- `AES_CBC_encrypt`
- `Transform_Sha256_Len`
- `AES_invert_key`
- `AES_GCM_encrypt`
- `GCM_gmult_len`
- `AES_set_encrypt_key`
- `AES_ECB_encrypt`

All these methods are defined as `extern` in the relevant files (`armv8-aes.c` and `armv8-sha256.c`). After searching through the code base, there seem to be no implementations for these methods except in the `armv8-32-x` and `thumb2-xxx` files, which don't work on AArch64. 

Do I miss some flags to map these methods to other implementations?

dgarske, 2024-10-30:
Hi @Frauschi ,

Thank you for the detailed report. I can reproduce on my Pi4. Building with `--enable-armasm` gets illegal instruction because chip doesn't support `sha256`/`aes` crypto instructions. 

When building with something like `./configure --enable-armasm=inline CFLAGS="-DWOLFSSL_ARMASM_NO_HW_CRYPTO -DWOLFSSL_ARMASM_NO_NEON"` there are missing link-time functions for `Transform_Sha256_Len`, `AES_set_encrypt_key`, etc... and I can't find an aarch64 variation with `WOLFSSL_ARMASM_NO_HW_CRYPTO` in the code.

@SparkiDev can you review our aarch64 support without crypto extensions?

Thanks,
David Garske, wolfSSL

SparkiDev, 2024-10-31:
Hi @Frauschi,

As far as I know we have never had support for ASM with ARMv8-A.

Have you had any success in the past or are you requesting this feature?

Thanks,
Sean

--
Sean Parkinson, wolfSSL
Senior Software Engineer

Frauschi, 2024-10-31:
Thanks for your fast responses @dgarske @SparkiDev!

> Have you had any success in the past or are you requesting this feature?

I haven't had any success on that side. I was looking into the ARM ASM stuff as we migrated our internal testing from Raspberry Pi 4 to Pi 5 and I wanted to test the performance boost of the ARM Crypto extensions (impressive results btw). In this process, I realized that the same config now (obviously) doesn't work anymore on the Pi 4. When I found the `WOLFSSL_ARMASM_NO_HW_CRYPTO` option, I thought that this also works for ARMv8-A. 

I don't really "need" the feature, tbh. However, I think it would be nice in the long run to at least use the other ASM optimizations (e.g. for Kyber etc.) on ARMv8-A, which work without the crypto extensions. But that is also no priority on my side. So feel free to close the issue if you don't see any additional discussion/work on this topic.

SparkiDev, 2024-10-31:
Hi @Frauschi,

I will add this to our feature request list.
Personally, I would like to see this done and I'll see if I can find time in the future do work on this.

I will close this ticket as this is a feature request.

Sean

closed 2024-10-31 by SparkiDev