wolfSSL/wolfssl #8131: WolfSSL FIPS Windows MSYS

volga629-1, 2024-10-30:
### Version

5.7.4 FIPS

### Description

Hello Everyone,
I am trying to compile WolfSSL on Windows MSYS and make -j4 quits, any help thank you. 

```
Configuration summary for wolfssl version 5.7.4

   * Installation prefix:        /home/volga629/strongswan-5.9.14/wolfssl-5.7.4-gplv3-fips-ready/wolfssl-fips-build
   * System type:                w64-mingw32
   * Host CPU:                   x86_64
   * C Compiler:                 x86_64-w64-mingw32-gcc
   * C Flags:                    -DWOLFSSL_STATIC_RSA -DWOLFSSL_LOG_PRINTF -DWOLFSSL_PUBLIC_MP -DHAVE_EX_DATA -DFP_MAX_BITS=8192   -Wno-pragmas -Wall -Wextra -Wunknown-pragmas --param=ssp-buffer-size=1 -Waddress -Warray-bounds -Wbad-function-cast -Wchar-subscripts -Wcomment -Wfloat-equal -Wformat-security -Wformat=2 -Wmaybe-uninitialized -Wmissing-field-initializers -Wmissing-noreturn -Wmissing-prototypes -Wnested-externs -Wnormalized=id -Woverride-init -Wpointer-arith -Wpointer-sign -Wshadow -Wsign-compare -Wstrict-overflow=1 -Wstrict-prototypes -Wswitch-enum -Wundef -Wunused -Wunused-result -Wunused-variable -Wwrite-strings -fwrapv
   * C++ Compiler:
   * C++ Flags:
   * CPP Flags:
   * CCAS Flags:                 -DWOLFSSL_STATIC_RSA -DWOLFSSL_LOG_PRINTF -DWOLFSSL_PUBLIC_MP -DHAVE_EX_DATA -DFP_MAX_BITS=8192
   * LD Flags:                    -lws2_32
   * LIB Flags:                   -pie
   * Library Suffix:
   * Debug enabled:              no
   * Coverage enabled:
   * Warnings as failure:        no
   * make -j:                    9
   * VCS checkout:               no

   Features
   * Experimental settings:      Forbidden
   * FIPS:                       v2
   * Single threaded:            no
   * Filesystem:                 yes
   * OpenSSH Build:              no
   * OpenSSL Extra API:          yes
   * OpenSSL Coexist:            no
   * Old Names:                  yes
   * Max Strength Build:         no
   * Distro Build:               no
   * Reproducible Build:         yes
   * Side-channel Hardening:     yes
   * Single Precision Math:      no
   * SP implementation:          no
   * Fast Math:                  yes
   * Heap Math:                  no
   * Assembly Allowed:           yes
   * sniffer:                    no
   * snifftest:                  no
   * ARC4:                       no
   * AES:                        yes
   * AES-NI:                     no
   * AES-CBC:                    yes
   * AES-CBC length checks:      no
   * AES-GCM:                    yes
   * AES-GCM streaming:          no
   * AES-CCM:                    yes
   * AES-CTR:                    yes
   * AES-CFB:                    no
   * AES-OFB:                    no
   * AES-XTS:                    no
   * AES-XTS streaming:          no
   * AES-SIV:                    no
   * AES-EAX:                    no
   * AES Bitspliced:             no
   * AES Key Wrap:               no
   * ARIA:                       no
   * DES3:                       yes
   * DES3 TLS Suites:            no
[... 538 lines cut ...]
 2025 |     byte     hash[FIPS_IN_CORE_DIGEST_SIZE];
      |              ^~~~
wolfcrypt/src/fips_test.c: At top level:
wolfcrypt/src/fips_test.c:2234: warning: ignoring '#pragma warning ' [                                         ]
 2234 |     #pragma warning(pop)
      |
 warning: ignoring '#pragma code_seg ' [-Wunknown-pragmas]
   38 |         #pragma code_seg(".fipsA$g")
      |
wolfcrypt/src/hmac.c:39: warning: ignoring '#pragma const_seg ' [-Wunknown-pragmas]
   39 |         #pragma const_seg(".fipsB$g")
      |
make[2]: *** [Makefile:7335: wolfcrypt/src/src_libwolfssl_la-fips_test.lo] Error 1
wolfcrypt/src/random.c:53: warning: ignoring '#pragma code_seg ' [-Wunknown-pragmas]
   53 |         #pragma code_seg(".fipsA$i")
      |
wolfcrypt/src/random.c:54: warning: ignoring '#pragma const_seg ' [-Wunknown-pragmas]
   54 |         #pragma const_seg(".fipsB$i")
      |
make[2]: *** [Makefile:7213: wolfcrypt/test/test.o] Error 1
make[2]: Leaving directory '/home/volga629/strongswan-5.9.14/wolfssl-5.7.4-gplv3-fips-ready'
make[1]: *** [Makefile:9026: all-recursive] Error 1
make[1]: Leaving directory '/home/volga629/strongswan-5.9.14/wolfssl-5.7.4-gplv3-fips-ready'
make: *** [Makefile:5376: all] Error 2
```

anhu, 2024-10-31:
Hi @volga629-1 

According to your configure script output, this is how you executed the configure script: 

./configure flags: --host=x86_64-w64-mingw32 --enable-keygen --enable-rsapss --enable-dtls --enable-certgen --enable-certreq --enable-certext --enable-sessioncerts --enable-crl --enable-ocsp --enable-secure-renegotiation --enable-strongswan 'CFLAGS=-DWOLFSSL_STATIC_RSA -DWOLFSSL_LOG_PRINTF -DWOLFSSL_PUBLIC_MP -DHAVE_EX_DATA -DFP_MAX_BITS=8192' --enable-ed25519 --enable-curve25519 --enable-fips=v2 --prefix=/home/volga629/strongswan-5.9.14/wolfssl-5.7.4-gplv3-fips-ready/wolfssl-fips-build --enable-ecc --disable-fpecc --disable-aligndata --disable-static --disable-jni --disable-crl-monitor --disable-examples host_alias=x86_64-w64-mingw32

I noticed `--enable-fips=v2` but you are using a fips-ready source package, so you need to replace that with ` --enable-fips=ready`

Warm regards, Anthony

volga629-1, 2024-11-01:
Thank you for reply, getting closer 
Currently get this error 

with options 

```
 ./configure --host=x86_64-w64-mingw32 --enable-reproducible-build --enable-keygen --enable-rsapss \
			--enable-certgen --enable-certreq --enable-certext \
			--enable-sessioncerts --enable-crl --enable-ocsp \
			--enable-tls13 --enable-dtls \
			--enable-secure-renegotiation --enable-strongswan  CPPFLAGS="-I/usr/include" CFLAGS="-DWOLFSSL_NO_ATOMICS -DWOLFSSL_LOG_PRINTF -DWOLFSSL_PUBLIC_MP -DHAVE_EX_DATA" \
			--enable-ed25519 --enable-curve25519 --enable-fips=ready  --prefix=$(pwd)/wolfssl-fips-build \
			--enable-ecc \
			--disable-fpecc \
			--disable-aligndata \
			--disable-static \
			--disable-jni \
			--disable-crl-monitor\
			--disable-examples 

```

```
configure:42436: Generating user options header...
configure:42550: checking for wolfssl/openssl/aes.h
configure:42550: x86_64-w64-mingw32-gcc -c -DWOLFSSL_NO_ATOMICS -DWOLFSSL_LOG_PRINTF -DWOLFSSL_PUBLIC_MP -DHAVE_EX_DATA   -Wno-pragmas -Wall -Wextra -Wunknown-pragmas --param=ssp-buffer-size=1 -Waddress -Warray-bounds -Wbad-function-cast -Wchar-subscripts -Wcomment -Wfloat-equal -Wformat-security -Wformat=2 -Wmaybe-uninitialized -Wmissing-field-initializers -Wmissing-noreturn -Wmissing-prototypes -Wnested-externs -Wnormalized=id -Woverride-init -Wpointer-arith -Wpointer-sign -Wshadow -Wsign-compare -Wstrict-overflow=1 -Wstrict-prototypes -Wswitch-enum -Wundef -Wunused -Wunused-result -Wunused-variable -Wwrite-strings -fwrapv -DHAVE_CONFIG_H -I. -I. -I/usr/include conftest.c >&5
In file included from ./wolfssl/wolfcrypt/types.h:35,
                 from ./wolfssl/wolfcrypt/aes.h:37,
                 from ./wolfssl/openssl/aes.h:35,
                 from conftest.c:63:
./wolfssl/wolfcrypt/wc_port.h:870:20: error: field 's' has incomplete type
  870 |         XSTAT_TYPE s;
      |                    ^
In file included from ./wolfssl/wolfcrypt/wc_port.h:1167:
C:/msys64/usr/include/sys/time.h:373: warning: "timerclear" redefined
  373 | #define timerclear(tvp)         ((tvp)->tv_sec = (tvp)->tv_usec = 0)
      |
In file included from C:/msys64/mingw64/include/winsock2.h:52,
                 from ./wolfssl/wolfcrypt/wc_port.h:81:
C:/msys64/mingw64/include/_timeval.h:20: note: this is the location of the previous definition
   20 | #define timerclear(tvp)         (tvp)->tv_sec = (tvp)->tv_usec = 0
      |
C:/msys64/usr/include/sys/time.h:375: warning: "timercmp" redefined
  375 | #define timercmp(tvp, uvp, cmp)                                 \
      |
C:/msys64/mingw64/include/_timeval.h:17: note: this is the location of the previous definition
   17 | #define timercmp(tvp,uvp,cmp)                                   \
      |
In file included from ./wolfssl/wolfcrypt/wc_port.h:1332:
C:/msys64/usr/include/stdatomic.h:201:17: error: unknown type name 'int_least8_t'
  201 | typedef _Atomic(int_least8_t)           atomic_int_least8_t;
      |                 ^~~~~~~~~~~~
C:/msys64/usr/include/stdatomic.h:202:17: error: unknown type name 'uint_least8_t'
  202 | typedef _Atomic(uint_least8_t)          atomic_uint_least8_t;
      |                 ^~~~~~~~~~~~~
C:/msys64/usr/include/stdatomic.h:203:17: error: unknown type name 'int_least16_t'
  203 | typedef _Atomic(int_least16_t)          atomic_int_least16_t;
      |                 ^~~~~~~~~~~~~
C:/msys64/usr/include/stdatomic.h:204:17: error: unknown type name 'uint_least16_t'
  204 | typedef _Atomic(uint_least16_t)         atomic_uint_least16_t;
      |                 ^~~~~~~~~~~~~~
C:/msys64/usr/include/stdatomic.h:205:17: error: unknown type name 'int_least32_t'
  205 | typedef _Atomic(int_least32_t)          atomic_int_least32_t;
      |                 ^~~~~~~~~~~~~
C:/msys64/usr/include/stdatomic.h:206:17: error: unknown type name 'uint_least32_t'
  206 | typedef _Atomic(uint_least32_t)         atomic_uint_least32_t;
      |                 ^~~~~~~~~~~~~~
C:/msys64/usr/include/stdatomic.h:207:17: error: unknown type name 'int_least64_t'
  207 | typedef _Atomic(int_least64_t)          atomic_int_least64_t;
      |                 ^~~~~~~~~~~~~
C:/msys64/usr/include/stdatomic.h:208:17: error: unknown type name 'uint_least64_t'
  208 | typedef _Atomic(uint_least64_t)         atomic_uint_least64_t;
      |                 ^~~~~~~~~~~~~~
C:/msys64/usr/include/stdatomic.h:209:17: error: unknown type name 'int_fast8_t'
  209 | typedef _Atomic(int_fast8_t)            atomic_int_fast8_t;
      |                 ^~~~~~~~~~~
C:/msys64/usr/include/stdatomic.h:210:17: error: unknown type name 'uint_fast8_t'
  210 | typedef _Atomic(uint_fast8_t)           atomic_uint_fast8_t;
      |                 ^~~~~~~~~~~~
C:/msys64/usr/include/stdatomic.h:211:17: error: unknown type name 'int_fast16_t'
  211 | typedef _Atomic(int_fast16_t)           atomic_int_fast16_t;
      |                 ^~~~~~~~~~~~
C:/msys64/usr/include/stdatomic.h:212:17: error: unknown type name 'uint_fast16_t'
  212 | typedef _Atomic(uint_fast16_t)          atomic_uint_fast16_t;
      |                 ^~~~~~~~~~~~~
C:/msys64/usr/include/stdatomic.h:213:17: error: unknown type name 'int_fast32_t'
  213 | typedef _Atomic(int_fast32_t)           atomic_int_fast32_t;
      |                 ^~~~~~~~~~~~
C:/msys64/usr/include/stdatomic.h:214:17: error: unknown type name 'uint_fast32_t'
  214 | typedef _Atomic(uint_fast32_t)          atomic_uint_fast32_t;
      |                 ^~~~~~~~~~~~~
C:/msys64/usr/include/stdatomic.h:215:17: error: unknown type name 'int_fast64_t'
  215 | typedef _Atomic(int_fast64_t)           atomic_int_fast64_t;
      |                 ^~~~~~~~~~~~
C:/msys64/usr/include/stdatomic.h:216:17: error: unknown type name 'uint_fast64_t'
  216 | typedef _Atomic(uint_fast64_t)          atomic_uint_fast64_t;
      |                 ^~~~~~~~~~~~~
configure:42550: $? = 1
configure: failed program was:

```

anhu, 2024-11-01:
Hi, please use visual studio.  That is the best method for building wolfSSL on windows.

volga629-1, 2024-11-01:
I need all headers  to build strongswan with wolfssl support.

closed 2024-11-01 by volga629-1