wolfSSL/wolfssl #8142: Hybrid Post Quantum Algorithms

Mohre, 2024-11-03:
### Version

5.7.4

### Description

I noticed Wolfssl supports these Post Quantum Algorithms. 
[image]

1. Does Wolfssl support composite ML-DSA algorithms like `MLDSA87-ECDSA-P384-SHA512` mentioned in [this draft](https://datatracker.ietf.org/doc/draft-ietf-lamps-pq-composite-sigs/)?
2. Do you have any plan to support hybrid ML-KEM algorithms mentioned in [this draft](https://datatracker.ietf.org/doc/draft-ietf-lamps-pq-composite-kem/)? It said "521 bit curve not widely used." and suggest `MLKEM1024-ECDH-P384`(Something like P384_KYBER_LEVEL5) for standardization.

kareem-wolfssl, 2024-11-05:
Hi @Mohre ,

Great question.  We don't currently support this, we do hope to support this in the future but don't have a concrete timeline yet.  We can set up a feature request for this if you'd like, please contact us at support [AT] wolfssl [DOT] com to set this up.
Can you share a bit of information on your use case for this?

opencrypto, 2024-11-13:
If you are interested in collaborating, I am working on a Pull Request to implement Composite Signatures. Here's the link to the fork I am using (main branch aligned with master in wolfSSL, mldsa-composite-1 is the branch I use to add the composite). Also, you can checkout the `examples/pktool/pktool.c` that I added to the examples build - it is a small tool intended to help generating keys, requests, and certs: https://github.com/opencrypto/wolfssl/tree/mldsa-composite-1.

Any help/comment/support/feedback is highly appreciated!

Mohre, 2024-11-13:
> Can you share a bit of information on your use case for this?

I investigating on PQS algorithms usage with TLS1.3 for communication.

anhu, 2024-12-02:
Hi @Mohre 

Please do have a look at @opencrypto 's PR.  We here at wolfSSL  have been following his work and are delighted with his efforts. 

We do not support `MLKEM1024-ECDH-P384` but instead do support `MLKEM1024-ECDH-P521` as well as `MLKEM768-ECDH-P384` .  Unless there is strong interest, I don't forsee us adding  `MLKEM1024-ECDH-P384` anytime soon.  That said, please send a feature request message to support@ if you feel you want to see wolfSSL support  `MLKEM1024-ECDH-P384` . 

Warm regards, Anthony