wolfSSL/wolfssl #8144: [Bug]: wolfSSL_CTX_use_certificate_buffer documentation argument nits

bagder, 2024-11-04:
### Contact Details

reply here

### Version

current online docs

### Description

The example snippet for wolfSSL_CTX_use_certificate_buffer does two "suspicious" things:

1. it sets `sz` to zero, when it should be the size of the buffer
2. the third argument wants a `long` so `sz` is using the wrong type

bagder, 2024-11-04:
This same mistake exists for multiple function calls in the documentation.

ColtonWilley, 2024-11-07:
@bagder I have a PR up here https://github.com/wolfSSL/wolfssl/pull/8161 which should fix this issue, please let me know if you had something else in mind.