wolfSSL/wolfssl #8148: [Bug]: Build fails on aarch64 with LTO

vifino, 2024-11-04:
### Contact Details

Here, on GitHub, preferrably.

### Version

5.7.4

### Description

The build fails with error messages like the following on aarch64 Linux machines, which should compile wolfssl for `-mcpu=armv8-a` with LTO. 5.7.2 worked fine.

./configure flags: `--disable-static --disable-dependency-tracking --prefix=/nix/store/q5adnsfcgdhbcc5siq3c5v8w5d9h19dr-wolfssl-all-5.7.4 --bindir=/nix/store/q5adnsfcgdhbcc5siq3c5v8w5d9h19dr-wolfssl-all-5.7.4/bin --sbindir=/nix/store/q5adnsfcgdhbcc5siq3c5v8w5d9h19dr-wolfssl-all-5.7.4/sbin --includedir=/nix/store/l33016xyxb23f8l4v0hbw7kx086hssff-wolfssl-all-5.7.4-dev/include --oldincludedir=/nix/store/l33016xyxb23f8l4v0hbw7kx086hssff-wolfssl-all-5.7.4-dev/include --mandir=/nix/store/q5adnsfcgdhbcc5siq3c5v8w5d9h19dr-wolfssl-all-5.7.4/share/man --infodir=/nix/store/q5adnsfcgdhbcc5siq3c5v8w5d9h19dr-wolfssl-all-5.7.4/share/info --docdir=/nix/store/fjivg6dwixisnz8vs29d74mkk0ch4f6d-wolfssl-all-5.7.4-doc/share/doc/wolfssl --libdir=/nix/store/bkkhgrpb580lvqgvwr06750pk9j12sjr-wolfssl-all-5.7.4-lib/lib --libexecdir=/nix/store/bkkhgrpb580lvqgvwr06750pk9j12sjr-wolfssl-all-5.7.4-lib/libexec --localedir=/nix/store/bkkhgrpb580lvqgvwr06750pk9j12sjr-wolfssl-all-5.7.4-lib/share/locale --enable-all --enable-reproducible-build --enable-pkcs11 --enable-writedup --enable-base64encode --enable-bigcache --enable-sp=yes,asm --enable-sp-math-all --enable-harden CC=gcc`

CFLAGS contain `-fPIC -O2 -U_FORTIFY_SOURCE -Wformat -Wformat-security -Werror=format-security -fzero-call-used-regs=used-gpr -fstack-protector-strong --param ssp-buffer-size=4 -fno-strict-overflow -march=armv8-a`
LDFLAGS contain `-flto`

### Reproduction steps

1) Set up [Nix](https://nixos.org/download/) on an aarch64-linux machine.
2) Clone my fork with wolfssl 5.7.4: `git clone -b wolfssl-5.7.4 --depth=1 https://github.com/vifino/nixpkgs`
3) `nix-build -A wolfssl`
4) Watch it fail. :(

### Relevant log output

```shell
CC       tests/unit_test-quic.o
  CCLD     src/libwolfssl.la
In function 'SetObjectId',
    inlined from 'wc_PKCS12_encrypt_content' at wolfcrypt/src/pkcs12.c:2194:24:
wolfcrypt/src/asn.c:5859:21: warning: writing 1 byte into a region of size 0 [8;;https://gcc.gnu.org/onlinedocs/gcc/Warning-Options.html#index-Wstringop-overflow=-Wstringop-overflow=8;;]
 5859 |         output[idx] = ASN_OBJECT_ID;
      |                     ^
In function 'wc_PKCS12_encrypt_content':
lto1: note: destination object is likely at address zero
In function 'SetObjectId',
    inlined from 'wc_PKCS12_encrypt_content' at wolfcrypt/src/pkcs12.c:2105:24:
wolfcrypt/src/asn.c:5859:21: warning: writing 1 byte into a region of size 0 [8;;https://gcc.gnu.org/onlinedocs/gcc/Warning-Options.html#index-Wstringop-overflow=-Wstringop-overflow=8;;]
 5859 |         output[idx] = ASN_OBJECT_ID;
      |                     ^
In function 'wc_PKCS12_encrypt_content':
lto1: note: destination object is likely at address zero
In function 'SetHeader',
    inlined from 'SetSequence' at wolfcrypt/src/asn.c:16025:12,
    inlined from 'wc_i2d_PKCS12' at wolfcrypt/src/pkcs12.c:927:20:
wolfcrypt/src/asn.c:16010:19: warning: writing 1 byte into a region of size 0 [8;;https://gcc.gnu.org/onlinedocs/gcc/Warning-Options.html#index-Wstringop-overflow=-Wstringop-overflow=8;;]
16010 |         output[0] = tag;
      |                   ^
In function 'wc_i2d_PKCS12':
lto1: note: destination object is likely at address zero
/build/ccfEpj3u.s: Assembler messages:
/build/ccfEpj3u.s:49: Error: selected processor does not support `aese v0.16b,v1.16b'
/build/ccfEpj3u.s:50: Error: selected processor does not support `aesmc v0.16b,v0.16b'
/build/ccfEpj3u.s:52: Error: selected processor does not support `aese v0.16b,v2.16b'
/build/ccfEpj3u.s:53: Error: selected processor does not support `aesmc v0.16b,v0.16b'
/build/ccfEpj3u.s:64: Error: selected processor does not support `aese v0.16b,v3.16b'
/build/ccfEpj3u.s:65: Error: selected processor does not support `aesmc v0.16b,v0.16b'
/build/ccfEpj3u.s:68: Error: selected processor does not support `aese v0.16b,v4.16b'
/build/ccfEpj3u.s:69: Error: selected processor does not support `aesmc v0.16b,v0.16b'
/build/ccfEpj3u.s:72: Error: selected processor does not support `aese v0.16b,v5.16b'
/build/ccfEpj3u.s:73: Error: selected processor does not support `aesmc v0.16b,v0.16b'
/build/ccfEpj3u.s:74: Error: selected processor does not support `aese v15.16b,v1.16b'
/build/ccfEpj3u.s:75: Error: selected processor does not support `aesmc v15.16b,v15.16b'
/build/ccfEpj3u.s:76: Error: selected processor does not support `aese v0.16b,v6.16b'
/build/ccfEpj3u.s:77: Error: selected processor does not support `aesmc v0.16b,v0.16b'
/build/ccfEpj3u.s:78: Error: selected processor does not support `aese v15.16b,v2.16b'
/build/ccfEpj3u.s:79: Error: selected processor does not support `aesmc v15.16b,v15.16b'
/build/ccfEpj3u.s:80: Error: selected processor does not support `aese v0.16b,v7.16b'
/build/ccfEpj3u.s:81: Error: selected processor does not support `aesmc v0.16b,v0.16b'
[... 511 lines cut ...]
/build/ccpHHaN6.s:663: Error: selected processor does not support `sha256su1 v13.4s,v11.4s,v12.4s'
/build/ccpHHaN6.s:664: Error: selected processor does not support `sha256h q16,q17,v9.4s'
/build/ccpHHaN6.s:665: Error: selected processor does not support `sha256h2 q17,q18,v9.4s'
/build/ccpHHaN6.s:666: Error: selected processor does not support `sha256su0 v14.4s,v11.4s'
/build/ccpHHaN6.s:669: Error: selected processor does not support `sha256su1 v14.4s,v12.4s,v13.4s'
/build/ccpHHaN6.s:670: Error: selected processor does not support `sha256h q16,q17,v10.4s'
/build/ccpHHaN6.s:671: Error: selected processor does not support `sha256h2 q17,q18,v10.4s'
/build/ccpHHaN6.s:672: Error: selected processor does not support `sha256su0 v15.4s,v12.4s'
/build/ccpHHaN6.s:675: Error: selected processor does not support `sha256su1 v15.4s,v13.4s,v14.4s'
/build/ccpHHaN6.s:676: Error: selected processor does not support `sha256h q16,q17,v11.4s'
/build/ccpHHaN6.s:677: Error: selected processor does not support `sha256h2 q17,q18,v11.4s'
/build/ccpHHaN6.s:681: Error: selected processor does not support `sha256h q16,q17,v12.4s'
/build/ccpHHaN6.s:682: Error: selected processor does not support `sha256h2 q17,q18,v12.4s'
/build/ccpHHaN6.s:685: Error: selected processor does not support `sha256h q16,q17,v13.4s'
/build/ccpHHaN6.s:686: Error: selected processor does not support `sha256h2 q17,q18,v13.4s'
/build/ccpHHaN6.s:689: Error: selected processor does not support `sha256h q16,q17,v14.4s'
/build/ccpHHaN6.s:690: Error: selected processor does not support `sha256h2 q17,q18,v14.4s'
/build/ccpHHaN6.s:693: Error: selected processor does not support `sha256h q16,q17,v15.4s'
/build/ccpHHaN6.s:694: Error: selected processor does not support `sha256h2 q17,q18,v15.4s'
make[3]: *** [/build/cca8wKCq.mk:2: /build/ccPSdxbB.ltrans0.ltrans.o] Error 1
lto-wrapper: fatal error: make returned 2 exit status
compilation terminated.
/nix/store/wj7hxrvf8jqipkdry7x6ii1ds1ry2hd5-binutils-2.43.1/bin/ld: error: lto-wrapper failed
collect2: error: ld returned 1 exit status
```

kareem-wolfssl, 2024-11-06:
Hi @vifino ,

Please try rebuilding with your mcpu/march args set to `armv8-a+crypto` and let me know if it helps.

vifino, 2024-11-06:
Hey @kareem-wolfssl, it probably would compile, except then it would require the crypto extensions, no?
We need this to run on plain `armv8-a`.

vifino, 2024-11-07:
To be clear: Building 5.7.2 for `armv8-a` with assembly speedups worked fine.
5.7.4 does not.

Disabling *all* of the previous optimized implementations because of a new implementation does not seem like a good solution to me but rather like a big regression.

Please gate the part requiring `+crypto` extensions. I would really appreciate it.

dgarske, 2024-11-07:
Hi @vifino ,

I think the difference between the versions is that we try and enable --enable-armasm by default. Can you please try with `--disable-armasm`?

Thanks,
David Garske, wolfSSL

vifino, 2024-11-15:
Hey @dgarske.

Yes, the code now compiles without error.

I find the naming of the `armasm` options very confusing. Is this specifically the option to enable the `crypto` extension?

What options would yield the fastest code on aarch64 `armv8-a` machines like the Raspberry Pi 4 (that lack `+crypto` extensions)?

dgarske, 2024-11-15:
Hi @vifino ,

Yes for the aarch64 the `--enable-armasm` option enables the aes/sha256 crypto assembly instruction speedups.

As of right now the only way to use the aarch64 speedups is with the crypto extensions, however I believe it is possible to enable the aarch32 assembly speedups, which would still be significantly faster than the C implementation.

Which algorithms are you interested in improving performance for? Based on that feedback I can make a recommendation for Raspberry Pi 4 build options to use.

Thanks,
David Garske, wolfSSL

SparkiDev, 2024-11-18:
Hi @vifino,

Unfortunately we don't support assembly for Aarch64 armv8-a. That is, there is only assembly when the crypto extensions are available.

Is it possible to have different builds for different versions of ARMv8-A?

Sean

vifino, 2024-12-03:
Hey @SparkiDev,

it's complicated. While there is the possibility of building the Nix derivations targeting armv8-a+crypto or similar, those are rarely used on anything but Darwin/Apple Silicon, as the supported linux target is plain `armv8-a`.

So, realistically, the answer is "no" - 99% of `armv8-a+crypto` capable Linux machines will only run the build that has the `armv8-a` baseline.

SparkiDev, 2024-12-03:
Hi @vifino,

I understand.
I have started working on a solution for this.
Basically, I'm making sure that there is a software implementation used when the cryptographic instructions are not available.

I'll let you know when I have a PR for this work.

Sean

SparkiDev, 2024-12-12:
Hi @vifino

There is a PR up that supports builds for Aarch64 when hardware crypto instructions are not available: #8277 
You can specify WOLFSSL_ARMASM_NO_HW_CRYPTO and it will use a software implementation.

Sean

SparkiDev, 2024-12-16:
Hi @vifino

The PR has been merged.

Let us know id this fixes your issue.

Thanks,
Sean

vifino, 2024-12-21:
Hey @SparkiDev, apologies for the late response.

I've tried the current master now that #8293 is merged. I appreciate the effort you all put in!

Problem is, I'm running into the same build failure with LTO as before. I've tried with `--enable-armasm=yes` and `--enable-armasm=inline`.
The only way I've got it to build is by using `-march=armv8-a+crypto` which results in the compiler emitting instructions that won't run, including the unit tests crashing because of the illegal instructions.

How can I get the linker to not complain without gcc emitting the instructions on its own?

dgarske, 2024-12-21:
Hi @vifino ,

Did you try adding CFLAGS=-DWOLFSSL_ARMASM_NO_HW_CRYPTO to your configure also? It will disable the ARM crypto assembly instructions. 

Thanks,
David Garske, wolfSSL

vifino, 2024-12-21:
Hi @dgarske,

I did not, but that seems to be missing most of the potential of that patch.

Given that it added CPUID gating of the instructions, that's what I'm aiming for, as that's the most ideal situation for packaging.

I believe the only missing step is to get the linker to accept the additional instructions without making the actual compiler emit the +crypto instructions.

SparkiDev, 2024-12-22:
Hi @vifino,

If __linux__ is defined, then the OS is checked for CPU features.
Let us know which functions are crashing so we can make sure we are using the CPU features correctly.

Thanks,
Sean

dgarske, 2024-12-23:
@SparkiDev here are my results with this PR on a Pi4 (this HW does not have crypto extensions). FYI: Works fine with `WOLFSSL_ARMASM_NO_HW_CRYPTO`

```
./configure '--enable-armasm' '--enable-all' '--disable-crl-monitor' 'CFLAGS=-DWOLFSSL_AARCH64_PRIVILEGE_MODE' '--enable-debug' '--disable-shared' && make
gdb ./wolfcrypt/test/testwolfcrypt
...
asn      test passed!
wolfSSL Entering random_test

Program received signal SIGILL, Illegal instruction.
0x00000055555caf60 in Sha256Final (sha256=0x7fffffee90, hash=0x7fffffee70 "") at wolfcrypt/src/port/arm/armv8-sha256.c:564
564	    __asm__ volatile (
(gdb) bt
#0  0x00000055555caf60 in Sha256Final (sha256=0x7fffffee90, hash=0x7fffffee70 "") at wolfcrypt/src/port/arm/armv8-sha256.c:564
#1  0x00000055555cb35c in wc_Sha256Final (sha256=0x7fffffee90, hash=0x7fffffee70 "") at wolfcrypt/src/port/arm/armv8-sha256.c:1830
#2  0x00000055555c8e18 in Hash_df (drbg=0x7fffffefd8, out=0x7fffffefdc "", outSz=55, type=4 '\004',
    inA=0x7ffffff2b0 "\246Z\320\363E\333N\016\377\350uâ\347\037B\307\022\235b\017\365\301\031\251\357U\360Q\205\340\373\205\201\371\061u\027'n\006\351`}\333\313\314.\260\246\202UU", inASz=48, inB=0x0, inBSz=0)
    at wolfcrypt/src/random.c:408
#3  0x00000055555c9608 in Hash_DRBG_Instantiate (drbg=0x7fffffefd8,
    seed=0x7ffffff2b0 "\246Z\320\363E\333N\016\377\350uâ\347\037B\307\022\235b\017\365\301\031\251\357U\360Q\205\340\373\205\201\371\061u\027'n\006\351`}\333\313\314.\260\246\202UU", seedSz=48, nonce=0x0,
    nonceSz=0, heap=0x0, devId=-2) at wolfcrypt/src/random.c:716
#4  0x00000055555ca0a0 in wc_RNG_HealthTest_ex (reseed=0, nonce=0x0, nonceSz=0,
    seedA=0x7ffffff2b0 "\246Z\320\363E\333N\016\377\350uâ\347\037B\307\022\235b\017\365\301\031\251\357U\360Q\205\340\373\205\201\371\061u\027'n\006\351`}\333\313\314.\260\246\202UU", seedASz=48, seedB=0x0,
    seedBSz=0, output=0x7ffffff0e0 "\200\361\377\377\177", outputSz=128, heap=0x0, devId=-2) at wolfcrypt/src/random.c:2085
#5  0x00000055555c9fe8 in wc_RNG_HealthTest (reseed=0,
    seedA=0x7ffffff2b0 "\246Z\320\363E\333N\016\377\350uâ\347\037B\307\022\235b\017\365\301\031\251\357U\360Q\205\340\373\205\201\371\061u\027'n\006\351`}\333\313\314.\260\246\202UU", seedASz=48, seedB=0x0,
    seedBSz=0, output=0x7ffffff0e0 "\200\361\377\377\177", outputSz=128) at wolfcrypt/src/random.c:2044
#6  0x000000555557b680 in random_test () at wolfcrypt/test/test.c:17869
#7  0x000000555555c890 in wolfcrypt_test (args=0x7ffffff340) at wolfcrypt/test/test.c:1626
#8  0x000000555555e194 in wolfcrypt_test_main (argc=1, argv=0x7ffffff4c8) at wolfcrypt/test/test.c:2609
#9  0x000000555555e0c4 in main (argc=1, argv=0x7ffffff4c8) at wolfcrypt/test/test.c:2517```

dgarske, 2024-12-24:
FYI: @SparkiDev is making progress on resolving this. The SHA2/SHA3 runtime CPUID detection was resolved for AARCH64. Next he will work on an AES GCM issue.
https://github.com/wolfSSL/wolfssl/pull/8314

dgarske, 2025-01-02:
Hi @vifino , the final piece for getting this working... https://github.com/wolfSSL/wolfssl/pull/8325 
Looking forward to your feedback

vifino, 2025-01-02:
Hey @dgarske, that's awesome news!

What's the arch set to in your Pi testing, armv8-a or armv8-a+crypto? Does it work with `-flto` for you?

I'll check it out tomorrow myself.
Unfortunate that the new release won't have this change.

dgarske, 2025-01-02:
Hi @vifino ,

I compiled it on the Pi4 and it defaults to including crypto extensions (`armv8-a+crypto`). I just tried with `-flto` and it builds and all tests pass. Looking forward to your feedback tomorrow.

That is correct about the release. We could not get that work completed in time for the release. You would need to use v5.7.6 + PR 8325.

Thanks,
David Garske, wolfSSL

dgarske, 2025-01-30:
Hi @vifino ,

I believe this issue has been fully resolved, but it would be great to get your feedback before closing this issue. Do you have any updates?

Thanks,
David Garske, wolfSSL

vifino, 2025-02-03:
Hey @dgarske,

apologies for the late response, extraordinarily busy the past few weeks.

I've applied PR 8325 on top of 5.7.6 and tested it on x86 and aarch64 hosts with crypto extensions.

On the ARM machine, if I `--enable-armasm=yes`, the unit tests fail. Without, they pass like usual.

The unit test fails in a very... interesting and verbose way: https://gist.github.com/vifino/35f705d2da913c8b0a505d9251147155

Hope this helps!

SparkiDev, 2025-02-04:
Hi @vifino,

I've tried to reproduce this by checking out the v5.7.6-stable tag and applying the PR 8345 patch on top and running with QEMU.
I don't have a failure.

In the verbose test output I don't see any failure either.
There should be a 'FAIL' line at the end when unit.test fails in the client/server testing.
Is this all the output?

Which Aarch64 CPU are you testing on?

Sean

vifino, 2025-02-04:
Hey @SparkiDev,

the Gist is much longer than GitHub shows by default, you have to [view the full file](https://gist.github.com/vifino/35f705d2da913c8b0a505d9251147155/raw/0739d919174129af165dcb9b4141c71553c77700/wolfssl-aarch64-test-fail.log).

That is running on an Ampere CPU in the Oracle Cloud - `/proc/cpuid` doesn't look very detailed however:
```
processor       : 0
BogoMIPS        : 50.00
Features        : fp asimd evtstrm aes pmull sha1 sha2 crc32 atomics fphp asimdhp cpuid asimdrdm lrcpc dcpop asimddp
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x3
CPU part        : 0xd0c
CPU revision    : 1
```

Hope this helps.

SparkiDev, 2025-02-07:
Hi @vifino 

Please try applying the patch in this PR: https://github.com/wolfSSL/wolfssl/pull/8275
It fixed a corner case in  Poly1305 assembly code for Aarch64.

Thanks,
Sean

vifino, 2025-02-08:
Hey @SparkiDev.

That commit is already in 5.7.6-stable, which I am using, plus PR 8325 on top.

dgarske, 2025-02-10:
Hi @vifino ,

Is the `scripts/unit.test` failure consistently failing for you in the same place? The error seems to be an issue with a non-blocking timeout on one of the test cases.

```
Early Data was not sent.
Session Ticket CB: ticketSz = 211, ctx = initiNonblocking read timeout
wolfSSL error: ClientRead failed
al session
```

I'm wondering if this error is not related to the aarch64 with LTO optimizations?

Thanks,
David Garske, wolfSSL

vifino, 2025-02-10:
Hi @dgarske!

I haven't tested if it fails at the same point, but it always fails when I have the 5.7.6-stable + PR 8325 applied and "--enable-armasm".
When I do not enable armasm, the unit tests pass, so I'm sure it is related.

The issue title is kinda outdated by now, yes, but the CPUID detection for arm64 is the fix for this issue, but as mentioned, I can't get the tests to pass.

SparkiDev, 2025-02-11:
Hi @vifino,

I can't think of anything different about the assembly code other than it should be faster and it may use more RAM.
Neither of these seem to match the situation you are seeing a failure.

In case of a memory issue, could you please try disabling SP, and leaving ARM asm on.
That is remove: --enable-sp=yes,asm' '--enable-sp-math-all'

Thanks,
Sean

dgarske, 2025-03-20:
Hi@vifino , are you able to share any details about the crash or the silicon you are running on? If you use `--enable-armasm CFLAGS="-DWOLFSSL_ARMASM_NO_HW_CRYPTO"` does it compile and work? Thanks, David Garske, wolfSSL

dgarske, 2025-04-03:
> Hey [@dgarske](https://github.com/dgarske), the details you are asking for are in an earlier comment [here](https://github.com/wolfSSL/wolfssl/issues/8148#issuecomment-2633189204), I can't provide much else.
> 
> When defining `WOLFSSL_ARMASM_NO_HW_CRYPTO`, the unit tests still fail. [Here](https://gist.githubusercontent.com/vifino/63f2b45fcfa41ae2d08b4f3a2653314f/raw/b1f934743805f5c3124c31e14555deafb099e127/wolfssl-aarch64-test-fail-no-hw-crypto.log) is the entire log.
> 
> Running the current master ([a8384bb](https://github.com/wolfSSL/wolfssl/commit/a8384bb42667ef3b764615a36bd45b7cad315f16)) instead of 5.7.6-stable + PR 8325 is unsuccessful as well, perhaps even less so, given the segfaulting test driver, see [here](https://gist.githubusercontent.com/vifino/1c69345af5f70a09431ae911032fbbe5/raw/f0466aa88610429e03804c6a7bc648c7668730e7/wolfssl-aarch64-master-test-fail-no-hw-crypto.log).

@SparkiDev , Any ideas?

vifino, 2025-05-02:
I have tried again with 5.8.0. Same issues.

In fact, even without LTO and with `--disable-armasm` and `WOLFSSL_EXTERNAL_TEST=0`, the testsuite's test driver segfaults.

I'd be willing to hop on a call (CEST) and share a terminal session into the Oracle Cloud ARM machine.
I'm willing to walk you through my test setup and packaging with Nix.

Having broken unit tests results in wolfssl not being upgraded for NixOS users, basically.

vifino, 2025-10-28:
Hey @dgarske,

I've successfully ran the testsuite on an aarch64 host **with crypto extensions**, with `--enable-armasm=yes` and with `--disable-armasm`, with and without `-DWOLFSSL_ARMASM_NO_HW_CRYPTO`. Progress!

However, the testsuite fails on a host **without crypto extensions** - Raspberry Pi 4 - with `--enable-armasm=yes`: [Here is the Log](https://gist.github.com/vifino/add8022c95dfbae6cdbcfa410a065d7c/raw/89acfea2a752f042813f7582a3af27ca70f5e57f/wolfssl-aarch64-master-pi4-test-fail.log)

vifino, 2025-11-08:
Hey @SparkiDev,

I can confirm that:
- the build is supposed to be generic (ie. run with or without crypto instructions)
- the device that is failing the testsuite is a Raspberry Pi 4 which does not support the crypto instructions
- `WOLFSSL_ARMASM_NO_HW_CRYPTO` is not defined
- the OS is a glibc-based Linux - NixOS specifically - where I am sure `__linux__` will be defined

Our (NixOS) build process runs the test suite during build, is the test suite not the most suited for debugging this issue?
I can also skip the test suite and run the benchmark, if you think it will help more than the build and test log I've linked above.

vifino, 2025-12-14:
Hi!
I've tried to get both going on the latest master, but they rather immediately segfault.

Benchmark (no options):
```
------------------------------------------------------------------------------
 wolfSSL version 5.8.4
------------------------------------------------------------------------------
Clock frequency: 54000000 Hz
wolfCrypt Benchmark (block bytes 1048576, min 1.0 sec each)
/nix/store/pm4wbkdadhx5gjz435cyjm646kzp6ddi-stdenv-linux/setup: line 1757: 14889 Illegal instruction        (core dumped) ./wolfcrypt/benchmark/benchmark
```

sha256 unit tests:
```
starting unit tests...
 Begin API Tests
/nix/store/pm4wbkdadhx5gjz435cyjm646kzp6ddi-stdenv-linux/setup: line 1760: 14914 Illegal instruction        (core dumped) ./tests/unit.test --group sha256
```

Unfortunately, it is really hard to do more in-depth tests within the nix environment.
I'll try to set up some Debian environment on a Pi 4 and try more.