wolfSSL/wolfssl #8152: WOLFSSL_SYS_CA_CERTS incorrectly disabled when WOLFSSL_FILESYSTEM=OFF

redbaron, 2024-11-06:
### Version

5.7.4

### Description

When compiling with:

`cmake -DWOLFSSL_SYS_CA_CERTS=ON -DWOLFSSL_FILESYSTEM=OFF` it incorrectly disables SYS_CA feature even on platforms where system CA are accessed without invoking any file API (Apple and Windows) .

kareem-wolfssl, 2024-11-07:
Hi redbaron,

It looks like we are tying NO_FILESYSTEM to WOLFSSL_SYS_CA_CERTS specifically for wolfSSL_CTX_set_default_verify_paths, which does require a filesystem.  But it looks like the other system CA cert APIs could still work without a filesystem, I will continue looking into this to confirm.
Can you elaborate on your use case?  Are you building for a Windows/Apple system that has no filesystem, or are you just wanting to disable our filesystem support despite having a filesystem?

redbaron, 2024-11-07:
Yes, we build wolfs on platforms where there is no direct filesystem access, but would like to build it with similar flags across all of them , because no FS access features are  required.

There is a `wolfSSL_CTX_load_system_CA_certs` which would be good to have working on platforms where it can work without changing wolfs configuration options so that it is built the same across all platforms.

dgarske, 2025-12-23:
Hi @redbaron , we put up a PR #9569 to resolve this. Please let us know if you have any issues. Thanks, David Garske, wolfSSL