wolfSSL/wolfssl #8165: CMake buid scripts don't offer `--enable-keylog-export` like option

redbaron, 2024-11-08:
### Version

5.7.4

### Description

configure script has `--enable-keylog-export` param which defines `WOLFSSL_SSLKEYLOGFILE`. This feature doesn't seem to be supported by CMake build scripts.

dgarske, 2024-11-08:
@bigbrett can you review this bug report and evaluate effort to fix?

bigbrett, 2024-11-08:
@redbaron Thanks for pointing that out, we can add it in.

In the meantime, if you ensure the C macros `-DSHOW_SECRETS -DHAVE_SECRET_CALLBACK -DWOLFSSL_SSLKEYLOGFILE -DWOLFSSL_KEYLOG_EXPORT_WARNED` are defined, it will have the equivalent behavior.

redbaron, 2024-11-11:
@bigbrett , yeah passing `-DWOLFSSL_DEFINITIONS` with list of defines does the trick nicely in all cases so far, but first class CMake support will help discovering these features easier.

In general it seems that autotools has more love: 148 vs 34 commits this year so far.  

```
$ git log  CMakeLists.txt cmake |grep -F ' 2024 ' | wc -l
34

$ git log  configure.ac |grep -F ' 2024 ' | wc -l  
148
```

It is probably because autotools is more popular among embedded devs, but your build [instructions](https://www.wolfssl.com/documentation/manuals/wolfssl/chapter02.html) don't mention CMake either :(

bigbrett, 2024-11-11:
@redbaron Yep I agree, our CMake coverage does often slip behind. wolfSSL has a long history with autotools, and for better or for worse, that is our default supported build system. So we do rely on the community for issues like this in keeping our CMake support up to date. Community PRs for CMake support are always appreciated :)

redbaron, 2024-11-11:
@bigbrett ,

> Community PRs for CMake support are always appreciated :)

took opportunity to add more ergonomic helper to define build option: https://github.com/wolfSSL/wolfssl/pull/8174

PR #8596 (2025-03-27): Various improvements to iscacii and CMake key log
# Description

Various improvements to iscacii and CMake key log:
* Detect 'isascii' at configuration (tested with `./configure CFLAGS="-DNO_STDLIB_ISASCII" && make check`).
* Add mew CMake option `WOLFSSL_KEYLOG_EXPORT` (fixes #8165) 
Replaces PR #8174 and #8158. Thank you @redbaron.

# Testing

```
./configure CFLAGS="-DNO_STDLIB_ISASCII" && make check
mkdir build && cd build && cmake .. -DWOLFSSL_KEYLOG_EXPORT=YES && make
```

# Checklist

 - [ ] added tests
 - [ ] updated/added doxygen
 - [ ] updated appropriate READMEs
 - [ ] Updated manua [... cut]
commit a59075b9 (2025-03-27): Various improvements to iscacii and CMake key log:
* Detect 'isascii' at configuration (tested with `./configure CFLAGS="-DNO_STDLIB_ISASCII" && make check`).
* Add mew CMake option `WOLFSSL_KEYLOG_EXPORT` (fixes #8165)
Replaces PR #8174 and #8158. Thank you @redbaron.
closed 2025-03-27 by SparkiDev