wolfSSL/wolfssl #8190: [Bug]: failed build on Fedora 40 with --enable-distro

sergey-safarov, 2024-11-15:
### Contact Details

s.safarov@gmail.com

### Version

5.7.4

### Description

```
  CC       examples/pem/pem.o
  CC       testsuite/testsuite_test-testsuite.o
  CC       tests/unit_test-unit.o
  CC       tests/unit_test-api.o
  CC       tests/unit_test-suites.o
  CC       tests/unit_test-hash.o
  CC       tests/unit_test-w64wrapper.o
  CC       tests/unit_test-srp.o
  CC       tests/unit_test-quic.o
  CCLD     src/libwolfssl.la
/usr/bin/ld: wolfcrypt/src/.libs/src_libwolfssl_la-ecc.o: unable to decompress section .gnu.debuglto_.debug_str
/usr/bin/ld: /tmp/ccxvhUPH.ltrans5.ltrans.o: in function `wolfCrypt_Init':
/root/rpmbuild/BUILD/wolfssl-5.7.4/wolfcrypt/src/wc_port.c:370:(.text+0x5e41): undefined reference to `wc_ecc_oid_cache_init'
/usr/bin/ld: /tmp/ccxvhUPH.ltrans5.ltrans.o: in function `wolfCrypt_Cleanup':
/root/rpmbuild/BUILD/wolfssl-5.7.4/wolfcrypt/src/wc_port.c:468:(.text+0x5e95): undefined reference to `wc_ecc_oid_cache_free'
/usr/bin/ld: /tmp/ccxvhUPH.ltrans19.ltrans.o: in function `ec_point_convert_to_affine':
/root/rpmbuild/BUILD/wolfssl-5.7.4/src/pk.c:10739:(.text+0x93e2): undefined reference to `ecc_map'
/usr/bin/ld: /tmp/ccxvhUPH.ltrans19.ltrans.o: in function `wolfssl_ec_point_add':
/root/rpmbuild/BUILD/wolfssl-5.7.4/src/pk.c:11040:(.text+0x9cb5): undefined reference to `ecc_projective_add_point'
/usr/bin/ld: /root/rpmbuild/BUILD/wolfssl-5.7.4/src/pk.c:11047:(.text+0x9cd9): undefined reference to `ecc_map'
/usr/bin/ld: /tmp/ccxvhUPH.ltrans7.ltrans.o: in function `EccSpecifiedECDomainDecode.lto_priv.0':
/root/rpmbuild/BUILD/wolfssl-5.7.4/wolfcrypt/src/asn.c:34137:(.text+0x5d01): undefined reference to `wc_ecc_free_curve'
/usr/bin/ld: /tmp/ccxvhUPH.ltrans20.ltrans.o: in function `wolfssl_ec_point_mul':
/root/rpmbuild/BUILD/wolfssl-5.7.4/src/pk.c:11151:(.text+0x2de5): undefined reference to `ecc_mul2add'
/usr/bin/ld: /tmp/ccxvhUPH.ltrans29.ltrans.o: in function `eccsi_mulmod_base_add.constprop.0':
/root/rpmbuild/BUILD/wolfssl-5.7.4/wolfcrypt/src/eccsi.c:1375:(.text+0x26e0): undefined reference to `ecc_projective_add_point'
/usr/bin/ld: /root/rpmbuild/BUILD/wolfssl-5.7.4/wolfcrypt/src/eccsi.c:1379:(.text+0x2700): undefined reference to `ecc_map'
collect2: error: ld returned 1 exit status
make[2]: *** [Makefile:6342: src/libwolfssl.la] Error 1
make[2]: *** Waiting for unfinished jobs....
make[2]: Leaving directory '/root/rpmbuild/BUILD/wolfssl-5.7.4'
make[1]: *** [Makefile:9026: all-recursive] Error 1
make[1]: Leaving directory '/root/rpmbuild/BUILD/wolfssl-5.7.4'
make: *** [Makefile:5376: all] Error 2
error: Bad exit status from /var/tmp/rpm-tmp.iP7so5 (%build)

RPM build errors:
    Bad exit status from /var/tmp/rpm-tmp.iP7so5 (%build)
```

### Reproduction steps

./configure --enable-distro
make rpm
mkdir -p ~/rpmbuild/SOURCES/
rpmbuild -bb --undefine=_disable_source_fetch rpm/spec

philljj, 2024-11-18:
Hi @sergey-safarov,

Thank you for the report. I found the same issue on Fedoras 39 and 40 (interestingly, the EOL Fedora 38 still worked!).

I'm suspicious this is a Fedora toolchain mismatch issue. I could circumvent the problem by using our docker rpm build (`make rpm-docker`) instead:

```
19. Building for RHEL, Fedora, CentOS, SUSE, and openSUSE

    To generate a .rpm package, configure wolfSSL with the desired
    configuration. Then run `make rpm` to generate a .rpm package
    with the current configuration. To build the package inside a
    Docker container, use `make rpm-docker`. In both cases the
    resulting packages are placed in the root directory of the
    project.
```

I would recommend trying this for now.

Let me know if this helps.
Best,
Jordan

philljj, 2024-11-18:
Hi @sergey-safarov,

I upgraded my system to Fedora 41 and this resolved the native toolchain rpm build issue.

I would recommend either using the docker rpm build method, or to try updating your system and see if that resolves your build issue.

Best,
Jordan

sergey-safarov, 2024-11-19:
I can confirm that using Fedora 41 rpm is packaged properly.

closed 2024-11-19 by sergey-safarov