wolfSSL/wolfssl #8199: Issue with DTLS 1.3 Server handshake

EscaVic, 2024-11-18:
### Version

v5.7.2-stable

### Description

Hi All,
I'm new to WolfSSL and I am replacing OpenSSL with WolfSSL for DTLS 1.3
I'm see the following using self-signed certs that worked for our OpenSSL Build

Can I get clarification on these statements - seems to occur after the handshake, 

wolfSSL Entering EmbedReceiveFrom
wolfSSL Entering wolfSSL_dtls_get_using_nonblock
Embed Receive From error
	Would block

The extended sequence is below

wolfSSL Entering DTLSv1_3_server_method_ex
wolfSSL Entering wolfSSL_CTX_new_ex
wolfSSL Entering wolfSSL_CertManagerNew
heap param is null
DYNAMIC_TYPE_CERT_MANAGER Allocating = 368 bytes
wolfSSL Entering wolfSSL_CTX_set_verify
wolfSSL Entering wolfSSL_CTX_set_mode
wolfSSL Entering wolfSSL_CTX_set_min_proto_version
wolfSSL Entering wolfSSL_CTX_get_max_proto_version
wolfSSL Entering wolfSSL_CTX_get_options
wolfSSL options are set through API calls and macros
wolfSSL Leaving wolfSSL_CTX_get_max_proto_version, return 772
wolfSSL Entering wolfSSL_CTX_clear_options
wolfSSL Entering wolfSSL_CTX_set_min_proto_version_ex
wolfSSL Entering wolfSSL_CTX_allow_anon_cipher
wolfSSL Leaving wolfSSL_CTX_new_ex, return 0
wolfSSL Entering wolfSSL_CTX_use_PrivateKey_file
wolfSSL Entering ProcessBuffer
wolfSSL Entering PemToDer
wolfSSL Leaving ProcessBuffer, return 1
wolfSSL Entering wolfSSL_CTX_use_certificate_file
wolfSSL Entering ProcessBuffer
wolfSSL Entering PemToDer
Checking cert signature type
Getting Cert Name
wolfSSL Entering wolfSSL_X509_NAME_new_ex
wolfSSL Entering wolfSSL_X509_NAME_add_entry_by_NID
Found place for name entry
wolfSSL Entering wolfSSL_sk_X509_NAME_new
wolfSSL Entering wolfSSL_sk_new_node
wolfSSL Entering wolfSSL_sk_push
wolfSSL Entering wolfSSL_X509_NAME_ENTRY_free
wolfSSL Entering wolfSSL_X509_NAME_add_entry_by_NID
Found place for name entry
wolfSSL Entering wolfSSL_sk_push
wolfSSL Entering wolfSSL_sk_new_node
wolfSSL Entering wolfSSL_X509_NAME_ENTRY_free
wolfSSL Entering wolfSSL_X509_NAME_add_entry_by_NID
Found place for name entry
wolfSSL Entering wolfSSL_sk_push
wolfSSL Entering wolfSSL_sk_new_node
wolfSSL Entering wolfSSL_X509_NAME_ENTRY_free
wolfSSL Entering wolfSSL_X509_NAME_add_entry_by_NID
Found place for name entry
wolfSSL Entering wolfSSL_sk_push
wolfSSL Entering wolfSSL_sk_new_node
wolfSSL Entering wolfSSL_X509_NAME_ENTRY_free
wolfSSL Entering wolfSSL_X509_NAME_add_entry_by_NID
Found place for name entry
wolfSSL Entering wolfSSL_sk_push
wolfSSL Entering wolfSSL_sk_new_node
[... 335 lines cut ...]
wolfSSL Entering DtlsMsgPoolTimeout
wolfSSL Leaving DtlsMsgPoolTimeout(), return 0
wolfSSL Entering Dtls13RtxSendBuffered
wolfSSL Leaving wolfSSL_dtls_get_current_timeout, return 32
wolfSSL Entering EmbedReceiveFrom
wolfSSL Entering wolfSSL_dtls_get_using_nonblock
Embed Receive From error
	Would block
wolfSSL Entering wolfSSL_dtls_get_using_nonblock
wolfSSL Entering DtlsMsgPoolTimeout
wolfSSL Leaving DtlsMsgPoolTimeout(), return 0
wolfSSL Entering Dtls13RtxSendBuffered
wolfSSL Leaving wolfSSL_dtls_get_current_timeout, return 64
wolfSSL Entering EmbedReceiveFrom
wolfSSL Entering wolfSSL_dtls_get_using_nonblock
Embed Receive From error
	Would block
wolfSSL Entering wolfSSL_dtls_get_using_nonblock
wolfSSL Entering DtlsMsgPoolTimeout
wolfSSL Leaving DtlsMsgPoolTimeout(), return -1
Error trying to retransmit DTLS buffered message
wolfSSL error occurred, error = 308 line:20837 file:external/wolfssl/src/intern
wolfSSL error occurred, error = 308 line:9844 file:external/wolfssl/src/ssl.c
wolfSSL Entering wolfSSL_get_error
wolfSSL Leaving wolfSSL_get_error, return -308

anhu, 2024-11-18:
Hi @EscaVic ,
My name is Anthony and I am a member of the wolfSSL team.  I will be delegating this to one of our team members.  In the meantime can you tell us a bit about yourself and your project? For example: 

- where are you located? 
- is this out of personal, academic or professional interest? 
- are there any organizations associated with this effort? 
- any other information that you care to share. 

Warm regards, Anthony

EscaVic, 2024-11-18:
Hi Anthony,

Thanks for responding quickly.

- where are you located?
    Foster City CA
- is this out of personal, academic or professional interest?
    Professional
- are there any organizations associated with this effort?
    Zoox, https://zoox.com/about
- any other information that you care to share.

anhu, 2024-11-18:
Hi, Can you please send a message to support@wolfssl.com referencing this ticket.  You should be more comfortable if we handle professional questions via our private support window. 

Warm regards, Anthony

closed 2025-01-15 by julek-wolfssl