wolfSSL/wolfssl #8203: Wolfssl FIPS   (library)  hash update issue

volga629-1, 2024-11-20:
### Version

5.7.4 FIPS

### Description

Configure 

```
 ./configure --host=x86_64-w64-mingw32 --enable-reproducible-build --enable-keygen --enable-rsapss \
			--enable-secure-renegotiation --enable-fastmath \
			--enable-ed25519 --enable-curve25519 \
			--enable-debug \
			--enable-fips=ready \
			--enable-opensslall \
			--enable-ecc \
			--enable-ocsp \
			--enable-crl \
			--enable-psk \
			--disable-fpecc \
			--disable-aligndata \
			--disable-jni \
			--disable-crl-monitor\
			--disable-examples \
			LDFLAGS="-lws2_32 -lcrypt32 -Wl,-s -Wl,--gc-sections"
```

Issue

I follow the process and compiled wolfssl fips multiply times with different variations on windows and  library not working working correct . When I tried to start strongswan it generate error -203 

Here forum thread which provide full information what was done before 

```
https://www.wolfssl.com/forums/post8085.html#p8085
```

Strongswan Start up with wolfssl dynamic library 

```
$ ldd charon-svc.exe
        ntdll.dll => /c/Windows/SYSTEM32/ntdll.dll (0x7ffb75230000)
        KERNEL32.DLL => /c/Windows/System32/KERNEL32.DLL (0x7ffb73c10000)
        KERNELBASE.dll => /c/Windows/System32/KERNELBASE.dll (0x7ffb726b0000)
        ADVAPI32.dll => /c/Windows/System32/ADVAPI32.dll (0x7ffb73890000)
        msvcrt.dll => /c/Windows/System32/msvcrt.dll (0x7ffb73e70000)
        sechost.dll => /c/Windows/System32/sechost.dll (0x7ffb73530000)
        bcrypt.dll => /c/Windows/System32/bcrypt.dll (0x7ffb72c10000)
        RPCRT4.dll => /c/Windows/System32/RPCRT4.dll (0x7ffb73720000)
        WS2_32.dll => /c/Windows/System32/WS2_32.dll (0x7ffb739b0000)
        fwpuclnt.dll => /c/Windows/SYSTEM32/fwpuclnt.dll (0x7ffb6f0e0000)
        IPHLPAPI.DLL => /c/Windows/SYSTEM32/IPHLPAPI.DLL (0x7ffb70d10000)
        libwolfssl-42.dll => /home/slava.bendersky/strongswan-5.9.14/strongswan-sec/libwolfssl-42.dll (0x7ffb148f0000)
        CRYPT32.dll => /c/Windows/System32/CRYPT32.dll (0x7ffb72540000)
        ucrtbase.dll => /c/Windows/System32/ucrtbase.dll (0x7ffb72d00000)
        WINHTTP.dll => /c/Windows/SYSTEM32/WINHTTP.dll (0x7ffb6e9c0000)
        libgcc_s_seh-1.dll => /mingw64/bin/libgcc_s_seh-1.dll (0x7ffb485b0000)
        libdl.dll => /mingw64/bin/libdl.dll (0x7ffb4b4d0000)
        libunbound-8.dll => /mingw64/bin/libunbound-8.dll (0x7ffb14050000)
        libldns-3.dll => /mingw64/bin/libldns-3.dll (0x7ffb335f0000)
        libwinpthread-1.dll => /mingw64/bin/libwinpthread-1.dll (0x7ffb43380000)
        libssl-3-x64.dll => /mingw64/bin/libssl-3-x64.dll (0x7ffb21d30000)
        libcrypto-3-x64.dll => /mingw64/bin/libcrypto-3-x64.dll (0x26be2410000)
        libcrypto-3-x64.dll => /mingw64/bin/libcrypto-3-x64.dll (0x26be28f0000)
        libcrypto-3-x64.dll => /mingw64/bin/libcrypto-3-x64.dll (0x7ffb05a80000)
        USER32.dll => /c/Windows/System32/USER32.dll (0x7ffb73a30000)
        win32u.dll => /c/Windows/System32/win32u.dll (0x7ffb72e20000)
        GDI32.dll => /c/Windows/System32/GDI32.dll (0x7ffb73be0000)
        gdi32full.dll => /c/Windows/System32/gdi32full.dll (0x7ffb72af0000)
        msvcp_win.dll => /c/Windows/System32/msvcp_win.dll (0x7ffb72ec0000)

volg629@Desktop1 MSYS ~/strongswan-5.9.14/strongswan-sec
$ ./charon-svc.exe
Starting Power On Self Test
Pre-Operational Self Test FAILURE
00[DMN] Starting IKE service charon-svc (strongSwan 5.9.14, Windows Client 10.0.22621 (SP 0.0))
00[LIB] wolfssl FIPS mode unavailable (-203)
00[LIB] plugin 'wolfssl': failed to load - wolfssl_plugin_create returned NULL
00[LIB] feature CUSTOM:libcharon in critical plugin 'charon-svc' has unmet dependency: NONCE_GEN
00[LIB] feature CUSTOM:libcharon-receiver in critical plugin 'charon-svc' has unmet dependency: HASHER:HASH_SHA1
00[LIB] feature CUSTOM:libcharon-sa-managers in critical plugin 'charon-svc' has unmet dependency: HASHER:HASH_SHA1
00[CFG] failed to read the resolver config: error reading file (No such file or directory)
00[CFG] failed to create a DNS resolver instance
00[LIB] failed to load 3 critical plugin features

```

Wolfssl crypt  produce error when try generate hash 

```
$ ./wolfcrypt/test/testwolfcrypt
------------------------------------------------------------------------------
 wolfSSL version 5.7.4
------------------------------------------------------------------------------
FIPS module version in use: wolfCrypt v7.0.0
error    test passed!
wolfSSL Entering memory_test
MEMORY   test passed!
wolfSSL Entering base64_test
Bad Base64 Decode data, too small
Bad Base64 Decode data, too small
Bad Base64 Decode data, too big
Bad Base64 Decode data, too small
Bad Base64 Decode data, too big
Bad Base64 Decode data, too small
Bad Base64 Decode data, too big
Bad Base64 Decode data, too small
Bad Base64 Decode data, too big
Bad Base64 Decode data, too small
Bad Base64 Decode data, too small
Bad Base64 Decode data, too small
Bad Base64 Decode data, too small
Bad Base64 Decode data, too small
Bad Base64 Decode data, too small
Bad Base64 Decode data, too small
Bad Base64 Decode data, too small
Bad Base64 Decode data, too small
Bad Base64 Decode data, too small
Bad Base64 Decode data, too small
Bad Base64 Decode data, too small
Bad end of line in Base64 Decode
Bad Base64 Decode data, too big
Bad Base64 Decode data, too big
Bad Base64 Decode data, too big
Bad Base64 Decode data, too big
Bad Base64 Decode data, too big
Bad Base64 Decode data, too big
Bad Base64 Decode data, too big
Escape buffer max too small
base64   test passed!
wolfSSL Entering base16_test
base16   test passed!
wolfSSL Entering asn_test
asn      test passed!
wolfSSL Entering random_test
in my Fips callback, ok = 0, err = -197
message = FIPS mode not allowed error
hash =
RANDOM   test failed!
 error L=17866 code=-197 (FIPS mode not allowed error)
 [fiducial line numbers: 9103 28041 46740 59294]
wolfSSL Entering wolfCrypt_Cleanup
Exiting main with return code: -1

```

or this 

```
Escape buffer max too small
base64   test passed!
wolfSSL Entering base16_test
base16   test passed!
wolfSSL Entering asn_test
asn      test passed!
wolfSSL Entering random_test
in my Fips callback, ok = 0, err = -203
message = In Core Integrity check FIPS error
hash = 11FC92013108BCB799AF1141F7BE8EB3E314240A8985736469BBDC33D5A94A0C
In core integrity hash check failure, copy above hash
into verifyCore[] in fips_test.c and rebuild
RANDOM   test failed!
 error L=17866 code=-197 (FIPS mode not allowed error)
 [fiducial line numbers: 9103 28041 46740 59294]
wolfSSL Entering wolfCrypt_Cleanup
Exiting main with return code: -1
```

anhu, 2024-11-20:
Hi @volga629-1 , 

This is the procedure: 
```
unzip wolfssl-5.7.2-gplv3-fips-ready.zip
cd wolfssl-5.7.2-gplv3-fips-ready
./configure --enable-fips=ready
make
./fips_hash.sh
make
make check 
```
You should probably try it on Linux as a sanity check.  Then once you have it working with strongswan on linux, move to windows. 

Here at wolfSSL we love knowing how people are using our software source code.  Can you let us know a bit about yourself and this project?  For example: 

- where are you located? 
- what are your goals? 
- is this out of personal, academic or professional interest? 

Please do let us know. 

Warm regards, Anthony

volga629-1, 2024-11-20:
I can reproduce same  problem on Linux Fedora 39 
I compiled same way  just generated RPM's.

```

[root@canldev01-fedora39 ~]# systemctl status strongswan
× strongswan.service - strongSwan IPsec IKEv1/IKEv2 daemon using swanctl
     Loaded: loaded (/usr/lib/systemd/system/strongswan.service; disabled; preset: disabled)
    Drop-In: /usr/lib/systemd/system/service.d
             └─10-timeout-abort.conf
     Active: failed (Result: exit-code) since Wed 2024-11-20 13:40:13 EST; 7s ago
    Process: 1614492 ExecStart=/usr/sbin/charon-systemd (code=exited, status=66)
   Main PID: 1614492 (code=exited, status=66)
     Status: "charon initialization failed"
        CPU: 22ms

Nov 20 13:40:13 canldev01-fedora39.networklab.prod charon-systemd[1614492]: wolfssl FIPS mode unavailable (-203)
Nov 20 13:40:13 canldev01-fedora39.networklab.prod charon-systemd[1614492]: plugin 'wolfssl': failed to load - wolfssl_plugin_create returned NULL
Nov 20 13:40:13 canldev01-fedora39.networklab.prod charon-systemd[1614492]: created TUN device: ipsec0
Nov 20 13:40:13 canldev01-fedora39.networklab.prod charon-systemd[1614492]: feature CUSTOM:libcharon in critical plugin 'charon-systemd' has unmet dependency: NONCE_GEN
Nov 20 13:40:13 canldev01-fedora39.networklab.prod charon-systemd[1614492]: feature CUSTOM:libcharon-receiver in critical plugin 'charon-systemd' has unmet dependency: HASHER:HASH_SHA1
Nov 20 13:40:13 canldev01-fedora39.networklab.prod charon-systemd[1614492]: feature CUSTOM:libcharon-sa-managers in critical plugin 'charon-systemd' has unmet dependency: HASHER:HASH_SHA1
Nov 20 13:40:13 canldev01-fedora39.networklab.prod charon-systemd[1614492]: failed to load 3 critical plugin features
Nov 20 13:40:13 canldev01-fedora39.networklab.prod systemd[1]: strongswan.service: Main process exited, code=exited, status=66/NOINPUT
Nov 20 13:40:13 canldev01-fedora39.networklab.prod systemd[1]: strongswan.service: Failed with result 'exit-code'.
Nov 20 13:40:13 canldev01-fedora39.networklab.prod systemd[1]: Failed to start strongswan.service - strongSwan IPsec IKEv1/IKEv2 daemon using swanctl.
[root@canldev01-fedora39 ~]# 

```

anhu, 2024-11-20:
When you do that, is it running ./fips_hash.sh to regenerated the fips hash?

volga629-1, 2024-11-20:
Yes, 100% , here are spec file 

```
%build
# --enable-tlsv10
#--enable-wpas \
#--enable-all-crypto \
#--enable-writedup \
autoreconf -i
%configure --enable-all \
           --enable-reproducible-build \
           --enable-secure-renegotiation --enable-fastmath \
           --enable-ed25519 --enable-curve25519 \
           --enable-opensslall \
           --enable-ecc \
           --enable-ocsp \
           --disable-fpecc \
           --disable-aligndata \
           --disable-static \
           --disable-jni \
           --disable-crl-monitor\
           --disable-examples \
           --enable-opensslextra \
           --enable-all-crypto \
           --enable-tlsv10 \
           --enable-shared \
           --enable-secure-renegotiation \
%if %{build_fips}
           --enable-fips=ready \
%endif
           CFLAGS="-DWOLFSSL_STATIC_RSA" \
           C_EXTRA_FLAGS="-fPIC"

%if %{build_fips}
%{__make} %{?_smp_mflags}
if [ "no" = "yes" ]
then
echo Updating wolfSSL FIPS hash
bash -x  ./fips-hash.sh
%{__make} %{?_smp_mflags}
fi
%else
 %{__make} %{?_smp_mflags}
%endif

```

anhu, 2024-11-20:
Can you first simply run the steps that I specified earlier: 
```
unzip wolfssl-5.7.2-gplv3-fips-ready.zip
cd wolfssl-5.7.2-gplv3-fips-ready
./configure --enable-fips=ready
make
./fips_hash.sh
make
make check 
```

volga629-1, 2024-11-20:
> Can you please help us prioritize this request by telling us about yourself and your project?

Yes, please provide email and I will provide you info.

volga629-1, 2024-11-20:
I will try to simplify, but I need to stick with RPM system, because I can't tell 100% if there are issue when I need to repoint build to non standard locations.

FIPs hash update 

```
CC       src/libwolfssl_la-tls13.lo
  CC       src/libwolfssl_la-ocsp.lo
  CC       src/libwolfssl_la-crl.lo
  CC       src/libwolfssl_la-quic.lo
  CC       src/libwolfssl_la-dtls.lo
  CC       wolfcrypt/test/test.o
  CCLD     src/libwolfssl.la
  CCLD     wolfcrypt/benchmark/benchmark
  CCLD     wolfcrypt/test/testwolfcrypt
make[2]: Leaving directory '/home/volga629/rpmbuild/BUILD/wolfssl-fips-5.7.4'
make[1]: Leaving directory '/home/volga629/rpmbuild/BUILD/wolfssl-fips-5.7.4'
+ echo Updating wolfSSL FIPS hash
Updating wolfSSL FIPS hash
+ bash -x ./fips-hash.sh
+ test '!' -x ./wolfcrypt/test/testwolfcrypt
+ test '!' -s ./wolfcrypt/src/fips_test.c
++ ./wolfcrypt/test/testwolfcrypt
++ sed -n 's/hash = \(.*\)/\1/p'
+ NEWHASH=E8231EC26852ADCAB614EC10A58EBBC00C766C87849B61872C32A33B831FCF3A
+ test -n E8231EC26852ADCAB614EC10A58EBBC00C766C87849B61872C32A33B831FCF3A
+ cp wolfcrypt/src/fips_test.c wolfcrypt/src/fips_test.c.bak
+ sed 's/^".*";/"E8231EC26852ADCAB614EC10A58EBBC00C766C87849B61872C32A33B831FCF3A";/' wolfcrypt/src/fips_test.c.bak
+ /usr/bin/make -j4
/usr/bin/make -j5  all-recursive
make[1]: Entering directory '/home/volga629/rpmbuild/BUILD/wolfssl-fips-5.7.4'
make[1]: warning: -j5 forced in submake: resetting jobserver mode.
make[2]: Entering directory '/home/volga629/rpmbuild/BUILD/wolfssl-fips-5.7.4'
make[2]: warning: -j5 forced in submake: resetting jobserver mode.
  CC       wolfcrypt/src/src_libwolfssl_la-fips_test.lo
  CCLD     src/libwolfssl.la
  CCLD     wolfcrypt/benchmark/benchmark
  CCLD     wolfcrypt/test/testwolfcrypt
make[2]: Leaving directory '/home/volga629/rpmbuild/BUILD/wolfssl-fips-5.7.4'
make[1]: Leaving directory '/home/volga629/rpmbuild/BUILD/wolfssl-fips-5.7.4'
+ RPM_EC=0
++ jobs -p
+ exit 0
Executing(%install): /bin/sh -e /var/tmp/rpm-tmp.Bu3s6T
+ umask 022
+ cd /home/volga629/rpmbuild/BUILD
+ '[' /home/volga629/rpmbuild/BUILDROOT/wolfssl-fips-5.7.4-3.fc39.x86_64 '!=' / ']'
```

volga629-1, 2024-11-20:
I recompiled again clean state and  this time wolfssl loaded correctly. 
Strongswan started with wolfssl fips.
I am not sure of possible to confirm which crypto engine is in use by strongswan 

```
[root@canldev01-fedora39 charon]# swanctl --stats
plugin 'sqlite': failed to load - sqlite_plugin_create not found and no plugin file available
uptime: 56 seconds, since Nov 20 14:23:25 2024
worker threads: 16 total, 7 idle, working: 4/0/5/0
job queues: 0/0/0/0
jobs scheduled: 0
IKE_SAs: 0 total, 0 half-open
mallinfo: sbrk 2555904, mmap 266240, used 431488, free 2124416
loaded plugins: charon-systemd unbound sha2 nonce pubkey ipseckey wolfssl curve25519 kernel-libipsec kernel-netlink socket-default bypass-lan vici counters
```

anhu, 2024-11-20:
@volga629-1 , 

> Yes, please provide email and I will provide you info.

You can reach me at anthony@wolfssl.com .  I look forward to your message. 

Warm regards, Anthony

volga629-1, 2024-11-20:
When I compile on Windows same think fips hash update empty 

```
*** libtool will only create a static version of it.
  CCLD     wolfcrypt/benchmark/benchmark.exe
  CCLD     wolfcrypt/test/testwolfcrypt.exe
make[2]: Leaving directory '/home/slava.bendersky/strongswan-5.9.14/wolfssl-5.7.4-gplv3-fips-ready'
make[1]: Leaving directory '/home/slava.bendersky/strongswan-5.9.14/wolfssl-5.7.4-gplv3-fips-ready'

slava.bendersky@VDT-SlavaB MSYS ~/strongswan-5.9.14/wolfssl-5.7.4-gplv3-fips-ready
$ bash -x ./fips-hash.sh
+ test '!' -x ./wolfcrypt/test/testwolfcrypt
+ test '!' -s ./wolfcrypt/src/fips_test.c
++ ./wolfcrypt/test/testwolfcrypt
++ sed -n 's/hash = \(.*\)/\1/p'
+ NEWHASH=
+ test -n ''
```

anhu, 2024-11-20:
Please look in the fips-hash.sh script to see what it is doing.  Basically it, run the tests, and lets them fail and the fail message will give you the new hash.  Are you getting the fail message with the new hash?

volga629-1, 2024-11-21:
I rebuilt Windows version  one time and  still fips hash is not updating properly.
I see that new hash is generated and updates the right file, but strongswan on load always get -203 

```
$ ./wolfcrypt/test/testwolfcrypt
------------------------------------------------------------------------------
 wolfSSL version 5.7.4
------------------------------------------------------------------------------
FIPS module version in use: wolfCrypt v7.0.0
error    test passed!
MEMORY   test passed!
base64   test passed!
base16   test passed!
asn      test passed!
in my Fips callback, ok = 0, err = -203
message = In Core Integrity check FIPS error
hash = 2CFD1B08B615068E2AD24A0E8CD9B00423414684111A76F1C0DA241068119C34
In core integrity hash check failure, copy above hash
into verifyCore[] in fips_test.c and rebuild
RANDOM   test failed!
 error L=17866 code=-197 (FIPS mode not allowed error)
 [fiducial line numbers: 9103 28041 46740 59294]
Exiting main with return code: -1

```

volga629-1, 2024-11-21:
Can you please post dependency for windows ? 
I want to validate that I am not missing anything.
I sent you email with details.

volga629-1, 2024-11-21:
@anhu  Should I try downgrade wolfssl ?  I so that you use 5.7.2

volga629-1, 2024-11-21:
I can't use visual studio, because need customize options for strongswan support, unless need adjust user_options.h.
I tried before and had similar issue.

anhu, 2024-11-21:
Yes, that is how you would do it. First you build on linux and make sure strongswan works there.  Then you look at the `options.h` that is generated on linux and use that as a guide to figure out what kind of changes you need to make in `user_settings.h` on windows.

anhu, 2024-11-21:
Lets take your fips hash problems out of the equation first.  You can do this: `--enable-fips=disabled`

anhu, 2024-11-21:
Please also note that you're rpm never executes the `fips_hash.sh`

volga629-1, 2024-11-21:
RPM process execute fips_hash.sh and  strongswan is loading the wolfssl fips library not issue 

```
CC       src/libwolfssl_la-ssl.lo
  CC       src/libwolfssl_la-tls.lo
  CC       src/libwolfssl_la-tls13.lo
  CC       src/libwolfssl_la-ocsp.lo
  CC       src/libwolfssl_la-crl.lo
  CC       src/libwolfssl_la-quic.lo
  CC       src/libwolfssl_la-dtls.lo
  CC       wolfcrypt/test/test.o
  CCLD     src/libwolfssl.la
  CCLD     wolfcrypt/benchmark/benchmark
  CCLD     wolfcrypt/test/testwolfcrypt
make[2]: Leaving directory '/home/volga629/rpmbuild/BUILD/wolfssl-fips-5.7.4'
make[1]: Leaving directory '/home/volga629/rpmbuild/BUILD/wolfssl-fips-5.7.4'
+ echo Updating wolfSSL FIPS hash
Updating wolfSSL FIPS hash
+ bash -x ./fips-hash.sh
+ test '!' -x ./wolfcrypt/test/testwolfcrypt
+ test '!' -s ./wolfcrypt/src/fips_test.c
++ ./wolfcrypt/test/testwolfcrypt
++ sed -n 's/hash = \(.*\)/\1/p'
+ NEWHASH=E8231EC26852ADCAB614EC10A58EBBC00C766C87849B61872C32A33B831FCF3A
+ test -n E8231EC26852ADCAB614EC10A58EBBC00C766C87849B61872C32A33B831FCF3A
+ cp wolfcrypt/src/fips_test.c wolfcrypt/src/fips_test.c.bak
+ sed 's/^".*";/"E8231EC26852ADCAB614EC10A58EBBC00C766C87849B61872C32A33B831FCF3A";/' wolfcrypt/src/fips_test.c.bak
+ /usr/bin/make -j4
/usr/bin/make -j5  all-recursive
make[1]: Entering directory '/home/volga629/rpmbuild/BUILD/wolfssl-fips-5.7.4'
make[1]: warning: -j5 forced in submake: resetting jobserver mode.
make[2]: Entering directory '/home/volga629/rpmbuild/BUILD/wolfssl-fips-5.7.4'
make[2]: warning: -j5 forced in submake: resetting jobserver mode.
  CC       wolfcrypt/src/src_libwolfssl_la-fips_test.lo
  CCLD     src/libwolfssl.la
  CCLD     wolfcrypt/benchmark/benchmark
  CCLD     wolfcrypt/test/testwolfcrypt
make[2]: Leaving directory '/home/volga629/rpmbuild/BUILD/wolfssl-fips-5.7.4'
make[1]: Leaving directory '/home/volga629/rpmbuild/BUILD/wolfssl-fips-5.7.4'
+ RPM_EC=0
++ jobs -p
+ exit 0

```

volga629-1, 2024-11-21:
```
[root@canldev01-fedora39 ~]# swanctl --stats
plugin 'sqlite': failed to load - sqlite_plugin_create not found and no plugin file available
uptime: 22 hours, since Nov 20 14:23:25 2024
worker threads: 16 total, 7 idle, working: 4/0/5/0
job queues: 0/0/0/0
jobs scheduled: 0
IKE_SAs: 0 total, 0 half-open
mallinfo: sbrk 2555904, mmap 266240, used 434080, free 2121824
loaded plugins: charon-systemd unbound sha2 nonce pubkey ipseckey wolfssl curve25519 kernel-libipsec kernel-netlink socket-default bypass-lan vici counters
[root@canldev01-fedora39 ~]# cat /etc/strongswan/strongswan.d/charon/wolfssl.conf 
wolfssl {

    # Enable to prevent loading the plugin if wolfSSL is not in FIPS mode.
    fips_mode = yes

    # Whether to load the plugin. Can also be an integer to increase the
    # priority of this plugin.
    load = yes

}

[root@canldev01-fedora39 ~]# 

```

anhu, 2024-11-21:
My colleagues pointed out you can use your current method to generate an options .h and then use that to customize your user_settings.h in the visual studio build.

volga629-1, 2024-11-21:
Let me try this method

volga629-1, 2024-11-21:
This list which I am going to try 

```
#define WOLFSSL_HAVE_ATOMIC_H
#define WOLFSSL_WOLFSSH
#define HAVE_C___ATOMIC 1
#define HAVE_THREAD_LS
#define NO_DO178
#define HAVE_REPRODUCIBLE_BUILD
#define HAVE_LINUXKM_PIE_SUPPORT
#define WOLFSSL_X86_64_BUILD
#define WOLFSSL_ASN_TEMPLATE
#define USE_FAST_MATH
#define HAVE_CRL_IO
#define HAVE_IO_TIMEOUT
#define WOLFSSL_DER_LOAD
#define KEEP_OUR_CERT
#define KEEP_PEER_CERT
#define WOLFSSL_SUBJ_DIR_ATTR
#define WOLFSSL_FPKI
#define WOLFSSL_SUBJ_INFO_ACC
#define WOLFSSL_CERT_NAME_ALL
#define WOLFSSL_VERBOSE_ERRORS
#define HAVE_AES_DECRYPT
#define HAVE_AES_ECB
#define WOLFSSL_ALT_NAMES
#define HAVE_FFDHE_2048
#define HAVE_FFDHE_3072
#define WOLFSSL_ASN_ALL
#define WOLFSSL_DH_EXTRA
#define WOLFSSL_ECDSA_DETERMINISTIC_K_VARIANT
#define WOLFSSL_HAVE_ISSUER_NAMES
#define HAVE_ECH
#define WOLFSSL_DTLS
#define WOLFSSL_QUIC
#define HAVE_EX_DATA
#define WOLFSSL_POST_HANDSHAKE_AUTH
#define WOLFSSL_SEND_HRR_COOKIE
#define WOLFSSL_LIBWEBSOCKETS
#define HAVE_EX_DATA
#define OPENSSL_NO_EC
#define ERROR_QUEUE_PER_THREAD
#define TFM_TIMING_RESISTANT
#define ECC_TIMING_RESISTANT
#define WC_RSA_BLINDING
#define PERSIST_SESSION_CACHE
#define PERSIST_CERT_CACHE
#define ATOMIC_USER
#define WOLFSSL_AES_CBC_LENGTH_CHECKS
#define HAVE_AESCCM
#define WOLFSSL_AES_EAX
#define WOLFSSL_AES_OFB
#define WOLFSSL_AES_DIRECT
#define WOLFSSL_AES_CFB
#define WOLFSSL_AESNI
#define USE_INTEL_SPEEDUP
#define HAVE_CAMELLIA
#define WOLFSSL_MD2
#define WOLFSSL_RIPEMD
#define HAVE_BLAKE2
#define HAVE_BLAKE2B
#define HAVE_BLAKE2S
#define WOLFSSL_SHA224
#define WOLFSSL_SHA512
#define WOLFSSL_SHA384
#define SESSION_CERTS
#define WOLFSSL_SEP
#define KEEP_PEER_CERT
#define HAVE_HKDF
#define HAVE_HPKE
[... 227 lines cut ...]
#define WOLFSSL_VALIDATE_FFC_IMPORT
#define HAVE_FFDHE_Q
#define HAVE_FFDHE_3072
#define HAVE_FFDHE_4096
#define HAVE_FFDHE_6144
#define HAVE_FFDHE_8192
#define WC_RNG_SEED_CB
#define WOLFSSL_ECDSA_SET_K
#define WOLFSSL_VALIDATE_ECC_IMPORT
#define WOLFSSL_VALIDATE_ECC_KEYGEN
#define HAVE_ECC192
#define HAVE_ECC224
#define HAVE_ECC256
#define HAVE_ED448
#define HAVE_ED448_KEY_IMPORT
#define WOLFSSL_NOSHA512_224
#define WOLFSSL_NOSHA512_256
#define WOLFSSL_AES_DIRECT
#define HAVE_AES_ECB
#define NO_OLD_TLS
#define WOLFSSL_AES_XTS
#define WOLFSSL_AES_DIRECT
#define HAVE_CURVE448
#define HAVE_ED448
```

volga629-1, 2024-11-21:
@anhu  can you please confirm that user_setting.h look correct 

```
#ifndef _WIN_USER_SETTINGS_H_
#define _WIN_USER_SETTINGS_H_
#define WOLFSSL_HAVE_ATOMIC_H
#define WOLFSSL_WOLFSSH
#define HAVE_C___ATOMIC 1
#define HAVE_THREAD_LS
#define NO_DO178
#define HAVE_REPRODUCIBLE_BUILD
#define HAVE_LINUXKM_PIE_SUPPORT
#define WOLFSSL_X86_64_BUILD
#define WOLFSSL_ASN_TEMPLATE
#define USE_FAST_MATH
#define HAVE_CRL_IO
#define HAVE_IO_TIMEOUT
#define WOLFSSL_DER_LOAD
#define KEEP_OUR_CERT
#define KEEP_PEER_CERT
#define WOLFSSL_SUBJ_DIR_ATTR
#define WOLFSSL_FPKI
#define WOLFSSL_SUBJ_INFO_ACC
#define WOLFSSL_CERT_NAME_ALL
#define WOLFSSL_VERBOSE_ERRORS
#define HAVE_AES_DECRYPT
#define HAVE_AES_ECB
#define WOLFSSL_ALT_NAMES
#define HAVE_FFDHE_2048
#define HAVE_FFDHE_3072
#define WOLFSSL_ASN_ALL
#define WOLFSSL_DH_EXTRA
#define WOLFSSL_ECDSA_DETERMINISTIC_K_VARIANT
#define WOLFSSL_HAVE_ISSUER_NAMES
#define HAVE_ECH
#define WOLFSSL_DTLS
#define WOLFSSL_QUIC
#define HAVE_EX_DATA
#define WOLFSSL_POST_HANDSHAKE_AUTH
#define WOLFSSL_SEND_HRR_COOKIE
#define WOLFSSL_LIBWEBSOCKETS
#define HAVE_EX_DATA
#define OPENSSL_NO_EC
#define ERROR_QUEUE_PER_THREAD
#define TFM_TIMING_RESISTANT
#define ECC_TIMING_RESISTANT
#define WC_RSA_BLINDING
#define PERSIST_SESSION_CACHE
#define PERSIST_CERT_CACHE
#define ATOMIC_USER
#define WOLFSSL_AES_CBC_LENGTH_CHECKS
#define HAVE_AESCCM
#define WOLFSSL_AES_EAX
#define WOLFSSL_AES_OFB
#define WOLFSSL_AES_DIRECT
#define WOLFSSL_AES_CFB
#define WOLFSSL_AESNI
#define USE_INTEL_SPEEDUP
#define HAVE_CAMELLIA
#define WOLFSSL_MD2
#define WOLFSSL_RIPEMD
#define HAVE_BLAKE2
#define HAVE_BLAKE2B
#define HAVE_BLAKE2S
#define WOLFSSL_SHA224
#define WOLFSSL_SHA512
#define WOLFSSL_SHA384
#define SESSION_CERTS
#define WOLFSSL_SEP
#define KEEP_PEER_CERT
[... 382 lines cut ...]
        #define NO_PSK
        #define HAVE_EXTENDED_MASTER
        #define WOLFSSL_SNIFFER
        #define HAVE_SECURE_RENEGOTIATION

        #define HAVE_AESGCM
        #define WOLFSSL_AES_XTS
        #define WOLFSSL_SHA384
        #define WOLFSSL_SHA512

        #define HAVE_SUPPORTED_CURVES
        #define HAVE_TLS_EXTENSIONS

        #define HAVE_ECC
        #define ECC_SHAMIR
        #define ECC_TIMING_RESISTANT
    #else
        /* The servers and clients */
        #define OPENSSL_EXTRA
        #define NO_PSK
    #endif
#endif /* HAVE_FIPS */

#endif /* _WIN_USER_SETTINGS_H_ */
```

anhu, 2024-11-21:
Hi Volga, this seems like a bit too much.  For example, you have HAVE_LINUXKM_PIE_SUPPORT which I would imagine isn't required. Same with HAVE_CAMELLIA and many others.  May I ask how you constructed this file?

volga629-1, 2024-11-21:
I pulled what is compiled on linux. 

Normally  I do configure 

```
./configure --build=x86_64-w64-mingw32 \
		    --host=x86_64-w64-mingw32 \
			--enable-all \
           --enable-reproducible-build \
           --enable-secure-renegotiation --enable-fastmath \
           --enable-ed25519 --enable-curve25519 \
           --enable-opensslall \
           --enable-ecc \
           --enable-ocsp \
           --disable-fpecc \
           --disable-aligndata \
           --disable-static \
           --disable-jni \
           --disable-crl-monitor\
           --disable-examples \
           --enable-opensslextra \
           --enable-all-crypto \
           --enable-shared \
           --enable-secure-renegotiation \
           --enable-fips=ready \
           CFLAGS="-DWOLFSSL_STATIC_RSA" \
		   LDFLAGS="-lws2_32 -lcrypt32 -Wl,-s -Wl,--gc-sections" \
           C_EXTRA_FLAGS="-fPIC"
```

volga629-1, 2024-11-21:
I am not 100% what required for strongswan that why is in use that large list.

volga629-1, 2024-11-21:
User_options.h alms impossible to figure out what exactly is need and what is valid.

anhu, 2024-11-21:
Hi, 
Here is the diff between `./configure` and `./configure --enable-strongswan`

```
--- /tmp/options.h	2024-11-21 16:41:28.088747244 -0500
+++ wolfssl/options.h	2024-11-21 16:41:44.626057902 -0500
@@ -57,6 +57,9 @@
 #undef  WOLFSSL_SHA384
 #define WOLFSSL_SHA384
 
+#undef  SESSION_CERTS
+#define SESSION_CERTS
+
 #undef  HAVE_HKDF
 #define HAVE_HKDF
 
@@ -108,6 +111,12 @@
 #undef  HAVE_HASHDRBG
 #define HAVE_HASHDRBG
 
+#undef  HAVE_OPENSSL_CMD
+#define HAVE_OPENSSL_CMD
+
+#undef  HAVE_CRL
+#define HAVE_CRL
+
 #undef  HAVE_TLS_EXTENSIONS
 #define HAVE_TLS_EXTENSIONS
 
@@ -162,8 +171,59 @@
 #undef  WOLFSSL_SYS_CA_CERTS
 #define WOLFSSL_SYS_CA_CERTS
 
-#undef  NO_DES3
-#define NO_DES3
+#undef  WOLFSSL_KEY_GEN
+#define WOLFSSL_KEY_GEN
+
+#undef  WOLFSSL_CERT_REQ
+#define WOLFSSL_CERT_REQ
+
+#undef  WOLFSSL_CERT_GEN
+#define WOLFSSL_CERT_GEN
+
+#undef  WOLFSSL_CERT_EXT
+#define WOLFSSL_CERT_EXT
+
+#undef  HAVE_OCSP
+#define HAVE_OCSP
+
+#undef  WOLFSSL_DES_ECB
+#define WOLFSSL_DES_ECB
+
+#undef  WOLFSSL_LOG_PRINTF
+#define WOLFSSL_LOG_PRINTF
+
+#undef  WOLFSSL_PUBLIC_MP
+#define WOLFSSL_PUBLIC_MP
+
+#undef  HAVE_EX_DATA
+#define HAVE_EX_DATA
+
+#undef  OPENSSL_ALL
+#define OPENSSL_ALL
+
+#undef  WOLFSSL_EITHER_SIDE
+#define WOLFSSL_EITHER_SIDE
+
+#undef  WC_RSA_NO_PADDING
+#define WC_RSA_NO_PADDING
+
+#undef  WC_RSA_PSS
+#define WC_RSA_PSS
+
+#undef  WOLFSSL_PSS_LONG_SALT
+#define WOLFSSL_PSS_LONG_SALT
+
+#undef  WOLFSSL_TICKET_HAVE_ID
+#define WOLFSSL_TICKET_HAVE_ID
+
+#undef  WOLFSSL_ERROR_CODE_OPENSSL
+#define WOLFSSL_ERROR_CODE_OPENSSL
+
+#undef  WOLFSSL_CERT_NAME_ALL
+#define WOLFSSL_CERT_NAME_ALL
+
+#undef  OPENSSL_EXTRA
+#define OPENSSL_EXTRA
 
 #undef  NO_DES3_TLS_SUITES
 #define NO_DES3_TLS_SUITES
@@ -174,12 +234,18 @@
 #undef  HAVE_AESGCM
 #define HAVE_AESGCM
 
+#undef  WOLFSSL_DTLS
+#define WOLFSSL_DTLS
+
 #undef  HAVE_TLS_EXTENSIONS
 #define HAVE_TLS_EXTENSIONS
 
 #undef  HAVE_SERVER_RENEGOTIATION_INFO
 #define HAVE_SERVER_RENEGOTIATION_INFO
 
+#undef  HAVE_COMP_KEY
+#define HAVE_COMP_KEY
+
 #undef  HAVE___UINT128_T
 #define HAVE___UINT128_T 1
```

In theory, if you make these changes (removals and additions) to the visual studio's user settings file, you should get what you need.

volga629-1, 2024-11-21:
Need  let me try 
I might will need to include more ciphers.

volga629-1, 2024-11-22:
Hello @anhu  

I tried to build and strongswan is failing , see like missing some option

```
In file included from ../../../../src/libstrongswan/plugins/plugin.h:28,
                 from wolfssl_plugin.h:34,
                 from wolfssl_plugin.c:29:
wolfssl_plugin.c: In function 'get_features':
../../../../src/libstrongswan/plugins/plugin_feature.h:321:119: error: 'FEATURE_WC_RNG' undeclared (first use in this function); did you mean 'FEATURE_RNG'?
  321 | #define __PLUGIN_FEATURE_REGISTER(type, _f)                                     (plugin_feature_t){ FEATURE_REGISTER, FEATURE_##type, .arg.reg.f = _f }
```

anhu, 2024-11-22:
@volga629-1 , 

I think this post might interest you: https://github.com/strongswan/strongswan/issues/2410

volga629-1, 2024-11-22:
Yes, I so it  I will  try compare configure options in options.h  with this list 

```
 ./configure --host=x86_64-w64-mingw32 --enable-reproducible-build --enable-keygen --enable-rsapss \
			--enable-secure-renegotiation --enable-fastmath \
			--enable-ed25519 --enable-curve25519 \
			--enable-debug \
			--enable-fips=ready \
			--enable-opensslall \
			--enable-ecc \
			--enable-ocsp \
			--enable-crl \
			--enable-psk \
			--disable-fpecc \
			--disable-aligndata \
			--disable-jni \
			--disable-crl-monitor\
			--disable-examples \
			LDFLAGS="-lws2_32 -lcrypt32 -Wl,-s -Wl,--gc-sections"
			 ./configure --host=x86_64-w64-mingw32 --enable-reproducible-build --enable-keygen --enable-rsapss \
			--enable-secure-renegotiation --enable-fastmath \
			--enable-ed25519 --enable-curve25519 \
			--enable-debug \
			--enable-fips=ready \
			--enable-opensslall \
			--enable-ecc \
			--enable-ocsp \
			--enable-crl \
			--enable-psk \
			--disable-fpecc \
			--disable-aligndata \
			--disable-jni \
			--disable-crl-monitor\
			--disable-examples \
			LDFLAGS="-lws2_32 -lcrypt32 -Wl,-s -Wl,--gc-sections" ./configure --host=x86_64-w64-mingw32 --enable-reproducible-build --enable-keygen --enable-rsapss \
			--enable-secure-renegotiation --enable-fastmath \
			--enable-ed25519 --enable-curve25519 \
			--enable-debug \
			--enable-fips=ready \
			--enable-opensslall \
			--enable-ecc \
			--enable-ocsp \
			--enable-crl \
			--enable-psk \
			--disable-fpecc \
			--disable-aligndata \
			--disable-jni \
			--disable-crl-monitor\
			--disable-examples \
			LDFLAGS="-lws2_32 -lcrypt32 -Wl,-s -Wl,--gc-sections" ./configure --host=x86_64-w64-mingw32 --enable-reproducible-build --enable-keygen --enable-rsapss \
			--enable-secure-renegotiation --enable-fastmath \
			--enable-ed25519 --enable-curve25519 \
			--enable-debug \
			--enable-fips=ready \
			--enable-opensslall \
			--enable-ecc \
			--enable-ocsp \
			--enable-crl \
			--enable-psk \
			--disable-fpecc \
			--disable-aligndata \
			--disable-jni \
			--disable-crl-monitor\
			--disable-examples \
			LDFLAGS="-lws2_32 -lcrypt32 -Wl,-s -Wl,--gc-sections"```

volga629-1, 2024-11-22:
Yes, when I tried before compile wolfssl with option above on Windows , there was no strongswan compiler exit ( seems like quite failure ) .
I never used  WS_NO_RNG

volga629-1, 2024-11-22:
@anhu  RNG is issues is passed, right no having issue where ld linker is finding -lwolfssl . I tried to add -L , but that din't helped 

```
libtool: link: x86_64-w64-mingw32-gcc -Wno-format -Wno-format-security -Wno-implicit-fallthrough -Wno-missing-field-initializers -Wno-pointer-sign -Wno-sign-compare -Wno-type-limits -Wno-unused-parameter -g -O2 -Wall -Wno-pointer-sign -Wno-format-security -mno-ms-bitfields -include /home/slava.bendersky/strongswan-5.9.14/config.h -o .libs/ipsecdump.exe ipsecdump.o  -L/home/slava.bendersky/strongswan-5.9.14/wolfssl-5.7.4-gplv3-fips-ready/wolfssl-visual ./.libs/libstrongswan-kernel-wfp.a -lfwpuclnt ../../../../src/libstrongswan/.libs/libstrongswan.a -ldl -lws2_32 -lpsapi -lwinhttp -lunbound -lldns -lwolfssl
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/13.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: cannot find -lwolfssl: No such file or directory
collect2.exe: error: ld returned 1 exit status
make[5]: *** [Makefile:597: ipsecdump.exe] Error 1
make[5]: Leaving directory '/home/volga629/strongswan-5.9.14/src/libcharon/plugins/kernel_wfp'
```

anhu, 2024-11-22:
Looks like you're missing `-L/path/to/wolfssl/library/file/`

anhu, 2024-11-22:
Interesting.  On linux,  `cannot find -lwolfssl: No such file or directory` means it cannot find `libwolfssl.so` ....but what is it looking for in the case of your environment?  I ask because for windows it builds a .dll file.

Have you considered trying to do this POC on Linux?

volga629-1, 2024-11-22:
POC Linux works fine
Major run in Windows 

Library location 

```

$ ls -la $(pwd)/wolfssl-5.7.4-gplv3-fips-ready/wolfssl-visual
total 11920
drwxr-xr-x 1 volga629 Domain Users       0 Nov 22 15:44 .
drwxr-xr-x 1 volga629 Domain Users       0 Nov 22 15:44 ..
-rwxr-xr-x 1 volga629 Domain Users  233472 Nov 22 15:40 test.exe
-rw-r--r-- 1 volga629 Domain Users 1028096 Nov 22 15:40 test.pdb
drwxr-xr-x 1 volga629 Domain Users       0 Nov 22 15:40 test_obj
-rwxr-xr-x 1 volga629 Domain Users 2157056 Nov 22 15:40 wolfssl-fips.dll
-rw-r--r-- 1 volga629 Domain Users  407811 Nov 22 15:40 wolfssl-fips.exp
-rw-r--r-- 1 volga629 Domain Users  686698 Nov 22 15:40 wolfssl-fips.lib
-rw-r--r-- 1 volga629 Domain Users 7655424 Nov 22 15:40 wolfssl-fips.pdb
drwxr-xr-x 1 volga629 Domain Users       0 Nov 22 15:40 wolfssl-fips_obj

```

Search 

```
 ld -L$(pwd)/wolfssl-5.7.4-gplv3-fips-ready/wolfssl-visual -lwolfssl
ld: cannot find -lwolfssl: No such file or directory

```

Search locations 

```

$ ld --verbose | grep SEARCH_DIR
SEARCH_DIR("/usr/x86_64-pc-msys/lib"); SEARCH_DIR("/usr/lib"); SEARCH_DIR("/usr/lib/w32api");
```

volga629-1, 2024-11-22:
@anhu  right now failing 

```
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/13.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: ../../../../src/libstrongswan/.libs/libstrongswan.a(wolfssl_crypter.o): in function `destroy':
C:/msys64/home/volga629/strongswan-5.9.14/src/libstrongswan/plugins/wolfssl/wolfssl_crypter.c:434:(.text+0x565): undefined reference to `wc_Des3Free'
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/13.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: ../../../../src/libstrongswan/.libs/libstrongswan.a(wolfssl_crypter.o): in function `wolfssl_crypter_create':
C:/msys64/home/volga629/strongswan-5.9.14/src/libstrongswan/plugins/wolfssl/wolfssl_crypter.c:619:(.text+0x670): undefined reference to `wc_Des3Init'
collect2.exe: error: ld returned 1 exit status
make[5]: *** [Makefile:597: ipsecdump.exe] Error 1
```

anhu, 2024-11-23:
Right.  You need to remove `NO_DES` and `NO_DES3`.  See if they are in your `user_setting.h`

volga629-1, 2024-11-25:
I solved issue with DES, but still issues with 

```
wolfssl_util.c:103:(.text+0x286): undefined reference to `sp_unsigned_bin_size'
```

volga629-1, 2024-11-25:
Thank you for reply build failing in visual studio, might missing something else

```
10:14:30:880	1>C:\msys64\home\volga629\strongswan-5.9.14\wolfssl-5.7.4-gplv3-fips-ready\wolfssl\wolfcrypt\sp_int.h(257,44): error C2061: syntax error: identifier '__attribute__'
10:14:30:880	1>(compiling source file '../../wolfcrypt/src/asn.c')
10:14:30:880	1>C:\msys64\home\volga629\strongswan-5.9.14\wolfssl-5.7.4-gplv3-fips-ready\wolfssl\wolfcrypt\sp_int.h(257,44): error C2059: syntax error: ';'
10:14:30:880	1>(compiling source file '../../wolfcrypt/src/asn.c')
10:14:30:944	1>C:\msys64\home\volga629\strongswan-5.9.14\wolfssl-5.7.4-gplv3-fips-ready\wolfssl\wolfcrypt\sp_int.h(258,44): error C2061: syntax error: identifier '__attribute__'
10:14:30:944	1>(compiling source file '../../wolfcrypt/src/asn.c')

```

anhu, 2024-11-25:
Hi @volga629-1 , 

You'll have to forgive me.  My co-working pointed out that I gave you bad advice.  Please don't use ` WOLFSSL_SP_MATH` , instead replace it with  `WOLFSSL_SP_MATH_ALL`.  My apologies. 

Warm regards, Anthony

volga629-1, 2024-11-25:
@anhu 

I removed USE_FAST_MATH, but build still failing is there are other conflicting  features  which need to be removed ?

volga629-1, 2024-11-25:
OK I found need remove 

```
#undef  HAVE___UINT128_T
#define HAVE___UINT128_T 1
```

volga629-1, 2024-11-25:
@anhu 

What else I need to add for FIPS build 
Strongswan comping is finished, but unable to load wolfssl FIPS 

```
$ ./usr/libexec/strongswan/charon-svc.exe
Starting Power On Self Test
Pre-Operational Self Test FAILURE
00[DMN] Starting IKE service charon-svc (strongSwan 5.9.14, Windows Client 10.0.22621 (SP 0.0))
00[LIB] wolfssl FIPS mode unavailable
00[LIB] plugin 'wolfssl': failed to load - wolfssl_plugin_create returned NULL
00[LIB] feature CUSTOM:libcharon in critical plugin 'charon-svc' has unmet dependency: NONCE_GEN
00[LIB] feature CUSTOM:libcharon-receiver in critical plugin 'charon-svc' has unmet dependency: HASHER:HASH_SHA1
00[LIB] feature CUSTOM:libcharon-sa-managers in critical plugin 'charon-svc' has unmet dependency: HASHER:HASH_SHA1
00[CFG] ipseckey plugin is enabled
00[LIB] failed to load 3 critical plugin features
```

anhu, 2024-11-25:
With all these configuration changes your fips hash has changed as well.  You'll need to recalculate your FIPS hash. 

Warm regards, Anthony

volga629-1, 2024-11-25:
How you update from Visual Studio ?

volga629-1, 2024-11-25:
You referring  to static library ?  Right now is build dynamic.

volga629-1, 2024-11-25:
You mean this ?

```
11:46:26:312	2>test.vcxproj -> C:\msys64\home\volga629\strongswan-5.9.14\wolfssl-5.7.4-gplv3-fips-ready\IDE\WIN10\DLL Release\x64\test.exe
11:46:26:333	2>Done building project "test.vcxproj".
11:46:26:357	========== Build: 2 succeeded, 0 failed, 0 up-to-date, 0 skipped ==========
11:46:26:357	========== Build completed at 11:46 AM and took 14.926 seconds ==========
```

volga629-1, 2024-11-25:
I replaced here are  is this correct ?

line 1916 fips_test.c 

#ifdef WOLFCRYPT_FIPS_CORE_HASH_VALUE

anhu, 2024-11-25:
Yes, the code looks like this: 

```
char verifyCore[] =
#ifdef WOLFCRYPT_FIPS_CORE_HASH_VALUE
WC_STRINGIFY(WOLFCRYPT_FIPS_CORE_HASH_VALUE);
#else
"027A1E13F50F2A56CAA5F655075F7C9DF96C35378B6DE2BD188EF44E4E57A757";
#endif

```

That value in the double quotes needs to change. 

Warm regards, Anthony

volga629-1, 2024-11-25:
@anhu  Might be I am not doing something incorrectly, but still unable to load wolfssl fips library. Same  error, I tried to load crypto library 

```
$ ./usr/libexec/strongswan/charon-svc.exe -v
Starting Power On Self Test
Pre-Operational Self Test SUCCESS
00[DMN] Starting IKE service charon-svc (strongSwan 5.9.14, Windows Client 10.0.22621 (SP 0.0))
00[LIB] wolfssl FIPS mode unavailable
00[LIB] plugin 'wolfssl': failed to load - wolfssl_plugin_create returned NULL
00[LIB] feature CUSTOM:libcharon in critical plugin 'charon-svc' has unmet dependency: NONCE_GEN
00[LIB] feature CUSTOM:libcharon-receiver in critical plugin 'charon-svc' has unmet dependency: HASHER:HASH_SHA1
00[LIB] feature CUSTOM:libcharon-sa-managers in critical plugin 'charon-svc' has unmet dependency: HASHER:HASH_SHA1
00[LIB] failed to load 3 critical plugin features
```

volga629-1, 2024-11-25:
Can you please clarify the order, when test.exe should be run in case of Visual studio is in use.
I ran test.exe and most tests is passed, but I don't see hash line .

```
$ ./test
Starting Power On Self Test
Pre-Operational Self Test SUCCESS
------------------------------------------------------------------------------
 wolfSSL version 5.7.4
------------------------------------------------------------------------------
FIPS module version in use: wolfCrypt v7.0.0
error    test passed!
wolfSSL Entering memory_test
MEMORY   test passed!
wolfSSL Entering base64_test
Bad Base64 Decode data, too big
Escape buffer max too small
base64   test passed!
wolfSSL Entering base16_test
base16   test passed!
wolfSSL Entering asn_test
asn      test passed!
wolfSSL Entering random_test
RNG_HEALTH_TEST_CHECK_SIZE = 128
sizeof(seedB_data)         = 128
RNG_HEALTH_TEST_CHECK_SIZE = 128
sizeof(seedB_data)         = 128
RANDOM   test passed!
wolfSSL Entering sha_test
SHA      test passed!
wolfSSL Entering sha224_test
SHA-224  test passed!
wolfSSL Entering sha256_test
SHA-256  test passed!
wolfSSL Entering sha384_test
SHA-384  test passed!
wolfSSL Entering sha512_test
SHA-512  test passed!
wolfSSL Entering sha3_test
SHA-3    test passed!
wolfSSL Entering hash_test
Hash     test passed!
wolfSSL Entering hmac_sha_test
HMAC-SHA test passed!
wolfSSL Entering hmac_sha224_test
HMAC-SHA224 test passed!
wolfSSL Entering hmac_sha256_test
HMAC-SHA256 test passed!
wolfSSL Entering hmac_sha384_test
HMAC-SHA384 test passed!
wolfSSL Entering hmac_sha512_test
HMAC-SHA512 test passed!
wolfSSL Entering hmac_sha3_test
HMAC-SHA3   test passed!
wolfSSL Entering hkdf_test
HMAC-KDF    test passed!
wolfSSL Entering sshkdf_test
SSH-KDF     test passed!
PRF         test passed!
TLSv1.2 KDF test passed!
wolfSSL Entering tls13_kdf_test
TLSv1.3 KDF test passed!
wolfSSL Entering srtpkdf_test
SRTP KDF test passed!
wolfSSL Entering gmac_test
RNG_HEALTH_TEST_CHECK_SIZE = 128
sizeof(seedB_data)         = 128
GMAC     test passed!
wolfSSL Entering des_test
DES      test passed!
[... 1497 lines cut ...]
GetLength - value exceeds buffer length
GetLength - value exceeds buffer length
GetLength - value exceeds buffer length
GetLength - value exceeds buffer length
GetLength - value exceeds buffer length
GetLength - value exceeds buffer length
GetLength - value exceeds buffer length
GetLength - value exceeds buffer length
GetLength - value exceeds buffer length
GetLength - value exceeds buffer length
GetLength - value exceeds buffer length
GetLength - value exceeds buffer length
ber-der  test passed!
wolfSSL Entering logging_test (debug)
logging  test passed!
wolfSSL Entering time_test
time     test passed!
wolfSSL Entering [wolfcrypt_]mutex_test (1)
mutex    test passed!
wolfSSL Entering memcb_test
memcb    test passed!
Test complete
wolfSSL Entering wolfCrypt_Cleanup
Exiting main with return code: 0
```

anhu, 2024-11-25:
OH, I see "RANDOM   test passed!" which means you have the correct hash.  The next possible problem might be that you need to call wc_SetSeed_Cb() and wolfCrypt_SetCb_fips() just before the wolfSSL initialization.  You can see the following example application to see what i mean: 

https://github.com/wolfSSL/wolfssl-examples/blob/master/X9.146/gen_dual_keysig_cert.c#L131

Warm regards, Anthony

volga629-1, 2024-11-25:
> OH, I see "RANDOM test passed!" which means you have the correct hash. The next possible problem might be that you need to call wc_SetSeed_Cb() and wolfCrypt_SetCb_fips() just before the wolfSSL [...]

My guess is in strongswan already should be done. I will check source code right now .

volga629-1, 2024-11-26:
Little a bit more details 

```
There isn't. As I pointed out already, certain aspects of the IKEv2 protocol ([RFC 7296](https://datatracker.ietf.org/doc/html/rfc7296)) require SHA-1.
````

volga629-1, 2024-11-26:
> Hi @volga629-1
> 
> I have setup a meeting with my colleagues for later today, but you seem to be giving this a high degree of urgency so I will give some interim advice.
> 
> Perhaps the best idea is to use our general wolfssl release (non-fips-ready) and get that working on windows as a proof of concept. Once you have seen how that functions and you are satisfied, then when you have a conversation with us about commercial licensing and support, you can highlight that you will need SHA1.
> 
> Please stay tuned; the meeting is set for late in the afternoon.
> 
> Warm regards, Anthony

Thank you for suggestion, I will try , but even for demo I will need FIPS 

@anhu  Question regard strongswan is failing to load  wolfssl 

Strongswan code is states 

```
/* New patched today */
#ifdef WC_RNG_SEED_CB
	wc_SetSeed_Cb(wc_GenerateSeed);
#endif

#ifdef HAVE_FIPS
	if (fips_mode)
	{
		int ret = wolfCrypt_GetStatus_fips();
		if (ret != 0)
		{
			DBG1(DBG_LIB, "wolfssl FIPS mode unavailable (%d)", ret);
			return NULL;
		}
	}
#else
	if (fips_mode)
	{
		DBG1(DBG_LIB, "wolfssl FIPS mode unavailable");
		return NULL;
	}
#endif

```

Is anything else prevent to load plugin ? 

In user_settings.h include 

```
#undef  HAVE_FIPS
#define HAVE_FIPS

#undef  HAVE_FIPS_VERSION
#define HAVE_FIPS_VERSION 7

#undef  HAVE_FIPS_VERSION_MAJOR
#define HAVE_FIPS_VERSION_MAJOR 7

#undef  HAVE_FIPS_VERSION_MINOR
#define HAVE_FIPS_VERSION_MINOR 0

#undef  HAVE_FIPS_VERSION_PATCH
#define HAVE_FIPS_VERSION_PATCH 0

```

anhu, 2024-11-26:
Hi  @volga629-1 , 

It seems we have many issues one after the other.  Why don't we setup a call over zoom or MS teams or Google Meet to get over these technical details?  Please send me an invite to a meeting with a link to your favourite meeting system for 9:30 am tomorrow morning.  You can send it to anthony@wolfssl.com . 

Warm regards, Anthony,.

volga629-1, 2024-11-26:
Thank you I will 

> Hi @volga629-1 ,
> 
> It seems we have many issues one after the other. Why don't we setup a call over zoom or MS teams or Google Meet to get over these technical details? Please send me an invite to a meeting with a link to your favourite meeting system for 9:30 am tomorrow morning. You can send it to [anthony@wolfssl.com](mailto:anthony@wolfssl.com) .
> 
> Warm regards, Anthony,.

I  will send you invite soon.

[22 more replies]

closed 2025-01-09 by anhu