wolfSSL/wolfssl #8220: If it is not a TLS1.3 website, and I set the client to use SNI, the program will crash

eighteen-k-gold-malow, 2024-11-24:
### Version

5.6.6

### Description

```

ctx ＝ wolfSSL_CTX_new (wolfTLS_client_method ())
wolfSSL_CTX_set_quiet_shutdown (ctx, 1)
wolfSSL_CTX_set_verify (ctx, #SSL_VM_NONE, 0)
wolfSSL_CTX_set_cipher_list (ctx, “ALL:!aPSK:!ECDSA+SHA1:!3DES”)

ssl ＝ wolfSSL_new (ctx)
if (hostname ≠ “”)
    wolfSSL_UseSNI (ssl, #WOLFSSL_SNI_HOST_NAME, hostname, 取文本长度 (hostname))

if (hostname ≠ “”) {
    wolfSSL_set_alpn_protos (ssl, "http/1.1", strlen ("http/1.1"))
}

wolfSSL_only_dhe_psk (ssl)

ssl_rbio = wolfSSL_BIO_new (wolfSSL_BIO_s_mem ())
ssl_wbio = wolfSSL_BIO_new (wolfSSL_BIO_s_mem ())
wolfSSL_set_bio (ssl, ssl_rbio, ssl_wbio)
wolfSSL_set_connect_state (ssl)

// send ssl handshake data
wolfSSL_SSL_do_handshake (ssl)
len ＝ wolfSSL_BIO_ctrl_pending (ssl_wbio)
wolfSSL_BIO_read (ssl_wbio, buf, len)
tcp_send(buf)

// the wolfSSL_SSL_do_handshake, it will crash
buf = tcp_read()
wolfSSL_write(ssl, buf, sizeof(buf))
wolfSSL_SSL_do_handshake (ssl)

```

embhorn, 2024-11-25:
Hi @eighteen-k-gold-malow 

The client should not crash if you send an SNI extension with TLS1.2

Can you try with the latest version of wolfSSL?

Have a look at this example using the BIO API:
https://github.com/wolfSSL/wolfssl-examples/blob/master/tls/client-tls-bio.c

Thanks,
@embhorn - wolfSSL Support

embhorn, 2025-02-24:
Since we never heard back, I'll go ahead and close this issue.

closed 2025-02-24 by embhorn