wolfSSL/wolfssl #8240: Support of WolfSSL on Aurix TC4d

Narratot, 2024-11-29:
### Version

5.7.4

### Description

I am trying to use WolfSSL for a research Project with the Aruix TC4D,
Is there a adapted WolfSSL version for this chip - Or a guide on how to get it working? 
Thank you

bigbrett, 2024-11-29:
Hi @Narratot,

Thanks for your interest in wolfSSL. There is no "out-of-the-box" configuration for AURIX TC4, nor special wolfSSL version for the chip, because there is not really anything AURIX-specific to get it to work. wolfSSL is highly portable, so the steps to build for AURIX would be the same as building for any generic target.

I recommend starting with the user_settings.h template in [examples/configs/user_settings.h](https://github.com/wolfSSL/wolfssl/blob/master/examples/configs/user_settings_template.h), and turning on/off features that you need via the #if 0 switches present in the file. See the[ "building wolfSSL" section of the wolfSSL manua](https://www.wolfssl.com/documentation/manuals/wolfssl/chapter02.html)l for more details on each feature.

 You can also look at [examples/IDE/GCC-ARM](https://github.com/wolfSSL/wolfssl/blob/master/IDE/GCC-ARM/Header/user_settings.h) for more inspiration on a configuration that targets a generic cortex-m microcontroller that would be a good start (though make sure not to uncomment/enable any of the ARM-specific optimizations). But if you compile the source files with a build configuration similar to cortex-m, making sure no ARM-specific assembly optimizations are used, then you should be able to arrive at a configuration that works. 

Don't hesitate to reach out to us at support@wolfssl.com if you have issues compiling the library.

Best,
Brett

anhu, 2024-11-29:
@bigbrett , Thanks for these excellent suggestions!! 

Hi @Narratot 

Here at wolfSSL we love learning about how the academic community is using our source code. Can you please tells more about yourself and your project?

    where are you located?
    what are the goals of your project?
    will there be a paper published based on your work on this project?
    is there a specific institution and/or professor associated to this project?
    any other relevant information you'd like to share.

If you are hesitant to share this information on a public platform, you can send me email at anthony@wolfssl.com

Warm regards, Anthony

anhu, 2025-03-29:
@Narratot ,  Care to share anything about your project?

closed 2025-05-15 by bigbrett